OE4T/meta-tegra #238: ixm219 camera branch?

deblauwetom, 2020-01-27:
Hello,

This is a question: what branch has been verified to work with the imx219 camera (raspberry pi v2 camera) on the jetson nano? Preferably jetpack 4.2 based, but can be later too.

Thanks,
Best regards

madisongh, 2020-01-27:
[This wiki page](https://github.com/madisongh/meta-tegra/wiki/Which-branch-should-I-use%3F) lists the branches and which versions of L4T/JetPack are available on them.  I don't have any cameras beyond the stock ones that come with the development kits I've got, though.

deblauwetom, 2020-01-27:
Ok thanks! Maybe I need to clarify a bit further:
My question was more like: did you actually see an image out of the camera, not just that /dev/video0 is created, but so it really works for you? I am now using the warrior branch. Because first I used zeus, but did not yet manage to get video images, it was all a bit scrambled, even though I got a /dev/video0 device. So it would be nice to know if anybody already saw it working using for instance the warrior branch. Thanks!

madisongh, 2020-01-27:
Yes, I have a simple script that sets up a gstreamer pipeline to capture from nvarguscamersrc and output on the display, similar to the examples that NVIDIA provides in the L4T documentation.

deblauwetom, 2020-01-28:
ok thanks, in the meantime the stock imx219 driver works indeed with the gstreamer stuff as you mentioned, thanks!

Just a sidenote: I actually need a custom camera driver in the kernel for another camera type, and it only works at runtime when compiled with the gcc that the kernel was compiled with. Just leaving this here as a warning to other users maybe: it seems for jetpack 4.2 there is a difference if you compile the kernel with the nvidia toolchain that is gcc 6.4 or the one in yocto which is gcc 7.x. I made a hybrid using KERNEL_CC to fix this.

closed 2020-01-28 by deblauwetom