OE4T/meta-tegra #239: nothing provides /bin/sh

paroque28, 2020-01-28:
DNF fails on recipe tegra-minimal-initramfs >> tegra-minimal-init with:

--> Starting dependency resolution
--> Finished dependency resolution
Error:
 Problem: conflicting requests
  - nothing provides /bin/sh needed by tegra-minimal-init-1.0-r0.jetson_nano
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)

paroque28, 2020-01-28:
I am still figuring it but I believe it needs something more in the RDEPENDS of tegra-minimal-init.bb
Adding: RDEPENDS = bash makes some progress

paroque28, 2020-01-28:
This is what happens if I try to add bash in the RDEPENDS:

Error Summary
-------------

2020-01-28T17:18:49Z CRITICAL Error: Transaction check error:
  file /proc conflicts between attempted installs of tegra-minimal-init-1.0-r0.jetson_nano and base-files-3.0.14-r89.jetson_nano
  file /sys conflicts between attempted installs of tegra-minimal-init-1.0-r0.jetson_nano and base-files-3.0.14-r89.jetson_nano

paroque28, 2020-01-28:
So I finally "fixed" it, don't really know what happened.

This were my changes:
diff --git a/recipes-core/initrdscripts/tegra-minimal-init_1.0.bb b/recipes-core/initrdscripts/tegra-minimal-init_1.0.bb
index 324dbd6..0ddf211 100644
--- a/recipes-core/initrdscripts/tegra-minimal-init_1.0.bb
+++ b/recipes-core/initrdscripts/tegra-minimal-init_1.0.bb
@@ -12,7 +12,7 @@ S = "${WORKDIR}"

 do_install() {
     install -m 0755 ${WORKDIR}/init-boot.sh ${D}/init
-    install -d ${D}/proc ${D}/sys ${D}/dev ${D}/tmp ${D}/mnt ${D}/run ${D}/usr
+    install -d ${D}/dev ${D}/tmp ${D}/mnt ${D}/run ${D}/usr
     mknod -m 622 ${D}/dev/console c 5 1
     install -d ${D}${sysconfdir}
     if [ -e ${WORKDIR}/platform-preboot-cboot.sh ]; then
@@ -24,7 +24,10 @@ do_install() {
     fi
 }

-RDEPENDS_${PN} = "${VIRTUAL-RUNTIME_base-utils}"
+RDEPENDS_${PN} = " \
+    ${VIRTUAL-RUNTIME_base-utils} \
+    bash \
+"
 RDEPENDS_${PN}_append_tegra = "${@' util-linux-blkid' if d.getVar('PREFERRED_PROVIDER_virtual/bootloader').startswith('cboot') else ''}"
 FILES_${PN} = "/"

madisongh, 2020-01-28:
The initrd normally gets built with busybox (the default value for VIRTUAL-RUNTIME_base-utils), which provides a shell.   Have you changed VIRTUAL-RUNTIME_base-utils in your build configuration, or done something else to exclude busybox from your builds?

paroque28, 2020-01-28:
Yes, that may be the problem. I am using Wind River Linux which by default doesn't have busybox included

madisongh, 2020-01-28:
Ouch.  OK, then perhaps I should explicitly use busybox for the initrd build.  It's supposed to be a very small image, and using something like bash plus the rest of packagegroup-core-base-utils would bulk it up pretty badly.

madisongh, 2020-01-29:
I wasn't able to reproduce the problem with dnf/rpm complaining about `/bin/sh` going missing, but I was able to reproduce the complaints about `/proc` and `/sys`.  You can work around those by adding a `tegra-minimal-init_%.bbappend` with:
```
do_install_append() {
    chmod 0555 ${D}/proc ${D}/sys
}
```
Alternatively (since you won't see the above complaints if not including base-files in the initrd's rootfs) - or in addition, include:
```
RDEPENDS_${PN}_remove = "packagegroup-core-base-utils"
RDEPENDS_${PN}_append = " busybox"
```
which should fix things so busybox provides the shell for the initrd.  I'll fix up the recipes to make this a bit cleaner, but for now this should take care of the issue.

(I ran some test builds with VIRTUAL_RUNTIME_base-utils set to packagegroup-core-base-utils, and the initrd size went from about 2MiB to over 30MiB.)

paroque28, 2020-01-29:
Adding your suggestions worked aswell. Thank you!

paroque28, 2020-02-04:
Your latest commits work aswell

closed 2020-02-04 by paroque28