OE4T/meta-tegra #240: libEGL_nvidia.so.0 : File not Found

paroque28, 2020-01-28:
Branch zeus

Error:

Exception: FileNotFoundError: [Errno 2] No such file or directory: '.../build/tmp-glibc/work/armv8a_tegra210-wrs-linux/tegra-libraries/32.2.3-r0/packages-split/tegra-libraries/usr/lib/aarch64-linux-gnu/tegra-egl/libEGL_nvidia.so.0'

ERROR: Logfile of failure stored in: .../build/tmp-glibc/work/armv8a_tegra210-wrs-linux/tegra-libraries/32.2.3-r0/temp/log.do_package_qa.16504
ERROR: Task (.../layers/meta-tegra/recipes-bsp/tegra-binaries/tegra-libraries_32.2.3.bb:do_package_qa) failed with exit code '1'

If I run a ls : 
```bash
$ ls -la .../build/tmp-glibc/work/armv8a_tegra210-wrs-linux/tegra-libraries/32.2.3-r0/packages-split/tegra-libraries/usr/lib/aarch64-linux-gnu/tegra-egl/libEGL_nvidia.so.0
lrwxrwxrwx 1 prodrigu users 29 Jan 28 17:16 .../build/tmp-glibc/work/armv8a_tegra210-wrs-linux/tegra-libraries/32.2.3-r0/packages-split/tegra-libraries/usr/lib/aarch64-linux-gnu/tegra-egl/libEGL_nvidia.so.0 -> /usr/lib64/libEGL_nvidia.so.0
```
So the problem is that I don't have anything installed on my machine and this file does not exist: /usr/lib64/libEGL_nvidia.so.0

Your do_install says:

```bash
do_install() {
    install -d ${D}${localstatedir}
    cp -R ${B}/var/nvidia ${D}${localstatedir}/
    install -d ${D}${libdir}/libv4l/plugins/
    for f in ${DRVROOT}/tegra/libv4l2_nv*; do
        install -m 0644 $f ${D}${libdir}/libv4l/plugins/
    done
    install -d ${D}${libdir}
    for f in ${DRVROOT}/tegra/lib*; do
        install -m 0644 $f ${D}${libdir}
    done
    for f in ${DRVROOT}/tegra-egl/lib*; do
        install -m 0644 $f ${D}${libdir}
    done
    ln -sf libcuda.so.1.1 ${D}${libdir}/libcuda.so
    ln -sf libcuda.so.1.1 ${D}${libdir}/libcuda.so.1
    for libname in nvbuf_fdmap nvbufsurface nvbufsurftransform nvbuf_utils; do
        ln -sf lib$libname.so.1.0.0 ${D}${libdir}/lib$libname.so.1
        ln -sf lib$libname.so.1.0.0 ${D}${libdir}/lib$libname.so
    done
    ln -sf libnvid_mapper.so.1.0.0 ${D}${libdir}/libnvid_mapper.so.1
    ln -sf libnvid_mapper.so.1.0.0 ${D}${libdir}/libnvid_mapper.so
    rm -f ${D}${libdir}/libdrm* ${D}${libdir}/libnvphsd* ${D}${libdir}/libnvgov*
    rm -f ${D}${libdir}/libv4l2.so* ${D}${libdir}/libv4lconvert.so*
    # argus and scf libraries hard-coded to use this path
    install -d ${D}/usr/lib/aarch64-linux-gnu/tegra-egl
    ln -sf ${libdir}/libEGL_nvidia.so.0 ${D}/usr/lib/aarch64-linux-gnu/tegra-egl/libEGL_nvidia.so.0
    .
    .
    .

```

Shouldn't it be like this?

` ln -sf ${D}${libdir}/libEGL_nvidia.so.0 ${D}/usr/lib/aarch64-linux-gnu/tegra-egl/libEGL_nvidia.so.0`

paroque28, 2020-01-28:
So I made this change and worked!

Although I got some warnings:
WARNING: tegra-libraries-32.2.3-r0 do_package_qa: QA Issue: Symlink /usr/lib/aarch64-linux-gnu/tegra-egl/libEGL_nvidia.so.0 in tegra-libraries points to TMPDIR [symlink-to-sysroot]
WARNING: tegra-libraries-32.2.3-r0 do_package_qa: QA Issue: tegra-libraries: found library in wrong location: /usr/lib/aarch64-linux-gnu/tegra-egl/libEGL_nvidia.so.0 [libdir]

madisongh, 2020-01-29:
No, the symlink is correct.  When installed on the device, the path will resolve correctly.

I'd say this is a bug in the package_qa_check_libdir() function, which should handle symlinks properly.  For now, just add a `tegra-libraries_%.bbappend` with
```
INSANE_SKIP_${PN} += "libdir"
```
to work around the issue.

This is only cropping up because your wrlinux distro is enabling multilib builds.  Turning that off should also resolve the issue.

paroque28, 2020-01-29:
Thanks, Madison. Adding that variable worked!

closed 2020-01-29 by paroque28