OE4T/meta-tegra #241: Zeus branch does not have the dosdcard.sh script for Jetson Nano

paroque28, 2020-01-28:
jetson-nano-20200128225419.tegraflash$ ls
bmp.blob     flash.xml        mkpkc                 sc7entry-firmware.bin                   tegrabct                tegrarcm
bsp_version  flash.xml.in     mksparse              sdcard-layout                           tegradevflash           tegrasign
cboot.bin    jetson-nano.cfg  nvtboot.bin           sdcard.xml.in                           tegraflash_internal.py  tos-mon-only.img
chkbdinfo    mkbctpart        nvtboot_cpu.bin       signed                                  tegraflash.py           u-boot-jetson-nano.bin
doflash.sh   mkbootimg        nvtboot_recovery.bin  tegra210-flash-helper.sh                tegrahost               verfile.txt
eks.img      mkgpt            rp4.blob              tegra210-p3448-0000-p3449-0000-a02.dtb  tegraparser             warmboot.bin

paroque28, 2020-01-28:
I found the image inside: tmp-glibc/deploy/images/jetson-nano/jetson-nano-20200128225419.sdcard

closed 2020-01-28 by paroque28

madisongh, 2020-01-29:
Right, the `dosdcard.sh` script was introduced with L4T R32.3.1 support.  For Nanos and R32.2.x, use the `.sdcard` image produced by the bitbake build.