OE4T/meta-tegra #242: Burn fuses on Jetson TX2 32.2.1

ichergui, 2020-01-29:
Hi Everyone,
I trying to burn fuses for the second time (new SBK and new RSA key). Unfortunately I did not succeed
For you information, I already applied the patch [fix-uid-check-in-odmfuse.patch.txt ](https://github.com/madisongh/meta-tegra/files/3746867/fix-uid-check-in-odmfuse.patch.txt)provided by Matt Madison
Please see the command line and the traces below:
**Command**
```bash 
$ sudo BOARDID=3310 FAB=C04 ./odmfuse.sh -j -i 0x18 -c PKC -k ./my_priv.pem -S ./my_sbk.key jetson-tx2                                       
```
**Traces**
```bash
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands
 
[   0.0032 ] Generating RCM messages
[   0.0040 ] tegrarcm_v2 --listrcm rcm_list.xml --chip 0x18 0 --download rcm /home/ilies/secureboot/mb1_recovery_prod.bin 0 0
[   0.0046 ] RCM 0 is saved as rcm_0.rcm
[   0.0050 ] RCM 1 is saved as rcm_1.rcm
[   0.0050 ] List of rcm files are saved in rcm_list.xml
[   0.0050 ] 
[   0.0050 ] Signing RCM messages
[   0.0056 ] tegrasign_v2 --key None --list rcm_list.xml --pubkeyhash pub_key.key
[   0.0062 ] Assuming zero filled SBK key
[   0.0087 ] 
[   0.0087 ] Copying signature to RCM mesages
[   0.0095 ] tegrarcm_v2 --chip 0x18 0 --updatesig rcm_list_signed.xml
[   0.0104 ] 
[   0.0104 ] Boot Rom communication
[   0.0110 ] tegrarcm_v2 --chip 0x18 0 --rcm rcm_list_signed.xml --skipuid
[   0.0117 ] RCM version 0X13
[   0.3470 ] Boot Rom communication failed
[   5.5740 ] 
Error: Return value 3
Command tegrarcm_v2 --chip 0x18 0 --rcm rcm_list_signed.xml --skipuid
Reading board information failed.
```
Have you guys an idea about the problem with Boot Room communication ?
Best regards, 
Ilies

madisongh, 2020-01-29:
When you say "for the second time", do you mean for the second time on the same device? Fuse burning is a one-time operation - once done, you cannot change them.

The odmfuse script can't talk to an already-secured device, which looks like the cause of the failure here.

ichergui, 2020-01-30:
Sorry for the late response.
Yes, I mean that I tried to burn the same device for the second time with new fuses.
Please see below what I have read from Nvidia documentation about the secure boot.

- NVIDIA SoCs contain multiple fuses that control different items for security and boot. Once a fuse bit is set to 1, you cannot change its value back to 0. For example, a fuse value of 1 (0x01) can be changed to 3 (0x03) or 5 (0x05), but not to 4 (0x4) because bit 0 is already programmed to 1.

madisongh, 2020-01-30:
Sure, but there are still a couple of problems:
1. As I mentioned above, the odmfuse script can't talk to an already-secured device.  Once a device is secured, communication over the USB port is signed and/or encrypted with the PKC.  As far as I know, NVIDIA doesn't provide a script that will let you burn other fuses via USB once the PKC/SBK are burned.
2. The PKC and SBK are essentially random sequences of 1s and 0s, so burning new ones isn't really feasible.  You'd only be able to change 0 bits to 1s, and the result in the fuses wouldn't match your new keys.

Oh, and if you burned the OdmProductionMode fuse on your first pass, the PKC and SBK are locked against any further changes anyway.

NVIDIA does provide access to the fuses from Linux through sysfs, and while I don't know if they officially support doing so, you can burn other fuses (e.g. the ODM_RESERVED fuses) using that interface.  I've used it to do some fixups on some fuse settings after partial failures with using the odmfuse script.

madisongh, 2020-03-04:
@ichergui did you get your fusing issues straightened out?

ichergui, 2020-03-09:
Hi @madisongh 
Not yet, still working on that
I'm waiting for the new JetPack 4.4 to do more tests
Please refer to the following link: https://devtalk.nvidia.com/default/topic/1064080/jetson-tx2/secure-boot-odmfuse-programming-step-fail-on-32-2-1/post/5425881/#5425881

madisongh, 2020-03-09:
Thanks for the pointer.  Since it doesn't look like there's a specific issue with the meta-tegra layer here, I'm going to close this.

closed 2020-03-09 by madisongh