OE4T/meta-tegra #244: Port jetson-nano from zeus to zeus-l4t-r32.3.1

paroque28, 2020-02-04:
It's much better to have the .sdcard image built than running the dosdcard.sh script each time

madisongh, 2020-03-04:
I moved away from pre-generating the sdcard image during the build to a post-build script for some good reasons:
1. With L4T R32.3.1, it looked like there could be different variants of Nano module in circulation, with each variant requiring a different device tree. With the post-build script, the device tree selection can be made based on the version information from the module.  (Or manually, if the user knows the specific variant.)
2. Sizing the SDcard image was a problem, especially if you use different SDcards of different sizes. With the script, the size can be determined when the script runs.  And if the SDcard is in a high-speed card reader/writer on your build host, the formatting and writing time can be much faster than `dd`ing the SDcard image.
3. USB flashing speed with the Nano improved with R32.3.1.  The slow flashing time was one reason I originally implemented the SDcard image approach.

That said, I'll see if I can at least provide the choice of generating an SDcard image.  It will likely be a bit different from the old mechanism, but it should be possible to do.

ichergui, 2021-03-03:
Hey guys, 

@madisongh Is there any action from us concerning this issue ?

madisongh, 2021-12-18:
With L4T R32.5.x and later, NVIDIA changed the partition layout for the SDcard-based Nanos such that only the rootfs gets loaded on the card. For Yocto builds, that means you can just transfer the `ext4` filesystem image into the `APP` partition on the card, without having to unpack the `tegraflash` package and run `dodscard.sh`.

So I don't think there's anything more we really need to do here.

closed 2021-12-18 by madisongh