OE4T/meta-tegra #245: cuda-samples:  cannot find -lcublas or any other cuda library

paroque28, 2020-02-05:
So I found that this is a multilib issue and found how to fix it.

If you look at my sysroot for this package you will see that it is being installed in lib64 rather than lib

build/tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-samples/10.0.326-1-r0$ find . -name *cublas*  ./recipe-sysroot/sysroot-providers/cuda-cublas
./recipe-sysroot/usr/lib64/pkgconfig/cublas-10.0.pc
./recipe-sysroot/usr/local/cuda-10.0/include/cublas_v2.h
./recipe-sysroot/usr/local/cuda-10.0/include/cublas_api.h
./recipe-sysroot/usr/local/cuda-10.0/include/cublasXt.h
./recipe-sysroot/usr/local/cuda-10.0/include/cublas.h
./recipe-sysroot/usr/local/cuda-10.0/lib64/libcublas.so.10.0.326
./recipe-sysroot/usr/local/cuda-10.0/lib64/libcublas.so
./recipe-sysroot/usr/local/cuda-10.0/lib64/libcublas_static.a
./recipe-sysroot/usr/local/cuda-10.0/lib64/libcublas.so.10.0
./recipe-sysroot/usr/local/cuda-10.0/lib64/stubs/libcublas.so
./recipe-sysroot-native/usr/local/cuda-10.0/targets/x86_64-linux/include/cublas_v2.h
./recipe-sysroot-native/usr/local/cuda-10.0/targets/x86_64-linux/include/cublas_api.h
./recipe-sysroot-native/usr/local/cuda-10.0/targets/x86_64-linux/include/cublasXt.h
./recipe-sysroot-native/usr/local/cuda-10.0/targets/x86_64-linux/include/cublas.h
./recipe-sysroot-native/usr/local/cuda-10.0/targets/x86_64-linux/lib/libcublas.so
./recipe-sysroot-native/usr/local/cuda-10.0/targets/x86_64-linux/lib/libcublas_static.a
./recipe-sysroot-native/usr/local/cuda-10.0/targets/x86_64-linux/lib/stubs/libcublas.so
./recipe-sysroot-native/installeddeps/cuda-cublas.complete
./recipe-sysroot-native/installeddeps/cuda-cublas
./recipe-sysroot-native/installeddeps/cuda-cublas.f387dceb5bc632678f48689fd8fb49f61fad92d3512172dd2369f6f4e0217ee2

To fix this I just edited cuda-samples file as follows:
```
diff --git a/recipes-devtools/cuda/cuda-samples_10.0.326-1.bb b/recipes-devtools/cuda/cuda-samples_10.0.326-1.bb
index 77353dd..8657ff3 100644
--- a/recipes-devtools/cuda/cuda-samples_10.0.326-1.bb
+++ b/recipes-devtools/cuda/cuda-samples_10.0.326-1.bb
@@ -56,7 +56,7 @@ def filtered_ldflags(d):
             newflags.append(flag)
     return ' '.join(newflags)

-LINKFLAGS = "-L${STAGING_DIR_HOST}${CUDA_PATH}/lib ${TOOLCHAIN_OPTIONS} ${@filtered_ldflags(d)} -lstdc++"
+LINKFLAGS = "-L${STAGING_DIR_HOST}${CUDA_PATH}/${baselib} ${TOOLCHAIN_OPTIONS} ${@filtered_ldflags(d)} -lstdc++"

 EXTRA_OEMAKE = ' \
     GENCODE_FLAGS="${CUDA_NVCC_ARCH_FLAGS}" SMS="${@extract_sm(d)}" OPENMP=yes \
```

paroque28, 2020-02-05:
My real worry is that cuda.bbclass has these paths hardcoded as well:
```
CUDA_NVCC_COMPAT_FLAGS ??= ""
CUDA_NVCC_PATH_FLAGS ??= "--include-path ${STAGING_DIR_HOST}/usr/local/cuda-${CUDA_VERSION}/include --library-path ${STAGING_DIR_HOST}/usr/local/cuda-${CUDA_VERSION}/lib"
CUDA_NVCC_EXTRA_FLAGS ??= ""
CUDA_NVCC_FLAGS ?= "${CUDA_NVCC_ARCH_FLAGS} ${CUDA_NVCC_COMPAT_FLAGS} ${CUDA_NVCC_PATH_FLAGS} ${CUDA_NVCC_EXTRA_FLAGS}"
```

Am I safe to use this cuda.bbclass?

madisongh, 2020-02-05:
You're right, the hard-coded `/lib` references should be changed to `${baselib}`.  As you can probably tell, multilib support hasn't been a high priority here.

paroque28, 2020-02-05:
Still without changing the cuda.bbclass it worked :)

```
root@jetson-nano-qspi-sd:~# /usr/bin/cuda-samples/deviceQuery
/usr/bin/cuda-samples/deviceQuery Starting...

 CUDA Device Query (Runtime API) version (CUDART static linking)

Detected 1 CUDA Capable device(s)

Device 0: "NVIDIA Tegra X1"
  CUDA Driver Version / Runtime Version          10.0 / 10.0
  CUDA Capability Major/Minor version number:    5.3
  Total amount of global memory:                 3956 MBytes (4148076544 bytes)
  ( 1) Multiprocessors, (128) CUDA Cores/MP:     128 CUDA Cores
  GPU Max Clock rate:                            922 MHz (0.92 GHz)
  Memory Clock rate:                             1600 Mhz
  Memory Bus Width:                              64-bit
  L2 Cache Size:                                 262144 bytes
  Maximum Texture Dimension Size (x,y,z)         1D=(65536), 2D=(65536, 65536), 3D=(4096, 4096, 4096)
  Maximum Layered 1D Texture Size, (num) layers  1D=(16384), 2048 layers
  Maximum Layered 2D Texture Size, (num) layers  2D=(16384, 16384), 2048 layers
  Total amount of constant memory:               65536 bytes
  Total amount of shared memory per block:       49152 bytes
  Total number of registers available per block: 32768
  Warp size:                                     32
  Maximum number of threads per multiprocessor:  2048
  Maximum number of threads per block:           1024
  Max dimension size of a thread block (x,y,z): (1024, 1024, 64)
  Max dimension size of a grid size    (x,y,z): (2147483647, 65535, 65535)
  Maximum memory pitch:                          2147483647 bytes
  Texture alignment:                             512 bytes
  Concurrent copy and kernel execution:          Yes with 1 copy engine(s)
  Run time limit on kernels:                     Yes
  Integrated GPU sharing Host Memory:            Yes
  Support host page-locked memory mapping:       Yes
  Alignment requirement for Surfaces:            Yes
  Device has ECC support:                        Disabled
  Device supports Unified Addressing (UVA):      Yes
  Device supports Compute Preemption:            No
  Supports Cooperative Kernel Launch:            No
  Supports MultiDevice Co-op Kernel Launch:      No
  Device PCI Domain ID / Bus ID / location ID:   0 / 0 / 0
  Compute Mode:
     < Default (multiple host threads can use ::cudaSetDevice() with device simultaneously) >

deviceQuery, CUDA Driver = CUDART, CUDA Driver Version = 10.0, CUDA Runtime Version = 10.0, NumDevs = 1
Result = PASS
```

closed 2020-03-04 by madisongh