OE4T/meta-tegra #250: nothing provides cuda-command-line-tools-libnvtoolsext needed by tensorrt-6.0.1+1-r0.armv8a_tegra210

paroque28, 2020-02-10:
--> Starting dependency resolution
--> Finished dependency resolution
Error:
 Problem: conflicting requests
  - nothing provides cuda-command-line-tools-libnvtoolsext needed by tensorrt-6.0.1+1-r0.armv8a_tegra210
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)

ERROR: Logfile of failure stored in: /yow-lpggp31/prodigu/lts19-jetson-nano/build/tmp-glibc/work/jetson_nano_qspi_sd-wrs-linux/wrlinux-image-std-sato/1.0-r1/temp/log.do_rootfs.18906

paroque28, 2020-02-10:
I see in "../layers/meta-tegra/recipes-devtools/gie/tensorrt_6.0.1-1.bb"
`the dependency very clearly 
```
RDEPENDS_${PN} += "libstdc++ cudnn cuda-cublas cuda-cudart cuda-command-line-tools-libnvtoolsext tegra-libraries libglvnd"
``

I am investigating

paroque28, 2020-02-10:
I found that the package is not being built:
```
prodrigu@yow-lpggp3:/yow-lpggp31/prodigu/lts19-jetson-nano/build/tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-command-line-tools/10.0.326-1-r0$ find . -name *.rpm
./deploy-rpms/armv8a_tegra210/cuda-command-line-tools-doc-10.0.326+1-r0.armv8a_tegra210.rpm
./deploy-rpms/armv8a_tegra210/cuda-command-line-tools-10.0.326+1-r0.armv8a_tegra210.rpm
./deploy-rpms/armv8a_tegra210/cuda-command-line-tools-libcupti-dev-10.0.326+1-r0.armv8a_tegra210.rpm
./deploy-rpms/armv8a_tegra210/cuda-command-line-tools-dev-10.0.326+1-r0.armv8a_tegra210.rpm
./deploy-rpms/armv8a_tegra210/cuda-command-line-tools-libcupti-10.0.326+1-r0.armv8a_tegra210.rpm
./deploy-rpms/armv8a_tegra210/cuda-command-line-tools-dbg-10.0.326+1-r0.armv8a_tegra210.rpm
```

madisongh, 2020-02-10:
Yeah, looks like another hard-coded `/lib` reference in the cuda-command-line-tools recipe, preventing the package from getting created.

madisongh, 2020-02-10:
I've also cherry-picked the fix into the zeus-l4t-r32.3.1 branch.

closed 2020-02-10 by madisongh