OE4T/meta-tegra #253: Create sdcard file for Jetson Nano (SKU 0000)

jbolduc01, 2020-02-12:
Hello,
I am having trouble to generate the sdcard file in order to test my yocto build.
I was able to generate my zeus image using bitbake core-image-minimal.
I was also able to include flashing tools to tegraimage.zip file. However, I am not able to flash my board even when putting device in recovery mode. I get the following message:
./doflash.sh 
ERR: chip does not identify as tegra210 ()

I am now trying to generate the image file for my sdcard directly but I have some issues with it.
udo ./make-sdcard flash.xml.in sd-blob.img -s 16G
About to make an SDcard image on sd-blob.img. OK? 
Please answer 'yes' or' no'.
Init...TBC...RP1...EBT...WB0...BPF...BPF-DTB...FX...TOS...DTB...LNX...EKS...BMP...RP4...APP...[OK]
Writing...ERR: cannot find file APPFILE for partition 1

Can you point me in the right direction?

Thanks,
Jérôme

jbolduc01, 2020-02-12:
I am sorry, everything works fine now. Seems like I used a bad jumper on REC pin.

paroque28, 2020-02-13:
@jbolduc01 As a tip, you can skip the doflash.sh script and leave your Jetson Nano unplugged if you run:

`$ make-sdcard -s 8G -b image-name signed/flash.xml`

closed 2020-03-04 by madisongh