OE4T/meta-tegra #254: Missing gstreamer plugins from nVidia

jbolduc01, 2020-02-13:
Hello,
I managed to build a Yocto image but there are some nVidia plugins that are missing. I am talking about nvcompositor and nvoverlaysink.

Here is my local.conf file:
#
# This file is your local configuration file and is where all local user settings
# are placed. The comments in this file give some guide to the options a new user
# to the system might want to change but pretty much any configuration option can
# be set in this file. More adventurous users can look at local.conf.extended
# which contains other examples of configuration which can be placed in this file
# but new users likely won't need any of them initially.
#
# Lines starting with the '#' character are commented out and in some cases the
# default values are provided as comments to show people example syntax. Enabling
# the option is a question of removing the # character and making any change to the
# variable as required.

#
# Machine Selection
#
# You need to select a specific machine to target the build with. There are a selection
# of emulated machines available which can boot and run in the QEMU emulator:
#
#MACHINE ?= "qemuarm"
#MACHINE ?= "qemuarm64"
#MACHINE ?= "qemumips"
#MACHINE ?= "qemumips64"
#MACHINE ?= "qemuppc"
#MACHINE ?= "qemux86"
#MACHINE ?= "qemux86-64"
#
# There are also the following hardware board target machines included for 
# demonstration purposes:
#
#MACHINE ?= "beaglebone-yocto"
#MACHINE ?= "genericx86"
#MACHINE ?= "genericx86-64"
#MACHINE ?= "mpc8315e-rdb"
#MACHINE ?= "edgerouter"
#
# This sets the default machine to be qemux86-64 if no other machine is selected:
MACHINE ??= "qemux86-64"

#
# Where to place downloads
#
# During a first build the system will download many different source code tarballs
# from various upstream projects. This can take a while, particularly if your network
# connection is slow. These are all stored in DL_DIR. When wiping and rebuilding you
# can preserve this directory to speed up this part of subsequent builds. This directory
# is safe to share between multiple builds on the same machine too.
#
# The default is a downloads directory under TOPDIR which is the build directory.
#
#DL_DIR ?= "${TOPDIR}/downloads"

#
# Where to place shared-state files
#
# BitBake has the capability to accelerate builds based on previously built output.
# This is done using "shared state" files which can be thought of as cache objects
# and this option determines where those files are placed.
#
# You can wipe out TMPDIR leaving this directory intact and the build would regenerate
# from these files if no changes were made to the configuration. If changes were made
# to the configuration, only shared state files where the state was still valid would
# be used (done using checksums).
#
# The default is a sstate-cache directory under TOPDIR.
#
[... 223 lines cut ...]
				usbutils \
				pciutils \
				ethtool \
				gstreamer1.0 \ 
				gstreamer1.0-libav \
				gstreamer1.0-meta-base \
				gstreamer1.0-plugins-bad \
				gstreamer1.0-plugins-base \
				gstreamer1.0-plugins-good \
				gstreamer1.0-plugins-nveglgles \
				gstreamer1.0-plugins-nvvideo4linux2 \
				gstreamer1.0-plugins-nvvideosinks \
				gstreamer1.0-plugins-tegra \
				gstreamer1.0-plugins-ugly \
				gstreamer1.0-python \
				gstreamer1.0-rtsp-server \			
				gstreamer1.0-vaapi \
				tegra-argus-daemon \
				nv-tegra-release \
				htop"

Do you have any idea why they are not part of the image?

Thanks,
Jérôme

madisongh, 2020-02-13:
The `nvcompositor` element is packaged separately into `gstreamer1.0-plugins-nvcompositor`, so you'll need to add that.   It's packaged separately because of some additional runtime dependencies.

The `nvoverlaysink` element should be in `gstreamer1.0-omx-tegra`, which should be getting included via settings in the machine config.  If that package is installed but the element isn't showing up, try using `gst-inspect-1.0 -b` to see if it's been blacklisted for some reason.

jbolduc01, 2020-02-13:
Thank you for your prompt reply!
How to I get this gstreamer1.0-plugins-tegra-nvcompositor recipe in order to build it?
Regards,
Jérôme

madisongh, 2020-02-13:
The [gstreamer1.0-plugins-tegra](https://github.com/madisongh/meta-tegra/blob/zeus-l4t-r32.3.1/recipes-bsp/tegra-binaries/gstreamer1.0-plugins-tegra_32.3.1.bb) that is already present does this - you just need to reference the extra package name to CORE_IMAGE_EXTRA_INSTALL, and make sure that the [additional gstreamer packages it depends on](https://github.com/madisongh/meta-tegra/blob/3fe0fe7d7223bd2e90cf1b78a08ce9603cb777b9/recipes-bsp/tegra-binaries/gstreamer1.0-plugins-tegra_32.3.1.bb#L61) can also be built.

jbolduc01, 2020-02-13:
Thanks for the info.
However, when I try to include the additional gstreamers to CORE_IMAGE_EXTRA_INSTALL, here is what I get: 
```
ERROR: core-image-minimal-1.0-r0 do_rootfs: Could not invoke dnf. Command '/home/jerome/poky-nano/build/tmp/work/jetson_nano_qspi_sd-poky-linux/core-image-minimal/1.0-r0/recipe-sysroot-native/usr/bin/dnf -v --rpmverbosity=info -y -c /home/jerome/poky-nano/build/tmp/work/jetson_nano_qspi_sd-poky-linux/core-image-minimal/1.0-r0/rootfs/etc/dnf/dnf.conf --setopt=reposdir=/home/jerome/poky-nano/build/tmp/work/jetson_nano_qspi_sd-poky-linux/core-image-minimal/1.0-r0/rootfs/etc/yum.repos.d --installroot=/home/jerome/poky-nano/build/tmp/work/jetson_nano_qspi_sd-poky-linux/core-image-minimal/1.0-r0/rootfs --setopt=logdir=/home/jerome/poky-nano/build/tmp/work/jetson_nano_qspi_sd-poky-linux/core-image-minimal/1.0-r0/temp --repofrompath=oe-repo,/home/jerome/poky-nano/build/tmp/work/jetson_nano_qspi_sd-poky-linux/core-image-minimal/1.0-r0/oe-rootfs-repo --nogpgcheck install ethtool gstreamer1.0 gstreamer1.0-libav gstreamer1.0-meta-base gstreamer1.0-omx-tegra gstreamer1.0-plugins-bad gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-nveglgles gstreamer1.0-plugins-nvvideo4linux2 gstreamer1.0-plugins-nvvideosinks gstreamer1.0-plugins-tegra gstreamer1.0-plugins-ugly gstreamer1.0-rtsp-server gstreamer1.0-vaapi htop libdrm2 libgstbadbase-1.0 libgstbadvideo-1.0 libgstvideo-1.0-0 nv-tegra-release openssh packagegroup-core-boot pciutils run-postinsts tegra-argus-daemon tegra-libraries usbutils v4l-utils' returned 1:
DNF version: 4.2.2
cachedir: /home/jerome/poky-nano/build/tmp/work/jetson_nano_qspi_sd-poky-linux/core-image-minimal/1.0-r0/rootfs/var/cache/dnf
Added oe-repo repo from /home/jerome/poky-nano/build/tmp/work/jetson_nano_qspi_sd-poky-linux/core-image-minimal/1.0-r0/oe-rootfs-repo
repo: using cache for: oe-repo
not found other for: 
not found modules for: 
not found deltainfo for: 
not found updateinfo for: 
oe-repo: using metadata from Thu 13 Feb 2020 07:30:25 PM UTC.
Last metadata expiration check: 0:00:01 ago on Thu 13 Feb 2020 07:30:25 PM UTC.
No module defaults found
No match for argument: libgstbadbase-1.0
No match for argument: libgstbadvideo-1.0
Error: Unable to find a match

ERROR: Logfile of failure stored in: /home/jerome/poky-nano/build/tmp/work/jetson_nano_qspi_sd-poky-linux/core-image-minimal/1.0-r0/temp/log.do_rootfs.3734
ERROR: Task (/home/jerome/poky-nano/meta/recipes-core/images/core-image-minimal.bb:do_rootfs) failed with exit code '1'
```
How can I build these two packages? I am sorry, I am quite new in building in Yocto.

Regards,
Jérôme

madisongh, 2020-02-13:
Hmm.  This could be a problem with using the nvcompositor plugin at all.  It was pre-built against an older version of gstreamer and was linked against the `libgstbadvideo-1.0` library.   That library existed in the version of gstreamer (1.14.x) that comes with the version of Ubuntu that NVIDIA puts into L4T/JetPack, but it looks like it was dropped from more recent versions of gstreamer - the functionality was migrated into the `base` or `good` package.  OE-Core zeus has gstreamer 1.16.x.

So if you really need the nvcompositor plugin, it could take some work.  It might be possible to fake the missing library (and remove the RDEPENDS on the nonexistent packages from the recipe), or it might mean having to downgrade gstreamer to 1.14.

jbolduc01, 2020-02-14:
I think you are right, seems like the `libgstcompositor.so` file is built inside my Yocto image. However, it relies on the `libgstbadvideo-1.0` library, which is not built.
I think I will need to downgrade to gstreamer 1.14.x in order to get rid of this. I really need the `nvcompositor` plugins since it manages DMA accesses.
Do you have any tips on how to downgrade to gstreamer 1.14.x? I think the latest Jetpack/L4T uses gstreamer 1.14.5.
Thanks for your help so far!
Regards,
Jérôme

madisongh, 2020-02-15:
You should be able to copy over all the gstreamer recipes from the openembedded-core or poky `warrior` branch into your own layer.  You'll need to set PREFERRED_VERSION_* for the gstreamer packages to have the build choose the older version.

jbolduc01, 2020-02-18:
I was able to include nvcompositor in my image when using gstreamer 1.14.4.
Thanks for your help!
Jérôme

closed 2020-02-18 by jbolduc01