OE4T/meta-tegra #255: Add-jetson-inference

elPrac, 2020-02-14:
Hi!

I tried to add [jetson-inference](https://github.com/dusty-nv/jetson-inference) in order to be able to use/install its python modules in my image, which will be running on a Jetson-Nano,  but i failed several times and my poor knowledge about Yocto and none experience with Cmake are making things worse...

So far i tried a very basic recipe
```bash
SUMMARY = "Deep-learning inference networks and deep vision primitives"
HOMEPAGE = "https://github.com/dusty-nv/jetson-inference"
LICENSE = "MIT"
DEPENDS = "glib-2.0 glib-2.0-native python3 python3-numpy"                                             
LIC_FILES_CHKSUM = "file://LICENSE.md;md5=eafbc60fe8e4e62ce0b37714974a87fc"
SRCREV = "99ecf1f41b47b6d4f8fb848963f4c6517cfa82c3"
SRC_URI = "gitsm://github.com/dusty-nv/jetson-inference"
S = "${WORKDIR}/git"
inherit cuda cmake pkgconfig
```
SRCEV is pointing to L4T-R32.2 [branch](https://github.com/dusty-nv/jetson-inference/tree/L4T-R32.2) and i'm using meta-tegra zeus branch.
I made the following changes to CMakeLists.txt
```bash
+++ tmp/work/armv8a_tegra210-poky-linux/jetson-inference/32.2.2-r0/git/CMakeLists.txt
@ Line: 39
 # if this is the first time running cmake, perform pre-build dependency install script (or if the user manually triggers re-building the dependencies)
 if( ${BUILD_DEPS} )
 #   message("-- Launching pre-build dependency installer script...")
 
 #   execute_process(COMMAND sh ../CMakePreBuild.sh
 #               WORKING_DIRECTORY ${PROJECT_BINARY_DIR}
 #               RESULT_VARIABLE PREBUILD_SCRIPT_RESULT)                                                                                               
```
[CMakePreBuild.sh](https://github.com/dusty-nv/jetson-inference/blob/master/CMakePreBuild.sh) install some packages that we don't need to be done by this script.
When I start the do_configure () task, problems with pthread.h appear
```
Log data follows:
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is GNU 7.3.0
| -- The CXX compiler identification is GNU 7.3.0
| -- Check for working C compiler: tmp/work/armv8a_tegra210-poky-linux/jetson-inference/32.2.2-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc
| -- Check for working C compiler: tmp/work/armv8a_tegra210-poky-linux/jetson-inference/32.2.2-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Check for working CXX compiler: tmp/work/armv8a_tegra210-poky-linux/jetson-inference/32.2.2-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++
| -- Check for working CXX compiler: tmp/work/armv8a_tegra210-poky-linux/jetson-inference/32.2.2-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++ -- works
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- using patched FindCUDA.cmake
| -- Looking for pthread.h
| -- Looking for pthread.h - not found
| CMake Error at tmp/work/armv8a_tegra210-poky-linux/jetson-inference/32.2.2-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
|   Could NOT find Threads (missing: Threads_FOUND)
| Call Stack (most recent call first):
|   tmp/work/armv8a_tegra210-poky-linux/jetson-inference/32.2.2-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
|   tmp/work/armv8a_tegra210-poky-linux/jetson-inference/32.2.2-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/FindThreads.cmake:220 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
|   utils/cuda/FindCUDA.cmake:869 (find_package)
|   CMakeLists.txt:52 (find_package)
| -- Configuring incomplete, errors occurred!
```
I'm not quite sure where Cmake is looking for the pthread but if i search for it in my ${WORKDIR} this is what i can see:
```
./recipe-sysroot/usr/lib/libpthread.so
./recipe-sysroot/usr/lib/libpthread.a
./recipe-sysroot/usr/include/pthread.h
./recipe-sysroot/usr/include/bits/pthreadtypes-64.h
./recipe-sysroot/usr/include/bits/pthreadtypes-arch.h
./recipe-sysroot/usr/include/bits/pthreadtypes-arch-64.h
./recipe-sysroot/usr/include/bits/pthreadtypes.h
```
I'm not convinced if i need something more in my DEPENDS list or maybe something to inherit, or perhaps, an additional value to EXTRA_OECMAKE.  I already google pthread and Cmake related problems, and solutions suggests that a missing -pthread flag in ${CMAKE_C_FLAGS} could be the problem, but, i don't think it would apply here because i can see in tmp/work/armv8a_tegra210-poky-linux/jetson-inference/32.2.2-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/FindThreads.cmake the following lines
```
PROPERTY INTERFACE_COMPILE_OPTIONS  "$<$<COMPILE_LANGUAGE:CUDA>:SHELL:-Xcompiler -pthread>"
                                   "$<$<NOT:$<COMPILE_LANGUAGE:CUDA>>:-pthread>")

```
However, I'm not really sure of the above because the error seems to happen before these compile options are set.

Also i don't think meta-tegra/recipes-devtools/cmake/cmake/findcuda-dont-reset-cflags.patch is causing problems because i already disabled it and nothing changes...

By the way I'm using the following in my host:
Distro: Ubuntu 18.04
SDK Manager: 1.0.0
JetPack 4.2.2
In my local.conf
GCCVERSION = "7.%"
require contrib/conf/include/gcc-compat.conf
DEBUG_PREFIX_MAP_remove = "-fmacro-prefix-map=${WORKDIR}=/usr/src/debug/${PN}/${EXTENDPE}${PV}-${PR}"
CUDA_VERSION="10.0"

NVIDIA_DEVNET_MIRROR = "file://home/me/Downloads/nvidia/sdkm_downloads"

Hope someone could give me some advise about the issue also, if it works,  i would be glad to make the pull request :) 

By the way, thank you so much for your work.
BR,

madisongh, 2020-02-14:
Looks like it's using its own copy of FindCUDA.cmake.  Try removing this line:
```
set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_CURRENT_SOURCE_DIR}/utils/cuda" )
```
from the CMakeLists.txt file. It should work better using the FindCUDA.cmake in the recipe sysroot. 

From a brief glance, it looks like there will be many more hurdles to get over.  Good luck!

elPrac, 2020-02-21:
Hi @madisongh thank you so much for your reply!

After doing your suggestion it finally found pthread
```
| -- Finished installing dependencies
| -- Looking for pthread.h
| -- Looking for pthread.h - found
| -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
| -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
| -- Looking for pthread_create in pthreads
| -- Looking for pthread_create in pthreads - not found
| -- Looking for pthread_create in pthread
| -- Looking for pthread_create in pthread - found
| -- Found Threads: TRUE

```
 _It should work better using the FindCUDA.cmake in the recipe sysroot._
That was my understanding(and my intention) but i'm glad it is doing now(thanks to your comment)
or at least that is what i can see in **tmp/work/armv8a_tegra210-poky-linux/jetson-inference/32.2.2-r0/build/CMakeFiles/CMakeError.log**

```
tmp/work/armv8a_tegra210-poky-linux/jetson-inference/32.2.2-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  
>>--sysroot=tmp/work/armv8a_tegra210-poky-linux/jetson-inference/32.2.2-r0/recipe-sysroot<<
```

 _it looks like there will be many more hurdles to get over._

Yes you are right, there are a lot of things that they do with scripts that should be changed to recipes, so the first thing(i think) is to download and install dependencies, so far I've been trying to install PyTorch but i failed several times. I've tried the [following](https://stackoverflow.com/questions/56563920/yocto-bitbake-recipes-for-nvidia-jetson-nano-for-python-whl-files-not-on-pypi) but it didn't work form me... but based on [this](https://stackoverflow.com/questions/48660051/yocto-recipe-python-whl-package) i came up with the following recipe.

```bash
FILESEXTRAPATHS_prepend := "${THISDIR}:"
SUMMARY = "PyTorch Python package"
DESCRIPTION = "provides two high-level features: Tensor computation & Deep neural networks"
LICENSE = "BSD-3-Clause"
PV = "1.1.0"
PN = "python3-torch"

# This is the default source of jetson-inference
SRC_URI = "https://nvidia.box.com/shared/static/j2dn48btaxosqp0zremqqm8pjelriyvs.whl;downloadfilename=torch-1.1.0-cp36-cp36m-linux_aarch64.zip;subdir=${BP}
                                                                                                                                                           
SRC_URI[md5sum] = "9ec85425a64ca266abbfdeddbe92fb18"
inherit python3-dir
LIC_FILES_CHKSUM = "file://${S}/torch-1.1.0.dist-info/LICENSE;md5=acf4d595f99e159bf31797aa872aef57"
do_unpack[depends] += "unzip-native:do_populate_sysroot"
PROVIDES += "python3-torch"
RDEPENDS_python3-torch += "libcurand.so.10.0 libcudart.so.10.0"
FILES_${PN} += "\
    ${libdir}/${PYTHON_DIR}/site-packages/* \
"
do_install() {
    TORCH_WHL="torch-1.1.0.dist-info torch caffe2"
    for f in ${TORCH_WHL}; do
        install -d ${D}${libdir}/${PYTHON_DIR}/site-packages/${f}
        cp -R --no-dereference --preserve=mode,links -v "${S}/${f}/" ${D}${libdir}/${PYTHON_DIR}/site-packages/${f}
    done
}
                                                                                                                                                           
```
 As you can see i'm not doing quite well in **RDEPENDS** i'm getting the following error:

```
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'libcudart.so.10.0' (but /mnt/development/emdep/environment/sources/meta-emdep/recipes-emdep/python3-torch/python3-torch_1.1.0.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'libcudart.so.10.0' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['libcudart.so.10.0']
ERROR: Required build target 'python3-torch' has no buildable providers.
Missing or unbuildable dependency chain was: ['python3-torch', 'libcudart.so.10.0']

```
where i can get these dependencies? can i get them from meta-tegra? i-'-m no quite sure how i should add them :/

Again, thank you for your time and advice. :)
If you have any suggestion or documentation that i could look at it will be very helpful!

BR,

madisongh, 2020-02-21:
>`RDEPENDS_python3-torch += "libcurand.so.10.0 libcudart.so.10.0"`

Instead of the libraries, you need to specify the packages that provide those libraries.  Try using `cuda-curand cuda-cudart` instead.

madisongh, 2020-11-08:
#440 looks related to this.

madisongh, 2020-12-11:
Will close this now in favor of the more recent #440

closed 2020-12-11 by madisongh