OE4T/meta-tegra #257: Auvidea Carrier Support

Austriker, 2020-02-17:
Hi,

I am looking at supporting a new carrier for the jetson TX2 with an Auvidea Carrier Board.
They provide the firmware but I am struggling to see how to integrate it. [The page for the firmware update](https://auvidea.eu/firmware/)

Below the readme of the firmware update. 

My question is should I reproduce that with a recipe or is there a better way ?

Best Regards

```
How to install the auvidea kernel the nvidia sdkmanager
####################################################################
This guid is writen for the J120 and the TX2 

1. prepare the sdk
ONLY NEEDED IF THE IF NO TX2 WAS FLASHED BEFORE
-start the sdkmanager
-select the Jetson TX2(P3310) as target Hardware
-start the installation until the sdk asks you for either use automatic or manual setup
 -> at this point you can choose to "skip" the rest of the installation and contine with step 2 of this instructions

2. copy the contents of the auvidea packet to the nvida_sdk folder
-> cp -r ~/J120_kernel/* /home/user/nvidia/nvidia_sdk/Jetpack_4.2_Linux_P3310/Linux_for_Tegra/

3. switch into the Linux_for_Tegra folder
-> /home/user/nvidia/nvidia_sdk/Jetpack_4.2_Linux_P3310/Linux_for_Tegra/

4. apply the changes
-> sudo ./apply_binaries.sh
 
5. start the sdkmanger and follow the normal installation process
```

madisongh, 2020-02-17:
From a quick glance and the downloads page and the instructions, they're just overlaying some files on top of the L4T/JetPack contents.  What you need to do is to dig into the source code of those changes (which it sounds like they provide).  It's likely there are some device tree modifications in the kernel source, at the very least.

For some hints, have a look at #67 - some folks did some work on supporting the ConnectTech carrier boards, and I would expect this to be similar.

Austriker, 2020-02-18:
Yes I have matched what they did in their PR and with the source provided by Auvidea I was able to have it running.

closed 2020-02-18 by Austriker

AtleStor, 2024-12-03:
Hi @madisongh or others - A bit on the side here: For a legacy system, I am trying to have the ConnectTech/Orbitty carrier board supported by a Yocto build. Did any1 have success with having BSPs successfully running a CTI boards that you are aware of?
BR, Atle

dwalkes, 2024-12-03:
You might want to [mine the issues list](https://github.com/search?q=repo%3AOE4T%2Fmeta-tegra+ConnectTech&type=issues) where it's been mentioned in the past

AtleStor, 2025-01-27:
> Hey, I have the same need with CT, maybe we can work together to put an
> unofficial repo for the layer?
> I've already asked support if they are willing to officially maintain it
> open-source.

Hi there.
Thanks, good idea.
I will look into this when I get back to office probably in the end of the 
week/start next.
Thanks again.
Br Atle

--194a839e23f25082822ac716ce
Content-Type: text/html; charset="UTF-8"
Content-Transfer-Encoding: quoted-printable

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN" "http://www.=
w3.org/TR/html4/loose.dtd"><html>
<head>
<meta http-equiv=3D"Content-Type" content=3D"text/html; charset=3Dutf-8"></=
head>
<body>
<div dir=3D"auto">
<div dir=3D"auto">Hi there.</div><div dir=3D"auto">Thanks, good idea.</div>=
<div dir=3D"auto">I will look into this when I get back to office probably =
in the end of the week/start next.</div><div dir=3D"auto">Thanks again.</di=
v><div dir=3D"auto">Br Atle </div><div dir=3D"auto"><br></div><div dir=
=3D"auto"><br></div><div dir=3D"auto"><br></div>
<div id=3D"aqm-original" style=3D"color: black;">

<div class=3D"aqm-original-body">
<div style=3D"color: black;">
<p style=3D"color: black; font-size: 10pt; font-family: sans-serif; margin:=
 8pt 0;">On January 27, 2025 15:30:39 Razvan Grigore ***@***.***=
ub.com&gt; wrote:</p>
<blockquote type=3D"cite" class=3D"gmail_quote" style=3D"margin: 0 0 0 0.75=
ex; border-left: 1px solid #808080; padding-left: 0.75ex;">
<p></p>
<p dir=3D"auto">Hey, I have the same need with CT, maybe we can work togeth=
er to put an unofficial repo for the layer?</p>
<p dir=3D"auto">I've already asked support if they are willing to officiall=
y maintain it open-source.</p>

<p style=3D"font-size:small;-webkit-text-size-adjust:none;color:#666;">=E2=
=80=94<br>
meta-tegra/issues/257#issuecomment-2615916142" originalsrc=3D"https://githu=
b.com/OE4T/meta-tegra/issues/257#issuecomment-2615916142" shash=3D"c6vwm23A=
fCnFxrfwmQSXmItVh1G5QGWAbPSYcSpNhI6MWiN2lnUHtQ336esiW7HfNQDAowB/iv+b2g+qMFr=
3LIX7JVbtHOL0DTdD2UdYYGqMsT+Nz7Z6XS3vn/5sYSEzoJW4MIwa84Pog03bB3371PYkwB05Bz=
J1wPAhZT4nD9A=3D">view it on GitHub</a>, or <a href=3D"https://github.com/n=
otifications/unsubscribe-auth/BNNE4FOBYR2YPRGDV2EB4PT2MY7IZAVCNFSM6AAAAABS5=
JM3YWVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDMMJVHEYTMMJUGI" originalsrc=
PT2MY7IZAVCNFSM6AAAAABS5JM3YWVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDMMJV=
HEYTMMJUGI" shash=3D"yqMLXyfwmTIq4NunTQPAB4dt991Z/anEDN2riWOTHm56FAP17LRGfX=
Wz1zBAbWs1v3blHeDbD5aGPQjVS2M7GWxiItfmgvW2AJ4l0lQ0oxY5Rx1WbU9mDZGEKHItr0wR8=
PnPa9pQe7OjjcyE4BUe5EJudA67QnjXkAizZTtmtQ4=3D">unsubscribe</a>.<br>You are =
receiving this because you commented.<img src=3D"https://github.com/notific=
ations/beacon/BNNE4FJ5YQHCQ3QRS3Z5WXL2MY7IZA5CNFSM6AAAAABS5JM3YWWGG33NNVSW4=
5C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTU35O3G4.gif" height=3D"1" =
width=3D"1" alt=3D""><span style=3D"color: transparent; font-size: 0; displ=
ay: none; visibility: hidden; overflow: hidden; opacity: 0; width: 0; heigh=
T/meta-tegra/issues/257/2615916142</span><span>@</span><span>github</span><=
span>.</span><span>com&gt;</span></span></p>

</blockquote>
</div>
</div>

</div><div dir=3D"auto"><br></div>
</div></body>
</html>

--194a839e23f25082822ac716ce--