OE4T/meta-tegra #259: Warrior Static IP Network

Austriker, 2020-02-18:
Hi,

I have been looking on how to fix the network interface IP. But the modifications doesn't stick. I have looked at this #225 should I overwrite the bbappend `systemd-conf_%.bbappend`. Also my second port eth1 is in IPV6 surprisingly so the wired.network in the meta-tegra seems not to apply to the second interface.

What is the right way to configure the network on the warrior branch

eth0.network
```
[Match]
Name=eth0

[Network]
DNS=8.8.8.8
Address=192.168.0.76
Gateway=192.168.0.1
```

eth1.network
```
[Match]
Name=eth0

[Network]
DNS=8.8.8.8
Address=192.168.1.76
Gateway=192.168.1.1
```

and the recipe : systemd_%.bbappend
```
PACKAGECONFIG_append = " networkd resolved"
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"

SRC_URI += " \
    file://eth0.network \
    file://eth1.network \
"
FILES_${PN} += " \
    ${sysconfdir}/systemd/network/eth0.network \
    ${sysconfdir}/systemd/network/eth1.network \
"
do_install_append() {
  if ${@bb.utils.contains('PACKAGECONFIG','networkd','true','false',d)}; then
        install -d ${D}${sysconfdir}/systemd/network
        install -m 0755 ${WORKDIR}/eth0.network ${D}${sysconfdir}/systemd/network
        install -m 0755 ${WORKDIR}/eth1.network ${D}${sysconfdir}/systemd/network
  fi
}
```

madisongh, 2020-02-18:
The `[Match]` section in your `eth1.network` file doesn't look right.  Don't you want `Name=eth1` there?

Austriker, 2020-02-18:
@madisongh Sorry It was a typo error.
When I do this : 
` ifconfig eth1 192.168.1.2 netmask 255.255.255.0 up`
The interface changes.

madisongh, 2020-02-19:
It could be that the warrior build is using `init-ifupdown` to manage the network links, in which case you'll need to provide a modified copy of `/etc/network/interfaces`  to configure the interfaces with their static IPs.  Either that, or modify the build to remove `init-ifupdown` so `systemd-networkd` takes control of configuring the interfaces (although that might require some other changes, I'm not 100% positive that will work).

Austriker, 2020-02-19:
Thank you for your feedback. I will try that after I have fixed the BSP for the ethernet ports.
Keep you posted/

Austriker, 2020-02-20:
It seems that it's actually systemd-networkd that is handling networking.
The file /etc/network/interfaces doesn't exist. I created it but it doesn't set the IP

When I restart networking I get this : 
```
Feb 20 18:04:55 j140-tx2 systemd-networkd[5049]: eth1: Gained IPv6LL
Feb 20 18:04:55 j140-tx2 systemd-networkd[5049]: enp2s0: Gained IPv6LL
Feb 20 18:04:55 j140-tx2 systemd-networkd[5049]: Enumeration completed
Feb 20 18:04:55 j140-tx2 systemd[1]: Started Network Service.
Feb 20 18:04:55 j140-tx2 systemd-networkd[5049]: eth1: DHCPv4 address 192.168.0.47/24 via 192.168.0.1
Feb 20 18:05:07 j140-tx2 systemd-networkd[5049]: eth1: Configured
Feb 20 18:07:02 j140-tx2 systemd[1]: Stopping Network Service...
Feb 20 18:07:02 j140-tx2 systemd[1]: systemd-networkd.service: Succeeded.
Feb 20 18:07:02 j140-tx2 systemd[1]: Stopped Network Service.
Feb 20 18:07:02 j140-tx2 systemd[1]: Starting Network Service...
Feb 20 18:07:02 j140-tx2 systemd-networkd[5134]: eth1: Gained IPv6LL
Feb 20 18:07:02 j140-tx2 systemd-networkd[5134]: enp2s0: Gained IPv6LL
Feb 20 18:07:02 j140-tx2 systemd-networkd[5134]: Enumeration completed
Feb 20 18:07:02 j140-tx2 systemd-networkd[5134]: eth1: DHCPv4 address 192.168.0.47/24 via 192.168.0.1
Feb 20 18:07:02 j140-tx2 systemd[1]: Started Network Service.
```
And before I fixed the file : 
```
-- Logs begin at Thu 2020-02-20 17:55:06 UTC, end at Thu 2020-02-20 18:05:25 UTC. --
Feb 20 17:55:06 j140-tx2 systemd[1]: Starting Network Service...
Feb 20 17:55:06 j140-tx2 systemd-networkd[3165]: [[0;1;31m[[0;1;39m[[0;1;31m/etc/systemd/network/enp2s0.network:4: An address '192.168.1.2' is specified without prefix length. The behavior of parsing addr
esses without prefix length will be changed in the future release. Please specify prefix length explicitly.[[0m
Feb 20 17:55:06 j140-tx2 systemd-networkd[3165]: Enumeration completed
Feb 20 17:55:06 j140-tx2 systemd[1]: Started Network Service.
Feb 20 17:55:06 j140-tx2 systemd-networkd[3165]: eth0: Interface name change detected, eth0 has been renamed to enp2s0.
-- Logs begin at Thu 2020-02-20 17:55:06 UTC, end at Thu 2020-02-20 18:07:03 UTC. --
Feb 20 17:55:06 j140-tx2 systemd[1]: Starting Network Service...
Feb 20 17:55:06 j140-tx2 systemd-networkd[3165]: [[0;1;31m[[0;1;39m[[0;1;31m/etc/systemd/network/enp2s0.network:4: An address '192.168.1.2' is specified without prefix length. The behavior of parsing addr
esses without prefix length will be changed in the future release. Please specify prefix length explicitly.[[0m
Feb 20 17:55:06 j140-tx2 systemd-networkd[3165]: Enumeration completed
```
So it seems it's reading it properly. But it doesn't apply it.

Austriker, 2020-02-20:
I have tested it on the standard Jetpack on a Xavier. To be able to make the `.network` file work I had to do this.
```
sudo systemctl stop NetworkManager.service 
sudo systemctl stop NetworkManager-wait-online.service 
sudo systemctl stop NetworkManager-dispatcher.service 
sudo systemctl stop network-manager.service 
```
But these services don't exist in the image.

madisongh, 2020-02-21:
Yes, thankfully NetworkManager is not used by default here.  Try naming your `.network` files such that they will be processed earlier than `80-wired.network` (which enables DHCP).  Per [the systemd documentation](https://www.freedesktop.org/software/systemd/man/systemd.network.html#%5BMatch%5D%20Section%20Options), whichever `.network` file matches first wins, and a numerically prefixed file name will precede an alphabetic filename lexicographically.

Austriker, 2020-02-21:
Thank you very much that was the solution.

closed 2020-02-21 by Austriker