OE4T/meta-tegra #260: Held broken packages packagegroup-base-extended depends packagegroup-base

Austriker, 2020-02-19:
Hi,

My package management is deb files.
Meta Tegra branch : warrior

I added the packagegroup-base in the image install list but I get another error with missing dependencies.
I rebuilded everything from scratch to be sure.

I am getting this error : 
```
ERROR: custom-image-1.0-r0 do_rootfs: Unable to install packages. Command '/home/ubuntu/tegra/build/tmp/work/j140_tx2-poky-linux/custom-image/1.0-r0/recipe-sysroot-native/usr/bin/apt-get  install --force-yes --allow-unauthenticated apt base-passwd cmake coreutils cuda-core cuda-samples libcudnn7 dpkg libc6 libc6-utils ldd libgomp1 libusb-1.0-0 lmsensors-sensors mender mesa-demos net-tools opencv packagegroup-base-extended packagegroup-core-boot packagegroup-core-ssh-openssh packagegroup-core-x11 procps qtbase qtbase-plugins qtdeclarative run-postinsts shadow sudo tegra-nvpmodel tegra-tools tensorrt ttf-dejavu-sans udev' returned 100:
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 packagegroup-base-extended : Depends: packagegroup-base but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

ERROR: custom-image-1.0-r0 do_rootfs: 
ERROR: custom-image-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/ubuntu/tegra/build/tmp/work/j140_tx2-poky-linux/custom-image/1.0-r0/temp/log.do_rootfs.17702
ERROR: Task (/home/ubuntu/tegra/build/../sources/meta-tegra-test/recipes-tegra-test/images/custom-image.bb:do_rootfs) failed with exit code '1'
```

my image is like this : 
```
inherit core-image extrausers

IMAGE_FEATURES += "package-management ssh-server-openssh x11"
IMAGE_FEATURES_remove = "ssh-server-dropbear"

ROOT_PASSWORD = "hello"

EXTRA_USERS_PARAMS_append = "\
                        usermod -p  '${ROOT_PASSWORD}' root;\
" 

IMAGE_INSTALL_append = "\
	glibc \
	glibc-utils \
	ldd \
	libgomp \
	ttf-dejavu-sans \
	ueye \
	qtbase \
	mesa-demos \
	qtbase-plugins \
	qtdeclarative \
	opencv \
	cuda-core \
	cuda-samples \
	cudnn \
	tensorrt \
	tegra-tools \
	tegra-nvpmodel \
	net-tools \
	coreutils \
	procps \
	libusb1 \
	usbutils \
	udev \
	sudo \
        packagegroup-core-buildessential \
"
```

madisongh, 2020-02-19:
Try adding this line to your local.conf:
```
APT_ARGS = "-oDebug::pkgProblemResolver=true -oDebug::pkgDPkgPM=true"
```
to coax apt into providing more info.  The default output from apt in situations like this is not exactly helpful in determing what the underlying problem is.

Austriker, 2020-02-19:
It didn't work in my local.conf but in my image recipe it did : 
```
ERROR: image-custom-1.0-r0 do_rootfs: Unable to install packages. Command '/home/ubuntu/tegra/build/tmp/work/j140_tx2-poky-linux/image-custom/1.0-r0/recipe-sysroot-native/usr/bin/apt-get -oDebug::pkgProblemResolver=true -oDebug::pkgDPkgPM=true install --force-yes --allow-unauthenticated apt base-passwd cmake coreutils cuda-core cuda-samples libcudnn7 dpkg libc6 libc6-utils ldd libgomp1 libusb-1.0-0 lmsensors-sensors mender mesa-demos net-tools opencv packagegroup-base-extended packagegroup-core-boot packagegroup-core-buildessential packagegroup-core-ssh-openssh packagegroup-core-x11 procps qtbase qtbase-plugins qtdeclarative run-postinsts shadow sudo tegra-nvpmodel tegra-tools tensorrt ttf-dejavu-sans udev usbutils ' returned 100:
Reading package lists...
Building dependency tree...
Reading state information...
Starting pkgProblemResolver with broken count: 1
Starting 2 pkgProblemResolver with broken count: 1
Investigating (0) packagegroup-base [ arm64 ] < none -> 1.0-r83 > ( base )
Broken packagegroup-base:arm64 Depends on packagegroup-machine-base [ arm64 ] < none -> 1.0-r83 > ( base )
  Considering packagegroup-machine-base:arm64 1 as a solution to packagegroup-base:arm64 0
  Holding Back packagegroup-base:arm64 rather than change packagegroup-machine-base:arm64
Investigating (1) packagegroup-base-extended [ arm64 ] < none -> 1.0-r83 > ( base )
Broken packagegroup-base-extended:arm64 Depends on packagegroup-base [ arm64 ] < none -> 1.0-r83 > ( base )
  Considering packagegroup-base:arm64 0 as a solution to packagegroup-base-extended:arm64 9999
    Reinst Failed early because of kernel-module-nvs:arm64
    Reinst Failed because of packagegroup-machine-base:arm64
Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 packagegroup-base-extended : Depends: packagegroup-base but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

ERROR: image-custom-1.0-r0 do_rootfs: 
ERROR: image-custom-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/ubuntu/tegra/build/tmp/work/j140_tx2-poky-linux/image-custom/1.0-r0/temp/log.do_rootfs.20208
ERROR: Task (/home/ubuntu/tegra/build/../sources/meta-tegra-bilberry/recipes-tegra-bilberry/images/image-custom.bb:do_rootfs) failed with exit code '1'
```

Austriker, 2020-02-19:
I didn't have this issue with the meta-tegra branch warrior-l4t-r32.2

madisongh, 2020-02-19:
Good, so it's `kernel-module-nvs` that's missing.  With L4T R32.1, the TX2 kernel config has `CONFIG_NVS=m`, so that module gets built, and kernel module is listed in MACHINE_EXTRA_RDEPENDS (in `conf/machine/include/tegra186.inc`).  If you're mixing a custom R32.2-based kernel or kernel config with L4T R32.1, you will need to do some work to either rebase your customizations on the R32.1 kernel, or remove that dependency for your builds.  But I'm not sure how well things will work for the latter case.

Austriker, 2020-02-19:
@madisongh Oh I see I forgot to modify the defconfig when I downgraded the kernel.

Question : How do I get the full defconfig for the good kernel except from your repo. Auvidea doesn't provide in their source modifications the defconfig for example.

madisongh, 2020-02-19:
The `defconfig` files in the repo here are ones I've extracted from the `kernel_headers.tbz2` tarball in the L4T BSP with each version.   Unless Auvidea makes modifications, you should be able to diff the configs from version to version to see what changed.

If Auvidea did make config changes, and you are able to boot their kernel on a device, you should be able to get the config used to build their kernel from `/proc/config.gz`.

Austriker, 2020-02-20:
Thank you for help. It was indeed the defconfig.

closed 2020-02-20 by Austriker