OE4T/meta-tegra #264: Warrior and Openssl utility not found

Austriker, 2020-03-04:
Hi,

I have added to my image openssl even if some recipes have` RDEPENDS += "openssl"`
```sh
IMAGE_INSTALL_append = "\
	iperf3 \
	jetsonclocks \
	openssl \
"
```

but when i execute openssl on the jetson the command is not found.
I was wondering if someone experienced the same issue. Is it possible I forgot to configure my machine badly and looking at the openssl recipe the switch case is not working properly ?
```sh
	case $target in
	linux-arm*)
		target=linux-armv4
		;;
	linux-aarch64*)
		target=linux-aarch64
		;;
	linux-i?86 | linux-viac3)
		target=linux-x86
		;;
	linux-gnux32-x86_64 | linux-muslx32-x86_64 )
		target=linux-x32
		;;
	linux-gnu64-x86_64)
		target=linux-x86_64
		;;
	linux-mips | linux-mipsel)
		# specifying TARGET_CC_ARCH prevents openssl from (incorrectly) adding target architecture flags
		target="linux-mips32 ${TARGET_CC_ARCH}"
		;;
	linux-gnun32-mips*)
		target=linux-mips64
		;;
	linux-*-mips64 | linux-mips64 | linux-*-mips64el | linux-mips64el)
		target=linux64-mips64
		;;
	linux-microblaze* | linux-nios2* | linux-sh3 | linux-sh4 | linux-arc*)
		target=linux-generic32
		;;
	linux-powerpc)
		target=linux-ppc
		;;
	linux-powerpc64)
		target=linux-ppc64
		;;
	linux-riscv32)
		target=linux-generic32
		;;
	linux-riscv64)
		target=linux-generic64
		;;
	linux-sparc | linux-supersparc)
		target=linux-sparcv9
		;;
	esac
```

madisongh, 2020-03-04:
I think the openssl command is packaged separately; try using `openssl-bin` in your IMAGE_INSTALL_append.

Austriker, 2020-03-05:
Yes that was it.

Thank you very much

closed 2020-03-05 by Austriker