OE4T/meta-tegra #265: How to get the signed binaries used for flashing the partitions?

deblauwetom, 2020-03-19:
Hello,

I used the warrior branch and jetpack 4.2 in the past and am now updating to jetpack 4.3 from the master branch. I could get the signed binaries that would be flashed onto an sdcard by applying a fix that I save the "signed" as a zip file, and I have put this just before the line where the "signed" directory is removed:
```
        zip -r ${IMGDEPLOYDIR}/${IMAGE_NAME}.tegraflash-signed.zip signed
        ln -sf ${IMAGE_NAME}.tegraflash-signed.zip ${IMGDEPLOYDIR}/${IMAGE_LINK_NAME}.tegraflash-signed.zip
	rm -r signed
```
I could then use this in my own "fwup" script to flash my sdcard. 
Now with jetpack 4.3 it seems it got much more complicated. I don't find how to get this signed directory now. What has changed and how could I get this functionality back?
Thanks a lot,
Best regards

deblauwetom, 2020-03-20:
I fixed it by setting a "flashvars" file, because I don't want to require a real device to make the firmware. 
The contents was like this:
```
BOARDID=3448 
FAB=301
BOARDSKU=0000 
BOARDREV=b00
```
And then I could run the dosdcard.sh script and this would create the signed directories too.

closed 2020-03-20 by deblauwetom

madisongh, 2020-03-20:
Yes, if you know the right settings for the device(s) you're going to use the signed binaries on, just setting those variables in the environment, or creating a flashvars file for the script to source, should work fine.

deblauwetom, 2020-03-20:
It still would be cool if I could already have only the signed binaries in a subdir in the deploy directory. Now I have to do a lot of effort to get these files... unzipping the tegraflash zipfile, then create the flashvars, then create the sdcard image, just to get these signed binaries. So if this flow could be improved it would be easier to for example add a step to create an "fwup" update file.

madisongh, 2020-03-20:
You could do this by adding a `tegraflash_custom_post` function in your image recipe, or creating a custom image type bbclass.

You might also consider using the OTA update support that NVIDIA added for the Nano and TX1 in R32.3.1, using the BUP payload instead of packaging up the signed binaries yourself.