OE4T/meta-tegra #268: Mender update for the Jetson Xavier build fails

focusware-develop, 2020-03-22:
Hi Matt,

I'm testing the test-distro build for the zeus-mender-l4t-r32.3.1 branch on a Jetson Xavier. When running the mender installer I get the following error:

`mender -install core-image-minimal-jetson-xavier.mender`

`INFO[0000] Loaded configuration file: /var/lib/mender/mender.conf  module=config
INFO[0000] Loaded configuration file: /etc/mender/mender.conf  module=config
INFO[0000] Mender running on partition: /dev/mmcblk0p1   module=main
INFO[0000] Start updating from local image file: [core-image-minimal-jetson-xavier.mender]  module=standalone
Installing Artifact of size 65272320...
INFO[0000] Update Module path "/usr/share/mender/modules/v3" could not be opened (open /usr/share/mender/modules/v3: no such file or directory). Update modules will not be available  module=modules
INFO[0000] installer: authenticated digital signature of artifact  module=installer
INFO[0000] native sector size of block device /dev/mmcblk0p42 is 512, we will write in chunks of 1048576  module=dual_rootfs_device
INFO[0000] opening device /dev/mmcblk0p42 for writing    module=block_device
INFO[0000] partition /dev/mmcblk0p42 size: 15034482688   module=block_device
................................   1% 1024 KiB
................................   3% 2048 KiB
................................   4% 3072 KiB
................................   6% 4096 KiB
...
................................  93% 59392 KiB
................................  94% 60416 KiB
................................  96% 61440 KiB
................................  97% 62464 KiB
................................  99% 63488 KiB
.......INFO[0145] wrote 14282760192/14282760192 bytes of update to device /dev/mmcblk0p42  module=dual_rootfs_device
                          100% 63742 KiB
INFO[0147] Enabling partition with new image installed to be a boot candidate: 42  module=dual_rootfs_device
ERRO[0147] stderr collected while running script /var/lib/mender/scripts/ArtifactInstall_Leave_80_bl-update [ERR: bootloader update failed
]  module=executor
ERRO[0147] ArtifactInstall_Leave script failed: statescript: error executing 'ArtifactInstall_Leave_80_bl-update': 1 : exit status 1  module=standalone
Rolling back Artifact...
INFO[0147] setting partition for rollback: 42            module=dual_rootfs_device
ERRO[0147] statescript: error executing 'ArtifactInstall_Leave_80_bl-update': 1 : exit status 1  module=main`

Do you have any suggestions on how to proceed with this?

Best regards,
Boris.

madisongh, 2020-03-22:
Try using core-image-base instead of core-image-minimal for your testing.  I'm don't think core-image-minimal will pull in everything that you need.  I just ran a quick test using core-image-base on my Xavier, and the mender update worked fine.

focusware-develop, 2020-03-22:
I'll try that and let you know. These are the steps I use to build the image

> git clone https://github.com/madisongh/test-distro.git
cd test-distro
git checkout zeus-mender-l4t-r32.3.1
git submodule update --init --recursive
source ./setup-env 
bitbake core-image-base

madisongh, 2020-03-22:
There are a couple of possibilities then.

First, I see your Xavier has a BOARDREV of "L.0".  The BUP generator doesn't handle that board revision yet, so nv_update_engine may be complaining because it's not finding some needed boot files that are matched by TNSPEC.  You could try adding
```
TEGRA_BUPGEN_SPECS_append = " fab=400;boardsku=0001;boardrev=L.0"
```
to your `local.conf` to see if that helps.

Second, if your device has the secure boot fuses set, then you'll need to make some modifications to digsigserver to have it sign the boot files properly.  Right now, digsigserver only supports signing TX2 boot files, I haven't had the guts to risk bricking my one Xavier dev kit to get the secure boot support straightened out on that platform.

focusware-develop, 2020-03-22:
Adding the "L.0" board revision the the local.conf file worked like a charm!

>INFO[0000] Loaded configuration file: /var/lib/mender/mender.conf  module=config
INFO[0000] Loaded configuration file: /etc/mender/mender.conf  module=config
INFO[0000] Mender running on partition: /dev/mmcblk0p1   module=main
INFO[0000] Start updating from local image file: [core-image-minimal-jetson-xavier.mender]  module=standalone
Installing Artifact of size 65401344...
INFO[0000] Update Module path "/usr/share/mender/modules/v3" could not be opened (open /usr/share/mender/modules/v3: no such file or directory). Update modules will not be available  module=modules
INFO[0000] installer: authenticated digital signature of artifact  module=installer
INFO[0000] native sector size of block device /dev/mmcblk0p42 is 512, we will write in chunks of 1048576  module=dual_rootfs_device
INFO[0000] opening device /dev/mmcblk0p42 for writing    module=block_device
INFO[0000] partition /dev/mmcblk0p42 size: 15034482688   module=block_device
................................   1% 1024 KiB
................................   3% 2048 KiB
................................   4% 3072 KiB
...
................................  97% 62464 KiB
................................  99% 63488 KiB
...........INFO[0144] wrote 14282760192/14282760192 bytes of update to device /dev/mmcblk0p42  module=dual_rootfs_device
                      100% 63868 KiB
INFO[0146] Enabling partition with new image installed to be a boot candidate: 42  module=dual_rootfs_device
Use -commit to update, or -rollback to roll back the update.
At least one payload requested a reboot of the device it updated.

Thank you for your assistance and keep up the good work!
Boris.

I'll submit two other pull requests that fixes some other small problems in the firmware image.

closed 2020-03-22 by focusware-develop