OE4T/meta-tegra #270: Xavier startup issue until "crng init done"

lfdmn, 2020-03-23:
Hi, 

I'm on warrior-l4t-r32.2 with a Xavier dev kit. My issue is with boot time.

According to systemd-analyse, the boot time is 25s and all systemd startup tasks are completed.

In reality, I cannot use the system for the first ~5mins. All the device applications hang until I see this in dmesg
```bash
crng init done
```

After that all the applications start and I can use the system normally.

Anyone seing the same behavior?

Thanks,

madisongh, 2020-03-23:
I've seen this, but not consistently.  I've never root-caused the exact issue, but near as I can tell there are multiple things going on - lack of HWRNG support in the BSP and maybe some bugs in rngd itself.

One thing to try... the `/etc/default/rng-tools` file that comes with OE-Core points it at `/dev/hwrng`, which doesn't exist on the Jetsons (it looks like the security engine might have that feature, but the driver isn't hooked up).   The stock L4T BSP uses `/dev/urandom`, which is probably wrong (using the receiver of entropy as an entropy generator?), but you could try that in your setup to see if it works better.

What I wound up doing is dumping rng-tools from my builds altogether and using haveged instead.  If/when NVIDIA hooks in hardware RNG support in their kernel, I'll probably try switching back again.

lfdmn, 2020-03-24:
Thanks Matt!

I'll check which option works best in my case.

lfdmn, 2020-03-24:
haveged worked for me. 

Just with that change, the boot time is now 15s and I have not started optimizing yet.

Thanks again.

closed 2020-03-24 by lfdmn