OE4T/meta-tegra #271: Factory Xavier EEPROMs + tegra-eeprom-tool

kekiefer, 2020-03-26:
Posting this here since there's a bit more visibility, but this is specifically related to https://github.com/madisongh/tegra-eeprom-tool

I don't know if the TX2 and Nano work properly. I think we need some additional considerations for Xavier, considering that the factory EEPROMs don't follow the documented format.

The EEPROM layout document that NVIDIA publish is geared towards making custom modules and carriers, so that the bootloader can apply its device tree overlays. A while back, I wrote my own tool to set up the EEPROM for custom carriers, and it's compatible with this tool, and works as expected with the bootloader.

My concern is that I don't think we should have to muck around with the factory module/CVM eeprom (0x50), but none of the update engine stuff will work unless we do.

Here is a factory-programmed Xavier module eeprom:

```
00000000  00 ff 00 48 0b 01 00 04  4a 00 00 00 00 00 00 00  |...H....J.......|
00000010  00 00 00 36 39 39 2d 38  32 38 38 38 2d 30 30 30  |...699-82888-000|
00000020  31 2d 34 30 30 20 4a 2e  30 00 00 00 00 00 00 00  |1-400 J.0.......|
00000030  00 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000040  ff ff ff 93 a2 cb 4b 04  00 30 34 32 33 34 31 38  |......K..0423418|
00000050  30 31 30 34 39 36 00 00  00 00 00 00 00 00 00 00  |010496..........|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000090  00 00 00 00 00 4e 56 43  42 1c 00 4d 31 00 00 ff  |.....NVCB..M1...|
000000a0  ff ff ff ff ff ff ff ff  ff ff ff 93 a2 cb 4b 04  |..............K.|
000000b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 7b 00  |..............{.|
00000100
```

And an 8GB module, for good measure:

```
00000000  00 ff 00 48 0b 06 00 04  42 00 00 00 00 00 00 00  |...H....B.......|
00000010  00 00 00 36 39 39 2d 38  32 38 38 38 2d 30 30 30  |...699-82888-000|
00000020  36 2d 34 30 30 20 42 2e  30 00 00 00 00 00 00 00  |6-400 B.0.......|
00000030  00 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000040  ff ff ff 02 84 e5 4b 04  00 31 34 32 33 32 31 39  |......K..1423219|
00000050  30 38 32 38 36 35 00 00  00 00 00 00 00 00 00 00  |082865..........|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000090  00 00 00 00 00 4e 56 43  42 1c 00 4d 31 00 00 ff  |.....NVCB..M1...|
000000a0  ff ff ff ff ff ff ff ff  ff ff ff 02 84 e5 4b 04  |..............K.|
000000b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 e5 00  |................|
00000100
```

As you can see, the checksum isn't in offset 255, like the eeprom layout doc specifies. Also we can also look at a stock Xavier carrier (0x56) eeprom, which is the more sensical target for customization considering that customer carrier boards will have different requirements than the stock carrier:

```
00000000  00 ff 00 06 0b 00 00 06  44 00 00 00 00 00 00 00  |........D.......|
00000010  01 3c 11 36 39 39 2d 38  32 38 32 32 2d 30 30 30  |.<.699-82822-000|
00000020  30 2d 36 30 30 20 44 2e  30 00 00 00 00 00 00 00  |0-600 D.0.......|
00000030  00 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000040  ff ff ff ff ff ff ff ff  ff 30 34 32 33 34 31 38  |.........0423418|
00000050  30 32 34 34 31 32 00 00  00 00 00 00 00 00 00 00  |024412..........|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000090  00 00 00 00 00 46 46 46  46 ff ff 46 46 ff ff ff  |.....FFFF..FF...|
000000a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000000b0  ff 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 9b 00  |................|
00000100
```

Anyhow, back to the problem at hand. For the CVM I can bypass the checksum, but as you might guess the results won't be quite right:

```
partnumber[nvidia]: 9-82888-0001-400 J.0
factory-default-wifi-mac: ff:ff:ff:ff:ff:ff
factory-default-bt-mac: ff:ff:ff:ff:ff:ff
factory-default-wifi-alt-mac: a2:93:ff:ff:ff:ff
factory-default-ether-mac: 34:30:00:04:4b:cb
asset-id: 23418010496
vendor-wifi-mac: ff:ff:ff:ff:ff:ff
vendor-bt-mac: a2:93:ff:ff:ff:ff
vendor-ether-mac: 00:00:00:04:4b:cb
```

Note that for example the MAC for this machine should be `00:04:4b:cb:a2:93`.

I think it's worth fixing up the tool to work with the factory Xavier EEPROM, would you agree? I'm going to look into decoding the factory format, since I have a handful of different modules to check against.

madisongh, 2020-03-26:
Looks like they're shifted left by one byte.  The first byte in the eeprom should be `01`, the lower half o f the two-byte layout version number.  The CVM EEPROM on my Xavier (also a J.0) reads OK.  Weird.

kekiefer, 2020-03-26:
It's probably worth pointing out that I do not have the AT24 module in this kernel, so the tool is using the i2c mode.

When I use the eeprog tool in 8-bit mode, it appears without that shift:

```
$ ./eeprog -f -8 /dev/i2c-0 0x50 -r 0:256 | hexdump -C
eeprog 0.7.5, a 24Cxx EEPROM reader/writer
Copyright (c) 2003 by Stefano Barbato - All rights reserved.
  Bus: /dev/i2c-0, Address: 0x50, Mode: 8bit
  Reading 256 bytes from 0x0
00000000  00 00 ff 00 48 0b 01 00  04 4a 00 00 00 00 00 00  |....H....J......|
00000010  00 00 00 00 36 39 39 2d  38 32 38 38 38 2d 30 30  |....699-82888-00|
00000020  30 31 2d 34 30 30 20 4a  2e 30 00 00 00 00 00 00  |01-400 J.0......|
00000030  00 00 ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000040  ff ff ff ff 93 a2 cb 4b  04 00 30 34 32 33 34 31  |.......K..042341|
00000050  38 30 31 30 34 39 36 00  00 00 00 00 00 00 00 00  |8010496.........|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000090  00 00 00 00 00 00 4e 56  43 42 1c 00 4d 31 00 00  |......NVCB..M1..|
000000a0  ff ff ff ff ff ff ff ff  ff ff ff ff 93 a2 cb 4b  |...............K|
000000b0  04 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 7b  |...............{|
00000100
```

madisongh, 2020-03-26:
Do you have i2c-tools in your build?  If so, does `i2c-dump -y -r 0-0xff 0 0x50 b` produce the same result?

It's certainly possible that the userland I2C interface isn't working for 100% of cases - either due to my misusing it, or something else - but for reading, at least, I haven't run into any issues with it garbling the data like this. I've only run into issues with writes, so for the one project where I needed to fill in the vendor MAC addresses, I use the at24 driver.

kekiefer, 2020-03-26:
i2cdump looks ok (version number discrepancy aside):

```
root@ccu3:~# i2cdump -y -r 0-0xff 0 0x50 b
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 00 00 ff 00 48 0b 06 00 04 42 00 00 00 00 00 00    ....H??.?B......
10: 00 00 00 00 36 39 39 2d 38 32 38 38 38 2d 30 30    ....699-82888-00
20: 30 36 2d 34 30 30 20 42 2e 30 00 00 00 00 00 00    06-400 B.0......
30: 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
40: ff ff ff ff 6d 83 e5 4b 04 00 31 34 32 33 32 31    ....m??K?.142321
50: 39 30 38 32 38 34 30 00 00 00 00 00 00 00 00 00    9082840.........
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
90: 00 00 00 00 00 00 4e 56 43 42 1c 00 4d 31 00 00    ......NVCB?.M1..
a0: ff ff ff ff ff ff ff ff ff ff ff ff 6d 83 e5 4b    ............m??K
b0: 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ?...............
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 60    ...............`
```

I'll do a little more digging and report back.

kekiefer, 2020-03-26:
It looks like i2c reads via tegra-eeprom-tool do not suffer the same offset. So those dumps were misleading, as I was using 16-bit mode (I have a 24C256 on my custom board). Sorry about that red herring.

Still, the checksum fails in tegra-eeprom-tool. However, if I I modify the version `00` that was read back to be `01`, the checksum becomes valid, and the tool is working as expected.

I've checked two 8GB Xavier modules and one 16GB Xavier module, and all three of them have `00` in the factory-programmed CVM eeprom. This fixup makes all of them check out OK.

madisongh, 2020-03-26:
One more thing to check - the flashing tools have the bootloader send over the CVM contents (`dump eeprom boardinfo <filename>` command in tegraflash.py).  The latest flash helper scripts use a temporary file and remove it after they're done, but it would be interesting to see if the bootloader's getting the same contents you're seeing.

kekiefer, 2020-03-26:
This was where I first started noticing something fishy, and got around it by manually specifying the environment it needed.

It fails to return data when the host invokes `dump eeprom boardinfo`. This is the log from the target:

```
[0322.727] I> Populate eeprom info
[0322.731] I> Populate eeprom info for module cvm
[0322.760] E> verify_cvm_eeprom_version: EEPROM incompatible version found
[0322.767] E> NV3P_SERVER: Failed to get eeprom handle for cvm from eeprom manager.
[0322.774] E> NV3P_SERVER: Could not read eeprom for module cvm.
```

When I set the eeprom data[0] byte to 1, it works as expected.

How is the factory flash program coping with this? Unfortunately I can't try with the nvidia carrier and flashing tools, since I don't have this setup at home. It would be easy to modify a properly versioned eeprom to look like the factory ones with `i2cset -f -y 0 0x50 0 0 b`.

madisongh, 2020-03-27:
OK, that at least confirms that NV's tools find that the EEPROMs are corrupted, too.  So the question is how did they get that way?  I wouldn't expect them to have left the factory that way... at least I'd hope not, since you've got different modules manufactured at different times, all with exactly the same defect, and that would be a heck of a coincidence.

Are you using that I2C bus for some other peripheral(s)?  Maybe an ill-behaved driver is accidentally writing a 0 to the wrong address on that bus?

kekiefer, 2020-03-27:
It looks like just using `eeprog` to read in 16-bit mode will corrupt the first byte! Yikes.

Thanks for your help on this one, and sorry for the noise.

closed 2020-03-27 by kekiefer