OE4T/meta-tegra #272: Displayport on Jetson Nano devkit not working

MJLHThomassen-Sorama, 2020-03-27:
I've created an core-image-base image using poky and the meta-tegra layer for the Jetson Nano (MACHINE = "jetson-nano-qspi-sd") . It boots the devkit correctly and shows console when i plug in hdmi. When i plug in the display port on the devkit i get a black screen. Is there anything i need to do extra to enable display port?

madisongh, 2020-03-27:
I don't think there is, but I haven't tested with a DP monitor.  I do see some issues over on the NVIDIA forums on this topic.  Does it work if you run the stock L4T/JetPack Nano build?

MJLHThomassen-Sorama, 2020-03-28:
Yeah i noticed those issues too (dont have the exact link atm). It does work with the default L4T sd-card image that comes from the nvidia site (which is why it took me a few hours to realize the image i was building was actually working, but the display was not showing anything). At first i thought my image was just not working or i flashed my devkit the wrong way but everything is working fine with hdmi.

madisongh, 2020-03-29:
I tried some tests with my 4K Dell monitor, and see the same.  It looks like the DP link training never succeeds when it's just the framebuffer being used.  (The NVIDIA boot logo never shows, either.) If I boot a sato or weston image, the display does come up in X/Weston.

MJLHThomassen-Sorama, 2020-03-30:
Alright, good to hear you can reproduce the issue and know where to look (i have no clue). I have not tried a sato image yet since we wanted to start building our own image from the core-image-minimal since our application actually does not need a display, but for debugging purposes it would be nice to have a display.

madisongh, 2020-04-01:
I don't think there's much I can really do about this, unfortunately.  I strongly suspect it is a problem in L4T itself - some kind of interaction between the DP driver and the framebuffer driver.

ichergui, 2021-09-07:
Hello guys @MJLHThomassen-Sorama @madisongh 
Anyone try the latest `L4T` release to check if it works ? if no, I will try it later and will let you know.
I'm planning to check with `stock L4T` and Yocto `tegra-demo-distro` ?

ichergui, 2021-09-12:
Hello  @MJLHThomassen-Sorama @madisongh 
I tried the `DP` with the latest release `L4T R32.6.1` and it doesn't work. I need to try with NV `stock L4T`

MJLHThomassen-Sorama, 2021-09-13:
We are currently not actively using the devkits anymore since we have our own boards now so I'm afraid I can't be of further assistance here.

madisongh, 2021-12-18:
Since this isn't an issue with the layer, I'm moving this to the discussion board.