OE4T/meta-tegra #273: Missing gstreamer file

vjdaemp, 2020-03-30:
It seems that a gstreamer file is missing from the "recipes-multimedia/gstreamer" folder:

> ERROR: ParseError at /home/vjd/Build/Poky/meta-tegra/recipes-multimedia/gstreamer/gstreamer1.0-plugins-nveglgles_1.2.3-r32.3.1.bb:56: Could not include required file recipes-multimedia/gstreamer/gstreamer1.0-plugins-packaging.inc

Where can I obtain this file?

madisongh, 2020-03-30:
Looks like you've got the `master` branch checked out, rather than `zeus-l4t-r32.3.1`.  The `.inc` file would come from the OE-Core layer (poky/meta or openembedded-core/meta), but the gstreamer include files got reorganized after the zeus release, and that line mentioned in the error is current for master.

vjdaemp, 2020-03-30:
Thank you @madisongh . I mistakenly thought I had checked out the zeus branch, but I was still on master as you mentioned.  Everything is working now -- thanks!

closed 2020-03-30 by vjdaemp