OE4T/meta-tegra #280: Nvidia SDK packages from repo.download.nvidia.com instead of mirror from sdkmanager download

ericvh, 2020-04-02:
Given NVIDIA has these in ubuntu package repos, can't we fetch them using apt (or some other direct method fetching from the source) instead of mirroring locally using our dev login?

```
apt-get install --reinstall nvidia-l4t-jetson-multimedia-api
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 70.4 MB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://repo.download.nvidia.com/jetson/t194 r32/main arm64 nvidia-l4t-jetson-multimedia-api arm64 32.3.1-20191209230245 [70.4 MB]
Fetched 70.4 MB in 9s (7,678 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 137462 files and directories currently installed.)
Preparing to unpack .../nvidia-l4t-jetson-multimedia-api_32.3.1-20191209230245_arm64.deb ...
Unpacking nvidia-l4t-jetson-multimedia-api (32.3.1-20191209230245) over (32.3.1-20191209230245) ...
Setting up nvidia-l4t-jetson-multimedia-api (32.3.1-20191209230245) ...
```

madisongh, 2020-04-02:
Thanks for pointing this out... it would definitely be more convenient!  I'll look into it.

madisongh, 2020-04-15:
I've looked into this a bit... it should be doable for the packages installed on the device, but if you're building any CUDA applications, you'll still need the NVIDIA_DEVNET_MIRROR setup to get at the host-side tools - they don't appear to be distributed via a package feed.

Even for the on-device packages, I'm not sure how easy it will be to track what's provided via the SDK Manager/JetPack vs. what they supply in the package feed, since the naming is different.  I'll keep looking at it, though.

madisongh, 2020-04-26:
For the L4T R32.4.2/JetPack 4.4DP update, I've moved ahead with using the Debian feeds where I can.  Changes are in `master` and the `dunfell-l4t-r32.4.2` branches.  As I mentioned earlier, you still have to have NVIDIA_DEVNET_MIRROR set up for  the CUDA host-side tools.

closed 2020-04-26 by madisongh