OE4T/meta-tegra #281: Updating firmware on the field

lfdmn, 2020-04-03:
Hi,

Not really a meta-tegra issue but just a general question to the users here.

Tegra flash will always format and overwrite the eMMC module. It’s obviously good for production.

I’d like to be able to upgrade our Tegra based firmware by inserting a USB stick / or download some sort of image on a different partition, then format the partitions one by one and exclude the ones with persistent data.

Where would you implement this?

I could for example format and write the image partition from initramfs script. Would it be wise to do it there or do you raise some red flags?

But what if I need to re-write the initramfs image or update the kernel then? Should this be implemented from u-boot or elsewhere? Any pointers at example or how to get started?

Thanks,

dwalkes, 2020-04-03:
@lfdmn you could accomplish this with [mender](https://hub.mender.io/t/nvidia-tegra-jetson-tx2/123) and [standalone deployments](https://docs.mender.io/2.3/architecture/standalone-deployments).  If you'd rather not use mender you could use the source code to help understand how to implement your own scheme.  See https://github.com/Trellis-Logic/meta-mender-community/pull/11 for the latest zeus branch I've tried based on @madisongh's latest, just waiting on a Zeus update for https://github.com/mendersoftware/meta-mender-community before merging this.

mirzak, 2020-04-03:
Do not mean to hog the thread, but @dwalkes, Just an FYI, I started a [zeus-test](https://github.com/mendersoftware/meta-mender-community/tree/zeus-test) branch and will accept PR's there and convert that to a proper `zeus` branch once upstream has been tagged.

dwalkes, 2020-04-03:
Thanks @mirzak I'll see if I can get a PR started there over the weekend.

lfdmn, 2020-04-06:
Thanks @dwalkes, it looks interesting

outbackdingo, 2020-05-25:
i think your better of using ruac, ostree and aktualizer then mender.....

madisongh, 2020-12-11:
This topic has come up several times in different contexts.  I'm going to close this issue now - please feel free to start a [discussion](https://github.com/OE4T/meta-tegra/discussions) if you'd like to discuss further.

closed 2020-12-11 by madisongh