OE4T/meta-tegra #284: NVIDIA_DEVNET_MIRROR mandatory on `warrior`

mirzak, 2020-04-07:
It seems with the update in https://github.com/madisongh/meta-tegra/commit/b5e7cdab655090d7989ef94b169e67837dde85a6#diff-615d50cc49b82960424c074e6054cac2, it is now mandatory to have `NVIDIA_DEVNET_MIRROR` defined or else you get an parse error. 

I do not know if this change was intentional, and I have not really tried how it is on other branches but I wanted to get your feedback on this.

madisongh, 2020-04-07:
Yeah, unfortunately, the secure boot tools for L4T R32.1 were only available via manual download from NVIDIA's developer network.  I've just put in a change that skips the secure boot tooling support if you don't have NVIDIA_DEVNET_MIRROR set, which should restore the prior behavior if you don't need the devnet extras.

closed 2020-04-07 by madisongh

mirzak, 2020-04-07:
Awesome! Thanks for the quick response and resolve. Closing this

PR #286 (2020-04-07): tegra186-flashtools-native: make NVIDIA_DEVNET_MIRROR optional
For builds not requiring secure boot support on tegra186
platforms, skip downloading the secureboot package if
NVIDIA_DEVNET_MIRROR is not defined.

[Fixes #284]