OE4T/meta-tegra #285: Question: how to read gpio pin from u-boot

lfdmn, 2020-04-07:
Hi,

I need to read the state of FORCE_RECOVERY# button at boot. When running linux, it is mapped as volume UP gpio-313.

I tried to read from u-boot shell but I could not find the bank.

Then I went editing p2771-0000.c

```c
#define DEBUG

#include <common.h>
#include <i2c.h>
#include <asm/gpio.h>
#include "../p2571/max77620_init.h"

void read_pin(void)
{
    if (gpio_request(313, "GPIO_VOLUME_UP") == -1) {
         printf("GPIO request failed for 313\n");   
    }
    else {
        printf("GPIO request succeeded for 313");
        if (gpio_direction_input(313) == -1) {
            printf("Failed to set the input direction for 313\n");   
        }
        else {
            printf("Succeeded to set the input direction for 313\n");

            udelay(3000000);
            int state = gpio_get_value(313);
            printf("The status is %s\n", state ? "HIGH" : "LOW");
        }
    }
}

int tegra_board_init(void)
{
	struct udevice *dev;
	uchar val;
	int ret;

	/* Turn on MAX77620 LDO3 to 3.3V for SD card power */
	debug("%s: Set LDO3 for VDDIO_SDMMC_AP power to 3.3V\n", __func__);
	ret = i2c_get_chip_for_busnum(0, MAX77620_I2C_ADDR_7BIT, 1, &dev);
	if (ret) {
		printf("%s: Cannot find MAX77620 I2C chip\n", __func__);
		return ret;
	}
	/* 0xF2 for 3.3v, enabled: bit7:6 = 11 = enable, bit5:0 = voltage */
	val = 0xF2;
	ret = dm_i2c_write(dev, MAX77620_CNFG1_L3_REG, &val, 1);
	if (ret) {
		printf("i2c_write 0 0x3c 0x27 failed: %d\n", ret);
		return ret;
	}

    read_pin();

	return 0;
}

int tegra_pcie_board_init(void)
{
	struct udevice *dev;
	uchar val;
	int ret;

	/* Turn on MAX77620 LDO7 to 1.05V for PEX power */
	debug("%s: Set LDO7 for PEX power to 1.05V\n", __func__);
	ret = i2c_get_chip_for_busnum(0, MAX77620_I2C_ADDR_7BIT, 1, &dev);
	if (ret) {
		printf("%s: Cannot find MAX77620 I2C chip\n", __func__);
		return -1;
	}
	/* 0xC5 for 1.05v, enabled: bit7:6 = 11 = enable, bit5:0 = voltage */
	val = 0xC5;
	ret = dm_i2c_write(dev, MAX77620_CNFG1_L7_REG, &val, 1);
	if (ret)
		printf("i2c_write 0 0x3c 0x31 failed: %d\n", ret);

    read_pin();

	return 0;
}

```

No error, but the state of 313 doesn't change, so 313 is most certainly not the correct u-boot gpio for this button.

Am I doing it wrong? Where can I find the mappings?

Also the gpio_request, gpio_direction_input and gpio_get_value as marked deprecated in favor of driver model. What is the driver model for Tegra? 

First time modifying u-boot, I'm not familiar with that stuff.

Thanks,

madisongh, 2020-04-07:
It's a complicated topic, and the documentation is somewhat lacking, at least when it comes to connecting the dots between the different hardware designations and how they're referenced in U-Boot and/or the Linux kernel.  What I usually do is:

1. Locate the pin I need in the [Pin and Function Names App Note](http://developer.nvidia.com/embedded/dlc/tx2-series-pin-function-names-guide-note) to get the hardware (SoM connector/SoC ball) name
2. Find that name in the  [Pinmux Excel spreadsheet](https://developer.nvidia.com/embedded/dlc/jetson-tx2-series-module-pinmux)
3. Look in the GPIO column of that spreadsheet to find the mapping to the controller port number, which is the lettter/number combination after the `_P` in the GPIO label - in this case the GPIO label is `GPIO3_PFF.01`, so, it's `FF` /`1`.

Then go looking in the sources for the letters - on the TX2 there are two GPIO controllers, the normal one and the always-on one.  In this case, it looks like the `FF` GPIOs are on the always-on controller, so in U-Boot you would use `TEGRA_AON_GPIO(FF, 1)` to reference this particular pin (that macro is defined in `include/dt-bindings/gpio/tegra186-gpio.h`).

Since you had a Linux naming reference already, in your case you could have found the right name by looking under `/sys/kernel/debug/pinctrl` to find `313` in the `pinmux-pins` file for the right pinmux controller, which would have got you the GPIO name (`GPIO_SW1_PFF1`) and whether it was AON or not.  (Although it doesn't hurt to cross-check the hardware docs to make sure you've got the right designation.)

Don't worry about the deprecation warning - U-Boot has been updating its driver APIs over the years, but for the version we have, I don't think the Tegra platforms have `CONFIG_DM_GPIO` set, so you have to use the legacy API.

(While the above references are TX2-specific, there are similar app notes and spreadsheets available for download for the other Jetsons. Pin-to-controller/port/register mappings will be different for each platform.)

lfdmn, 2020-04-07:
Thanks a lot!

I actually was browsing all these file today and was wondering how to connect the dots as you said.
```bash
#include <dt-bindings/gpio/tegra186-gpio.h>
// ...
int pin = TEGRA_AON_GPIO(FF,1);
gpio_request(pin, "GPIO_VOLUME_UP");
```

Works now and I can find GPIO_VOLUME_UP in the gpio register
```bash
# gpio status H -a
Bank H:
H1: input: 0 [x] GPIO_VOLUME_UP
```

Is u-boot reading the pin state each time I query H1 or only at initialization?
```bash
# gpio input H1
gpio: pin H1 (gpio 57) value is 0
```
No matter if I press the button or not, the value always is 0.

madisongh, 2020-04-07:
Hmm.  Something doesn't look right there.  That label should be getting assigned to `FF1`, not `H1`.

lfdmn, 2020-04-07:
Alright, I probably screwed up =) I'll keep debugging tomorrow. Thanks!

lfdmn, 2020-04-08:
```c
int pin = TEGRA_AON_GPIO(FF,1);
```
Returns 57 which probably is the number in the AON controller. I guess if I needed to request the port I should use gpio_lookup_name or name_to_gpio instead to get the correct pin number.

But since I can read the value directly from script now, no need for all this. Thanks!

Do you happen to know where the key combination for entering the device into flashing mode is set?

madisongh, 2020-04-08:
>Do you happen to know where the key combination for entering the device into flashing mode is set?

Do you mean which button controls recovery mode entry? I don't know for sure, but my guess is it's hard-coded in the on-chip boot ROM.

lfdmn, 2020-04-16:
It seems to be the case.

closed 2020-04-16 by lfdmn