OE4T/meta-tegra #287: NVIDIA Container Runtime support wiki

vjdaemp, 2020-04-08:
I was able to successfully build meta-tegra (thud-l4t-r32.3.1) and deploy to device. Now I'd like to add NVIDIA Container Runtime support , and have been following instructions on Wiki page. I encountered an error that meta-filesystems was required, so I added this layer. Now I am encountering another error after executing  'bitbake core-image-base':

> ERROR: No recipes available for:
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/meta-tegra/external/openembedded-layer/recipes-support/libusbgx/libusbgx-config.bbappend
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/meta-tegra/external/openembedded-layer/recipes-support/libusbgx/libusbgx_%.bbappend
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/meta-tegra/external/openembedded-layer/recipes-support/opencv/opencv_4.1.0.bbappend

**Added the following to bblayers.conf: **

> BBLAYERS ?= " \
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/meta \
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/meta-poky \
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/meta-yocto-bsp \
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/meta-virtualization \
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/meta-openembedded/meta-oe \
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/meta-openembedded/meta-networking \
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/meta-openembedded/meta-python \
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/meta-openembedded/meta-filesystems \
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/meta-tegra \
  "

**Added the following to local.conf:**

> DISTRO_FEATURES_append = " virtualization"

** Below are the versions I am using: **

-  poky / thud-20.0.4

- meta-tegra / thud-l4t-r32.3.1

- meta-virtualization / thud

- meta-openembedded / thud

Any help would be appreciated.

madisongh, 2020-04-08:
Sorry, but the container support hasn't been back-ported to the thud-l4t-r32.3.1 branch.  I've updated the wiki to specify the right branches.

You can use GCC 7 with the zeus-l4t-r32.3.1 branch by adding `meta-tegra/contrib` to your `bblayers.conf` file - recipes for the gcc 7 toolchain are present in that layer.

vjdaemp, 2020-04-08:
Thank you very much for the feedback. I did not realize I could use `zeus` and still have GCC 7 / CUDA 10. I've updated `poky`, `meta-tegra`, `meta-virtualization`, `meta-openembedded` from `thud` to `zeus`. I also added `meta-tegra/contrib` to the `bblayers.conf` file as you advised.

Unfortunately I am getting a new error after running `bitbake core-image-base`. Do you have any recommendations ( I am new to yocto ):

> Loading cache: 100% |############################################| Time: 0:00:00
Loaded 3671 entries from dependency cache.
Parsing recipes: 100% |##########################################| Time: 0:00:00
Parsing of 2480 .bb files complete (2479 cached, 1 parsed). 3672 targets, 130 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-nano-qspi-sd"
DISTRO               = "poky"
DISTRO_VERSION       = "3.0.2"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "jetson-devkit-yocto-zeus:5e1f52edb7a9f790fb6cb5d96502f3690267c1b1"
meta-virtualization  = "zeus:69f08a9f5672468fc41081e2dbb0dcc666684b64"
meta-oe              
meta-networking      
meta-python          
meta-filesystems     = "zeus:9e60d30669a2ad0598e9abf0cd15ee06b523986b"
meta-tegra           
contrib              = "zeus-l4t-r32.3.1:799bd56a509667ae131baec285e486e4a44ddce4"

Initialising tasks: 100% |#######################################| Time: 0:00:01
Sstate summary: Wanted 875 Found 0 Missed 875 Current 573 (0% match, 39% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
ERROR: libgcc-initial-7.3.0-r0 do_configure: Execution of '/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/libgcc-initial/7.3.0-r0/temp/run.do_configure.2630' failed with exit code 1:
configure: WARNING: unrecognized options: --disable-silent-rules, --disable-dependency-tracking, --with-libtool-sysroot, --with-gnu-ld, --enable-languages, --enable-threads, --disable-multilib, --enable-default-pie, --enable-c99, --enable-long-long, --enable-symvers, --enable-libstdcxx-pch, --without-local-prefix, --enable-lto, --disable-libssp, --enable-libitm, --disable-bootstrap, --disable-libmudflap, --with-system-zlib, --with-linker-hash-style, --enable-linker-build-id, --with-ppl, --with-cloog, --enable-checking, --enable-cheaders, --without-isl, --with-gxx-include-dir, --with-sysroot, --with-build-sysroot, --disable-static, --disable-nls, --enable-initfini-array, --enable-__cxa_atexit
checking build system type... x86_64-pc-linux-gnu
checking host system type... aarch64-poky-linux-gnu
checking for --enable-version-specific-runtime-libs... no
checking for a BSD-compatible install... /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/hosttools/install -c
checking for gawk... gawk
checking for aarch64-poky-linux-ar... aarch64-poky-linux-gcc-ar
checking for aarch64-poky-linux-lipo... no
checking for lipo... no
checking for aarch64-poky-linux-nm... aarch64-poky-linux-nm
checking for aarch64-poky-linux-ranlib... aarch64-poky-linux-gcc-ranlib
checking for aarch64-poky-linux-strip... aarch64-poky-linux-strip
checking whether ln -s works... yes
checking for aarch64-poky-linux-gcc... aarch64-poky-linux-gcc  -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/libgcc-initial/7.3.0-r0/recipe-sysroot
checking for suffix of object files... configure: error: in `/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/libgcc-initial/7.3.0-r0/gcc-7.3.0/build.aarch64-poky-linux.aarch64-poky-linux/libgcc':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details.
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/libgcc-initial/7.3.0-r0/temp/log.do_configure.2630
Log data follows:
| DEBUG: Executing shell function autotools_preconfigure
| DEBUG: Shell function autotools_preconfigure finished
| DEBUG: Executing python function autotools_aclocals
| DEBUG: SITE files ['endian-little', 'bit-64', 'arm-common', 'arm-64', 'common-linux', 'common-glibc', 'aarch64-linux', 'common']
| DEBUG: Python function autotools_aclocals finished
| DEBUG: Executing python function gcc_multilib_setup
| DEBUG: Python function gcc_multilib_setup finished
| DEBUG: Executing python function extract_stashed_builddir
| DEBUG: sed -e 's:^[^/]*/:/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/libgcc-initial/7.3.0-r0/gcc-7.3.0/build.aarch64-poky-linux.aarch64-poky-linux/:g' /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/sysroots-components/x86_64/gcc-stashed-builddir-aarch64-poky-linux/fixmepath | xargs sed -i -e 's:FIXMESTAGINGDIRTARGET:/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/libgcc-initial/7.3.0-r0/gcc-7.3.0/build.aarch64-poky-linux.aarch64-poky-linux:g; s:FIXMESTAGINGDIRHOST:/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/libgcc-initial/7.3.0-r0/gcc-7.3.0/build [... cut ...]
ERROR: Task (/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/meta-tegra/recipes-kernel/linux/linux-tegra_4.9.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1760 tasks of which 1758 didn't need to be rerun and 2 failed.

Summary: 2 tasks failed:
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/meta-tegra/contrib/recipes-devtools/gcc/libgcc-initial_7.3.bb:do_configure
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/meta-tegra/recipes-kernel/linux/linux-tegra_4.9.bb:do_compile
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.

madisongh, 2020-04-08:
I should have pointed you at [the wiki page](https://github.com/madisongh/meta-tegra/wiki/Using-gcc7-from-the-contrib-layer)... add `required contrib/conf/include/gcc-compat.conf` to your local.conf to get the settings needed to turn off the post-gcc-7 compiler flags that are enabled by default in OE-Core zeus.

vjdaemp, 2020-04-08:
Thanks @madisongh , the build worked!

vjdaemp, 2020-04-09:
@madisongh Yesterday the build worked, but I forgot to add `nvidia-docker` to the `local.conf` file.  After adding `nvidia-docker` to the `local.conf` file, I received another error related to the `meta-virtualization` layer.

The lines below show the additional components that I added in addition to `nvidia-docke`r:

`MACHINE = "jetson-nano-qspi-sd"
LICENSE_FLAGS_WHITELIST = "commercial"
IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash"
GCCVERSION = "7.%"
require contrib/conf/include/gcc-compat.conf
CUDA_VERSION= "10.0"
IMAGE_INSTALL_append = "  openssh nvidia-container-runtime cuda-samples nvidia-docker cudnn tensorrt usbutils"
NVIDIA_DEVNET_MIRROR = "file:///home/zebra/Downloads/nvidia/sdkm_downloads"
DISTRO_FEATURES_append = " systemd virtualization"`

The `meta-virtualization` error that I received is shown below:

```
NOTE: Executing Tasks
NOTE: Setscene tasks completed
ERROR: tini-0.18.0-r0 do_compile: Execution of '/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/temp/run.do_compile.25105' failed with exit code 1:
[1/1] : && /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc -march=armv8-a+crc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0=/usr/src/debug/tini/0.18.0-r0                           -fdebug-prefix-map=/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/recipe-sysroot=                           -fdebug-prefix-map=/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/recipe-sysroot-native=   -march=armv8-a+crc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/recipe-sysroot -std=gnu99 -Werror -Wextra -Wall -pedantic-errors -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat  -march=armv8-a+crc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/recipe-sysroot  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -Wl,-Bsymbolic-functions -Wl,-z,relro  -rdynamic -Wl,--no-export-dynamic -static CMakeFiles/tini-static.dir/src/tini.c.o  -o tini-static   && :
FAILED: tini-static 
: && /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc -march=armv8-a+crc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0=/usr/src/debug/tini/0.18.0-r0                           -fdebug-prefix-map=/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/recipe-sysroot=                           -fdebug-prefix-map=/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/recipe-sysroot-native=   -march=armv8-a+crc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/recipe-sysroot -std=gnu99 -Werror -Wextra -Wall -pedantic-errors -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat  -march=armv8-a+crc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/recipe-sysroot  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -Wl,-Bsymbolic-functions -Wl,-z,relro  -rdynamic -Wl,--no-export-dynamic -static CMakeFiles/tini-static.dir/src/tini.c.o  -o tini-static   && :
/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/ld: cannot find rcrt1.o: No such file or directory
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
WARNING: exit code 1 from a shell command.
|
ERROR: Logfile of failure stored in: /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/poky/build/tmp/work/aarch64-poky-linux/tini/0.18.0-r0/temp/log.do_compile.25105
Log data follows:
| DEBUG: Executing shell function do_compile
[... cut ...]
ERROR: Task (/home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/meta-virtualization/recipes-containers/tini/tini_0.18.0.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3537 tasks of which 3524 didn't need to be rerun and 1 failed.
|
Summary: 1 task failed:
  /home/vjdaemp/Jetson/Build/jetson-nano-dev-kit/meta-virtualization/recipes-containers/tini/tini_0.18.0.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

Any ideas on why this might be occurring?

madisongh, 2020-04-09:
Looks like the tini recipe tries to disable PIE for its compilation, but didn't get it quite right.  Try adding:
```
SECURITY_CFLAGS_append_pn-tini = " ${SECURITY_NOPIE_CFLAGS}"
```
to your `local.conf`. That should take care of the problem.

The test distro I use for my builds doesn't enable the security flags by default like poky does, so I didn't pick up on this before.

vjdaemp, 2020-04-09:
@madisongh Yes, that did the trick! Thanks

vjdaemp, 2020-04-10:
@madisongh  Also another note, I had to modify the instruction step to include TensorRT files in the NoDLA folder. Below is the line I used:

`sudo mv tensorrt*.deb libnvinfer*.deb libnvonnxparsers*.deb libnvparsers*.deb NoDLA/`

madisongh, 2020-04-10:
That is documented elsewhere; I've added a link to the wiki page that mentions it.

madisongh, 2020-04-10:
I've pushed fb3edb1aeded98aa47743b7d3bf17560d1a0fd57 to resolve the tini SECURITY_CFLAGS issue.

closed 2020-05-07 by madisongh