OE4T/meta-tegra #294: cannot find /dev/v4l2

pixellon, 2020-04-14:
Hi, 

I am currently using zeus branch l4t on a jetson tx2. I was able to successfully build core-image-weston. However when i try to run the on board camera capture application i do not see a v4l2 device in /dev/v4l2 folder. I dont know if this is a known issue. I would appreciate any help related to this.

Regards,
-Ash.

madisongh, 2020-04-14:
I see `/dev/v4l` on my TX2, but not `/dev/v4l2`.  Do you have `/dev/video0`?  Also check that the Argus daemon is running: `systemctl status nvargus-daemon`.

Also, I'd recommend using the `zeus-l4t-r32.3.1` for the latest L4T BSP content.  I've found it to be more reliable, particularly for weston.

pixellon, 2020-04-14:
Thanks for your suggestion to use `zeus-l4t-r32.3.1` . I can confirm I am using `zeus-l4t-r32.3.1` branch, However i do not see '/dev/v4l' and '/dev/video0'. the other thing I am observing is that the default build is using systemV and not systemd. do I have to install systemd ?

madisongh, 2020-04-15:
You can use sysvinit instead of systemd if you prefer (although I test more often with systemd than with sysvinit).  Definitely check that `nvargus-daemon` is running (e.g. with `ps -elf | grep nvargus-daemon`).  If it's not, try starting it manually (`/etc/init.d/nvargus-daemon restart`).  Also use `lsmod` to see if the `ov5693` driver has been loaded.

If the driver isn't present at all in your image, try adding `kernel-modules` (or at least `kernel-module-ov5693`) to CORE_IMAGE_EXTRA_INSTALL - the default build settings don't include all the kernel modules.  If nvargus-daemon isn't present, then add, say, `nvgstcapture`, or `gst-play` - something that will pull in the user-land camera support.

pixellon, 2020-04-15:
@madisongh many thanks for your suggestion after installing the modules i am now able to see the dev/video and dev/v4l2.

closed 2020-04-15 by pixellon

madisongh, 2020-04-15:
I could make it a bit more automatic... I noticed that for the machine config for the TX1, I have a reference to the kernel module for the camera.   I should probably do the same for the TX2.

pixellon, 2020-04-15:
I will save you the trouble. I have create a PR including the kernel-module-ov5693 for the TX2 machine configuration.

cg3dland, 2020-09-10:
I am working on dunfell-l4t-r32.4.3 on Jetson Nano now, I build the core-image-sato successfully, but I could not find '/dev/v4l' and '/dev/video0' even if the line 'CORE_IMAGE_EXTRA_INSTALL += " kernel-modules"' is added in local.conf, the nvargus-daemon runs on my Nano.

Anything do I miss?

pixellon, 2020-09-10:
kernel-module-ov5693 was missing in my case. Please check if you have this module installed.

cg3dland, 2020-09-10:
Any command to do this check?

pixellon, 2020-09-10:
In your machine/jetson-nano.conf check if you have 

MACHINE_EXTRA_RRECOMMENDS += "kernel-module-ov5693"

I had to include the above rebuild the image reflash and everything worked fine. I am not sure if the sensor module on tx2 is the same as the one on nano though.

madisongh, 2020-09-10:
@cg3dland Do you have a camera attached to your Nano?  You won't see `/dev/video0` show up unless you have a camera attached, with a driver for it loaded, and with appropriate device tree entries (at least for MIPI-CSI cameras).  If you haven't already, you should consult the [L4T documentation](https://docs.nvidia.com/jetson/l4t/index.html) on camera development to get started.

cg3dland, 2020-09-12:
@pixellon @madisongh thank you very much! the problem is resolved after I attahed a new USB camera to Nano instead of my old camera, it seem there is something mismatch with my old camera.

A bit confusion, with the new USB camera attached, the devices '/dev/v4l' and '/dev/video0' could be found if the core-image-base built with the line 'CORE_IMAGE_EXTRA_INSTALL += " kernel-modules"' in local.conf, but the devices '/dev/v4l' and '/dev/video0' could NOT be found if the core-image-base built with the line 'CORE_IMAGE_EXTRA_INSTALL += " kernel-module-ov5693"' in local.conf.

pixellon, 2020-09-12:
Kernel module kernel-module-ov5693 is specific to the omnivision sensor that comes as a part of jetson tx2. Usb camera modules seems to me are built as a part or general kernel modules kernel-modules. Both are different kernel-module-ov5693 is for the MIPI csi sensor.