OE4T/meta-tegra #295: tensorrt-staticdev and DEPENDS

deblauwetom, 2020-04-15:
Hello,

I want to use the static version of libnvinfer and added a line to my application package:
```
DEPENDS += "tensorrt-staticdev"
```
But then I got an error that it only "RPROVIDES" tensorrt-staticdev. So how come? Do I need to use "RDEPENDS" ? that seems odd because that means runtime dependency I think.

madisongh, 2020-04-15:
If it's a build-time dependency, you only need to use the recipe name, so just add a dependency on `tensorrt`.  That will make all of the headers and libraries (including static ones) available in your recipe's sysroot.

deblauwetom, 2020-04-15:
Ok thanks. But then in the final image to run on my nano device, I don't need the .so files included. I'm trying to link statically to reduce my image size. So how do I do that then?

madisongh, 2020-04-15:
The build should automatically pick up on which shared libraries your application links against and automatically add runtime dependencies on the package(s) that supply those libraries.  If you're statically linking, the shared libraries won't be brought in.  Just be sure you don't have explicit RDEPENDS settings somewhere to pull them in (or list them in CORE_IMAGE_EXTRA_INSTALL or some other variable that would do the same).

deblauwetom, 2020-04-29:
Hi, I finally had some more time and indeed that works, thanks!

Now it seems that TensorRT static is precompiled with  _GLIBCXX_USE_CXX11_ABI=0 and the rest of the system uses default =1. Do you happen to know how to set =0 as the default value for the whole system?

madisongh, 2020-04-29:
I don't know how to make it the default for the toolchain build (and I'm not sure what impact that would have on building everything else in the system if you did make that change), but for anything you're linking against that library, I think it should just be a matter of adding `-D_GLIBCXX_USE_CXX11_ABI=0` to the compiler flags (typically CXXFLAGS) - either in your recipe or in your make/cmake/etc. files.

deblauwetom, 2020-05-05:
Hi, I tried this and finally it worked, but the static library linking with tensorrt is as big as using the shared libs. I thought maybe the size would be less, but it isn't. It seems in the next jetpack 4.4 it would be solved that they split up libnvinfer and cudnn into smaller libraries(infer and training separated)

closed 2020-05-05 by deblauwetom