OE4T/meta-tegra #298: CUDA programs don't build with the SDK

eduardcazacu-sorama, 2020-04-16:
Hello!

We have successfully built an SDK using the latest zeus-l4t branch and managed to compile a simple non-cuda program. 

However, when trying to compile a CUDA program using cmake, we get the following error:

-- Check for working CUDA compiler: /workdir/sdk-env/sysroots/x86_64-pokysdk-linux/usr/local/cuda-10.0/bin/nvcc
-- Check for working CUDA compiler: /workdir/sdk-env/sysroots/x86_64-pokysdk-linux/usr/local/cuda-10.0/bin/nvcc -- broken
CMake Error at /workdir/sdk-env/sysroots/x86_64-pokysdk-linux/usr/share/cmake-3.15/Modules/CMakeTestCUDACompiler.cmake:46 (message):
  The CUDA compiler

 "/workdir/sdk-env/sysroots/x86_64-pokysdk-linux/usr/local/cuda-10.0/bin/nvcc"

  is not able to compile a simple test program.

with [CMakeError.log](https://github.com/madisongh/meta-tegra/files/4487176/CMakeError.log)

When checking, it seems like the CUDAHOSTCXX environment variable is not set. CUDAFLAGS are correctly set and so is CXX. 

Is it possible that nvcc is not correctly configured to use c++ compiler provided by the SDK?

We are using Linaro and we have added cuda-cudart to the nativesdk-packagegroup-cuda-sdk-host in addition to the cuda-compiler.

madisongh, 2020-04-16:
The environment-setup script for the SDK *should* be setting CUDAHOSTCXX if `cuda-target-environment` got included in your SDK... which it should if `cuda-cudart-dev` is in your SDK.  Make sure you've at least included `cuda-cudart-dev` (`cuda-toolkit-dev` would be more complete) in the target-side content.

eduardcazacu-sorama, 2020-04-16:
Thank you!
I only had the cuda-toolkit-dev added to the host side, adding it to target as well fixed it.

closed 2020-04-16 by eduardcazacu-sorama