OE4T/meta-tegra #299: Enable SPI with a patch

JW94, 2020-04-21:
Hey guys,

I want to enable SPI-support for the Jetson Nano Developer Kit p3449 0000 a02. I started to create a new patch for the linux-tegra recipe. I add the diff to this issue that somebody can understand and help me. 

As far as I know, I need to edit the pinmux configuration of the SPI-Pins but I cannot find any source which tells we what pins have to be edited. The only information I have is this commit https://github.com/rt-net/JetsonNano_DT_SPI/commit/0124b15ea6e29a0a45246223b7c892aa39e20678

Thank you in advance!

```diff --git a/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-porg-pinmux-p3448-0000-a02.dtsi b/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-porg-pinmux-p3448-0000-a02.dtsi
index 6bb2c60..0c87050 100644
--- a/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-porg-pinmux-p3448-0000-a02.dtsi
+++ b/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-porg-pinmux-p3448-0000-a02.dtsi
@@ -1220,7 +1220,7 @@
 
 			spi1_mosi_pc0 {
 				nvidia,pins = "spi1_mosi_pc0";
-				nvidia,function = "rsvd1";
+				nvidia,function = "spi1";
 				nvidia,pull = <TEGRA_PIN_PULL_DOWN>;
 				nvidia,tristate = <TEGRA_PIN_ENABLE>;
 				nvidia,enable-input = <TEGRA_PIN_DISABLE>;
@@ -1228,7 +1228,7 @@
 
 			spi1_miso_pc1 {
 				nvidia,pins = "spi1_miso_pc1";
-				nvidia,function = "rsvd1";
+				nvidia,function = "spi1";
 				nvidia,pull = <TEGRA_PIN_PULL_DOWN>;
 				nvidia,tristate = <TEGRA_PIN_ENABLE>;
 				nvidia,enable-input = <TEGRA_PIN_DISABLE>;
@@ -1236,7 +1236,7 @@
 
 			spi1_sck_pc2 {
 				nvidia,pins = "spi1_sck_pc2";
-				nvidia,function = "rsvd1";
+				nvidia,function = "spi1";
 				nvidia,pull = <TEGRA_PIN_PULL_DOWN>;
 				nvidia,tristate = <TEGRA_PIN_ENABLE>;
 				nvidia,enable-input = <TEGRA_PIN_DISABLE>;
@@ -1244,7 +1244,7 @@
 
 			spi1_cs0_pc3 {
 				nvidia,pins = "spi1_cs0_pc3";
-				nvidia,function = "rsvd1";
+				nvidia,function = "spi1";
 				nvidia,pull = <TEGRA_PIN_PULL_DOWN>;
 				nvidia,tristate = <TEGRA_PIN_ENABLE>;
 				nvidia,enable-input = <TEGRA_PIN_DISABLE>;
@@ -1252,7 +1252,7 @@
 
 			spi1_cs1_pc4 {
 				nvidia,pins = "spi1_cs1_pc4";
-				nvidia,function = "rsvd1";
+				nvidia,function = "spi1";
 				nvidia,pull = <TEGRA_PIN_PULL_DOWN>;
 				nvidia,tristate = <TEGRA_PIN_ENABLE>;
 				nvidia,enable-input = <TEGRA_PIN_DISABLE>;
```

madisongh, 2020-04-21:
That looks close.  pc1 is an input pin for SPI, so you want to change its `nvidia,enable-input` property to be `TEGRA_PIN_ENABLE`.  You probably also want to disable tristate on all five pins.

One way to get the right settings is to download the pinmux spreadsheet for the Nano dev kit from the NV download center, load it into Excel on Windows, carefully reconfigure the pins you want to modify, and have the spreadsheet's macros generate the `-pinmux-` and `-gpio-` DTS files for you.

Another would be to use the `jetson-io` python scripts that NV started including in L4T 32.3.1.  You'd have to load up your Nano with the stock L4T image, run the scripts there, then look at the changes it makes to the device tree for your configuration choices and apply them back to your device tree source.  They document its use [here](https://docs.nvidia.com/jetson/l4t/Tegra%20Linux%20Driver%20Package%20Development%20Guide/hw_setup_jetson_io.html#wwconnect_header).  I haven't tried it myself, though.

JW94, 2020-04-21:
Thank you for your super fast response!
You're right, there were some misconfigurations and they dropped after using the spreadsheets you recommended. 

After generating a dts with the spreadsheet the spi is working properly!

I attach the patch file which I added to the bbappend in my custom meta-layer.

Thank you!

<details>
<summary> SPI Patch</summary>

```diff
From afe87c9bcb3cb7ea4a22ecd597c79a0d3e619261 Mon Sep 17 00:00:00 2001
From: Jonathan Weinert <jo.weinert@h2o-protec.de>
Date: Tue, 21 Apr 2020 15:15:17 +0000
Subject: [PATCH] add custom dts with spi support

---
 ...egra210-jetson_nano_module_hs-gpio-default.dtsi |   77 ++
 .../tegra210-jetson_nano_module_hs-pinmux.dtsi     | 1347 ++++++++++++++++++++
 .../tegra210-p3448-0000-p3449-0000-a02.dts         |    4 +-
 3 files changed, 1426 insertions(+), 2 deletions(-)
 create mode 100644 nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-jetson_nano_module_hs-gpio-default.dtsi
 create mode 100644 nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-jetson_nano_module_hs-pinmux.dtsi

diff --git a/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-jetson_nano_module_hs-gpio-default.dtsi b/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-jetson_nano_module_hs-gpio-default.dtsi
new file mode 100644
index 0000000..8f57388
--- /dev/null
+++ b/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-jetson_nano_module_hs-gpio-default.dtsi
@@ -0,0 +1,77 @@
+/*This dtsi file was generated by jetson_nano_module.xlsm Revision: 1,01 */
+/*
+ * Copyright (c) 2019, NVIDIA CORPORATION.  All rights reserved.
+ *
+ * This program is free software; you can redistribute it and/or modify
+ * it under the terms of the GNU General Public License as published by
+ * the Free Software Foundation; either version 2 of the License, or
+ * (at your option) any later version.
+ *
+ * This program is distributed in the hope that it will be useful, but WITHOUT
+ * ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or
+ * FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License for
+ * more details.
+ *
+ * You should have received a copy of the GNU General Public License along
+ * with this program; if not, write to the Free Software Foundation, Inc.,
+ * 51 Franklin Street, Fifth Floor, Boston, MA  02110-1301, USA.
+ */
+
+#include <dt-bindings/gpio/tegra-gpio.h>
+
+/ {
+	gpio: gpio@6000d000 {
+		gpio-init-names = "default";
+		gpio-init-0 = <&gpio_default>;
+
+		gpio_default: default {
+			gpio-input = <
+				TEGRA_GPIO(BB, 0)
+				TEGRA_GPIO(B, 4)
+				TEGRA_GPIO(B, 5)
+				TEGRA_GPIO(B, 6)
+				TEGRA_GPIO(B, 7)
+				TEGRA_GPIO(DD, 0)
+				TEGRA_GPIO(E, 6)
+				TEGRA_GPIO(S, 5)
+				TEGRA_GPIO(A, 5)
+				TEGRA_GPIO(X, 4)
[... 1385 lines cut ...]
+			};
+		};
+
+		drive_default: drive {
+		};
+	};
+};
diff --git a/nvidia/platform/t210/porg/kernel-dts/tegra210-p3448-0000-p3449-0000-a02.dts b/nvidia/platform/t210/porg/kernel-dts/tegra210-p3448-0000-p3449-0000-a02.dts
index 4ab61c9..9786b73 100644
--- a/nvidia/platform/t210/porg/kernel-dts/tegra210-p3448-0000-p3449-0000-a02.dts
+++ b/nvidia/platform/t210/porg/kernel-dts/tegra210-p3448-0000-p3449-0000-a02.dts
@@ -19,8 +19,8 @@
  */
 
 #include "tegra210-porg-p3448-common.dtsi"
-#include "porg-platforms/tegra210-porg-pinmux-p3448-0000-a02.dtsi"
-#include "porg-platforms/tegra210-porg-gpio-p3448-0000-a02.dtsi"
+#include "porg-platforms/tegra210-jetson_nano_module_hs-pinmux.dtsi"
+#include "porg-platforms/tegra210-jetson_nano_module_hs-gpio-default.dtsi"
 
 / {
 	model = "NVIDIA Jetson Nano Developer Kit";
```

</details>

closed 2020-05-07 by madisongh