OE4T/meta-tegra #300: ERROR: libvisionworks rdepends on cuda-cudart-dev [dev-deps]

paroque28, 2020-04-22:
ERROR: libvisionworks-1.6.0.500n-r0 do_package_qa: QA Issue: libvisionworks rdepends on cuda-cudart-dev [dev-deps]
ERROR: libvisionworks-1.6.0.500n-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: Logfile of failure stored in: ..../build/tmp-glibc/work/armv8a_tegra210-wrs-linux/libvisionworks/1.6.0.500n-r0/temp/log.do_package_qa.30017
ERROR: Task (..../layers/meta-tegra/recipes-devtools/visionworks/libvisionworks_1.6.0.500n.bb:do_package_qa) failed with exit code '1'

madisongh, 2020-04-22:
Branch?

paroque28, 2020-04-22:
Hi Matt ,

This was on `zeus-l4t-r32.3.1`
```
IMAGE_INSTALL_append = "docker nvidia-docker nvidia-container-runtime cudnn tensorrt libvisionworks libvisionworks-sfm libvisionworks-tracking cudnn-container-csv tensorrt-container-csv libvisionworks-container-csv libvisionworks-sfm-container-csv libvisionworks-tracking-container-csv"
MACHINE='jetson-nano-qspi-sd'
```

madisongh, 2020-04-22:
Based on the shared library info, libvisionworks should be depending on cuda-cudart (provider of libcudart.so.10.0), not cuda-cudart-dev.  Just ran a quick build myself, adding libvisionworks, and the dependency was picked up correctly.

paroque28, 2020-04-22:
I will look into it thanks

madisongh, 2020-11-08:
@paroque28 did you figure out what was triggering this?

ichergui, 2021-04-06:
Hey @paroque28

paroque28, 2021-04-06:
@ichergui  sorry, I lost the environment where I was reproducing this issue.
Let me try to replicate it. Sorry about it

ichergui, 2021-04-06:
@paroque28 No worries.

ichergui, 2021-04-11:
Hey @paroque28 
I have done the same test as you, it works fine to me.
Tests are done by using `zeus-l4t-r32.3.1` branch of `meta-tegra` and `Jetson TX2` as a machine.
Hope it will work for you too

paroque28, 2021-04-15:
Hi @ichergui,
As you say, I have done the same test and it works fine to me

closed 2021-04-15 by paroque28