OE4T/meta-tegra #302: BUP package signing l4t-r32.3.1

ichergui, 2020-04-24:
Hi @madisongh 

This is not an issue but a question regarding BUP packages signing with redundant boot feature.

I'm testing the test-distro build for the zeus-mender-l4t-r32.3.1 branch on a Jetson TX2.

Is there any possibilty to sign the BUP packages after finishing the build ? 

For you infmation, **TEGRA_SIGNING_ARGS** variable is not define in **conf/local.conf**

Actually, I have 2 Jetson TX2, I wish flash them by using different keys (**SBK** + **PKC**), I mean sign the same BUP packge twice with different keys.

Can I use the digsigserver for that ? 

Any help would be appreciated. Please let me know if you need some more information in order to proceed.

Thanks, 
Ilies

lfdmn, 2020-04-24:
Maybe this help https://github.com/madisongh/meta-tegra/wiki/Secure-Boot-Support

ichergui, 2020-04-24:
Thank you for sharing the link, I've already take a look on that link.
I confirm the manual signing is working fine but following command is not working for BUP package.
```bash
$ ./doflash.sh -u /path/to/signing-key.pem -v /path/to-encryption-key
```
I wish perform the signature via the above command for the BUP package.

madisongh, 2020-04-24:
The flash helper script that doflash.sh calls on does have a `--bup`  option for BUP generation, so it is possible to do.  It just gets a bit complicated... to fully generate a BUP payload, you have to call on the helper script multiple times with different environment variable settings to handle the different SoC variants, then accumulate the results.

I don't currently have a handy wrapper script to do this manually at the moment, but it should be possible to create one.

madisongh, 2020-04-25:
>Actually, I have 2 Jetson TX2, I wish flash them by using different keys (SBK + PKC), I mean sign the same BUP packge twice with different keys.
>
>Can I use the digsigserver for that ?

Sure.  The easiest way is to have separate MACHINE configs for the two; digsigserver by default looks up keys based on the machine name.  That way you would ensure that the BUP payload has the right signature for each (at the expense of having different images for the two largely-identical machines).  That would also be safest, since the nv_update_engine tool has no way to check that the signatures are correct - it just trusts that you've delivered a correctly-signed payload.

ichergui, 2020-04-25:
> The flash helper script that doflash.sh calls on does have a `--bup` option for BUP generation, so it is possible to do. It just gets a bit complicated... to fully generate a BUP payload, you have [...]

Hi @madisongh 

I tried by using the tegraflash packge then perform the helper script **tegra186-flash-helper.sh** ,I got issues.

Here is the steps that I took:

```bash
$ unzip jetson-tx2.tegraflash.zip -d tests_bup_signing
$ cd tests_bup_signing
$ MACHINE="jetson-tx2" FAB="B00" BOARDSKU="" BOARDREV="" ./tegra186-flash-helper.sh --bup -u /home/ilieschergui/projects/test_keys/my_privkey.pem -v /home/ilieschergui/projects/test_keys/my_sbk.key flash.xml.in tegra186-quill-p3310-1000-c03-00-base.dtb jetson-tx2-cboot.cfg 0x1090000 tegra-minimal-initramfs-jetson-tx2-cboot.cboot
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands
 
[   0.0017 ] Generating RCM messages
[   0.0025 ] tegrarcm_v2 --listrcm rcm_list.xml --chip 0x18 0 --download rcm mb1_recovery_prod.bin 0 0
[   0.0033 ] RCM 0 is saved as rcm_0.rcm
[   0.0038 ] RCM 1 is saved as rcm_1.rcm
[   0.0039 ] List of rcm files are saved in rcm_list.xml
[   0.0039 ] 
[   0.0039 ] Signing RCM messages
[   0.0046 ] tegrasign_v2 --key None --list rcm_list.xml --pubkeyhash pub_key.key
[   0.0054 ] Assuming zero filled SBK key
[   0.0092 ] 
[   0.0092 ] Copying signature to RCM mesages
[   0.0099 ] tegrarcm_v2 --chip 0x18 0 --updatesig rcm_list_signed.xml
[   0.0109 ] 
[   0.0109 ] Boot Rom communication
[   0.0117 ] tegrarcm_v2 --chip 0x18 0 --rcm rcm_list_signed.xml
[   0.0124 ] BootRom is not running
[   0.0127 ] 
[   1.0150 ] tegrarcm_v2 --isapplet
[   1.0157 ] USB communication failed.Check if device is in recovery
[   1.0158 ] 
[   1.0166 ] tegradevflash_v2 --iscpubl
[   1.0173 ] Cannot Open USB
[   1.0174 ] 
[   2.0221 ] tegrarcm_v2 --isapplet
[   2.0251 ] USB communication failed.Check if device is in recovery
```

Is that the right way to sign the BUP package ? if no, which package I should use to perform the BUP package signing ?

Thank you for your help.

ichergui, 2020-04-25:
> Sure. The easiest way is to have separate MACHINE configs for the two; digsigserver by default looks up keys based on the machine name. That way you would ensure that the BUP payload has the right [...]

Thank you @madisongh for your help.

My goal is to avoid the build for several time, I thought that I can use the tegraflash package then use differents keys (SBK and PKC) for the signing then finally,  generate the BUP packages ready for the flashing.

madisongh, 2020-04-26:
>Here is the steps that I took:

Looks like you forgot to set `BOARDID=3310` on the command. For the TX2, the helper script will try to read the device EEPROM if either BOARDID or FAB is undefined in the environment.

ichergui, 2020-04-26:
Thank you @madisongh for your help

I tried with the following command:

```bash
$ MACHINE="jetson-tx2" BOARDID="3310" FAB="B00" BOARDSKU="" BOARDREV="" ./tegra186-flash-helper.sh --bup -u /home/ilieschergui/projects/test_keys/my_privkey.pem -v /home/ilieschergui/projects/test_keys/my_sbk.key flash.xml.in tegra186-quill-p3310-1000-c03-00-base.dtb jetson-tx2.cfg 0x1090000 tegra-minimal-initramfs-jetson-tx2-cboot.cboot
```
I got the following issue 

```bash
signing images succeeded

Creating update payloads for Jetson-3310-B00-1000--1-0-jetson-tx2-mmcblk0p1 board ...
ls: cannot access 'multi_signed/*/tegra186*bpmp*_sigheader.dtb.encrypt.signed': No such file or directory
WARNING: Failed to find image for bpmp-fw-dtb
ls: cannot access 'multi_signed/*/tegra186-quill*_sigheader.dtb.encrypt.signed': No such file or directory
WARNING: Failed to find image for bootloader-dtb
ls: cannot access 'multi_signed/*/tegra186-quill*_sigheader.dtb.signed': No such file or directory
WARNING: Failed to find image for kernel-dtb
ls: cannot access 'multi_signed/*/mb1_cold_boot*.bct.encrypt.signed': No such file or directory
WARNING: Failed to find image for MB1_BCT
creating bl_and_kernel payload
```

**Note:**  I have copy manually rollback directory to this folder from **L4T_BSP/L4T-32.3.1-tegra186/Linux_for_Tegra/bootloader** 

Should we add the rollback directory in the tegraflash package ?

Attached is the file contains the full traces

[BUP_package_signing_traces.log](https://github.com/madisongh/meta-tegra/files/4536631/BUP_package_signing_traces.log)

madisongh, 2020-04-27:
Ah, right.  Sigh.  The zip file doesn't include the rollback config, which the BUP generator requires.  The other thing is that the `flash.xml.in` file needs some modification:
  - The <filename> entry for the APP partition should be deleted (it's not relevant for BUP payloads)
  - The <filename> entry for the kernel/kernel_b partitions *must* be called `boot.img` (and that means your u-boot or kernel+initrd file must also be called `boot.img`).

The `oe_make_bup_payload` function in image_types_tegra.bbclass performs the necessary hackery to make this work when performing the signing steps as part of the build.

When I get a chance, I'll look into making the tegraflash zip generator more compatible for running BUP generation.

ichergui, 2020-04-27:
Thank you @madisongh for your help.

I succeeded to have the BUP payloads.

Here is my steps:
1. Unzip tegraflash into a folder (e.g. bup_payload_gen)
2. Copy rollback from L4T_BSP r32.3.1 into bup_payload_gen directory
3. Apply the attached patch to update the flash.xml.in file
4. boot.img symlink to u-boot-jetson-tx2.bin
5. Call tegra186-flash-helper.sh multiple times with different environment variable settings to handle the different SoC variants.   

```bash
$ mkdir tegraflash_sign_bup
$ unzip test-jetson-tx2.tegraflash.zip -d ./tegraflash_sign_bup
$ cd tegraflash_sign_bup
$ cp -r /opt/nvidia/L4T-32.3.1-tegra186/Linux_for_Tegra/bootloader/rollback .
$ ln -s u-boot-jetson-tx2.bin boot.img
$ cp -v flash.xml.in flash.xml.in.orig
$ mv 0001-update-flash.xml.in-for-bup-payload.patch.log 0001-update-flash.xml.in-for-bup-payload.patch
$ patch -p0 < 0001-update-flash.xml.in-for-bup-payload.patch
$ mv tests_bup_gen.sh.log tests_bup_gen.sh
$ chmod +x tests_bup_gen.sh
$ ./tests_bup_gen.sh
```

Attached are:

The script [tests_bup_gen.sh.log](https://github.com/madisongh/meta-tegra/files/4540921/tests_bup_gen.sh.log)

The patch [0001-update-flash.xml.in-for-bup-payload.patch](https://github.com/madisongh/meta-tegra/files/4540901/0001-update-flash.xml.in-for-bup-payload.patch.log) to modify the **flash.xml.in**

The log file [tegraflash_sign_bup.log](https://github.com/madisongh/meta-tegra/files/4540903/tegraflash_sign_bup.log)

madisongh, 2020-04-28:
>How can I flash my device ?
Do I launch doflash.sh script with -u /path/to/rsa-priv.key -v /path/to/sbk.key ?

Yes, but use the tegraflash package without your BUP-related changes applied.

To use the BUP payload you generated above, copy the `bl_update_payload` file that got generated to `/opt/ota_package` on the device, and use `nv_update_engine` to install it.  (You have to include `tegra-redundant-boot` in your image).

madisongh, 2020-05-03:
For the L4T R32.4.2-based branches, I've added support for generating BUP payloads out of the tegraflash packages.  The changes are a bit too intrusive for retrofitting onto the R32.3.1-based branches at this point, I think.

closed 2020-05-03 by madisongh

ichergui, 2020-05-03:
Thank you very much @madisongh

PR #311 (2020-05-03): Update tegraflash packaging for BUP generation and other enhancements
This patch series updates how tegraflash packages are created to include the tools and files needed to create BUP payloads from the unpacked package, including a `generate_bup_payload.sh` script to automate the payload creation.

There are related changes to the flash helper scripts.

Other changes include:
* a fix for the tegra194 helper script to better handle a null BOARDEV setting for BUP payload generation
* Extending the recipes that generate BUP payloads to deploy the kernel-only, bl-only, and any other (platform-spec [... cut]
PR #323 (2020-05-24): Backport kernel recipe and tegraflash enhancements
Backports from master/dunfell:

* Backport kernel recipe change to using linux-yocto.inc (#317)
* Make the tegraflash package usable for BUP payload generation (fixes #302)
* Deploy all generated BUP payloads
* Enhancements for SDcard generation
  * Use bmaptool, if available
  * Reduce need for environment variables when invoking dosdcard.sh script
  * Fixes for helper scripts
* Add `tar.gz` as alternative format for tegraflash package