OE4T/meta-tegra #303: linux-tegra_4.9 compilation error - zeus

lfdmn, 2020-04-24:
Using GCC 7.3 I get 
```
cc1: error: -Werror=address-of-packed-member: no option -Waddress-of-packed-member
cc1: error: -Werror=missing-attributes: no option -Wmissing-attributes
```

I tried adding 
`TARGET_CFLAGS += " -Wno-error=address-of-packed-member -Wno-error=missing-attributes"`

and
`export KCFLAGS = "-Wno-error=address-of-packed-member -Wno-error=missing-attributes"`

but it didn't help

dwalkes, 2020-04-24:
The suggestion in https://github.com/madisongh/meta-tegra/wiki/Using-linaro-gcc7-for-CUDA-support might be helpful

madisongh, 2020-04-25:
@lfdmn have you customized your kernel at all - your own fork with added patches, modified kernel config, etc.?  I run all my test builds with gcc 7.3 and don't get these errors.

lfdmn, 2020-04-27:
@madisongh this is the vanilla zeus branch with jetson-xavier machine and my distro.

Somehow the KCFLAGS export doesn't work. I was looking at the meta-boot2qt distro layer and they also apply the same fix:
[https://code.qt.io/cgit/yocto/meta-boot2qt.git/tree/meta-boot2qt-distro/dynamic-layers/tegra/recipes-kernel/linux/linux-tegra_%25.bbappend?h=zeus](https://code.qt.io/cgit/yocto/meta-boot2qt.git/tree/meta-boot2qt-distro/dynamic-layers/tegra/recipes-kernel/linux/linux-tegra_%25.bbappend?h=zeus) 

@dwalkes thanks, I had looked at this article but it didn't help for kernel recipe. Also from the latest [yocto manual](https://www.yoctoproject.org/docs/latest/ref-manual/ref-manual.html) there a section in 4.16.8. Miscellaneous Changes.

> TARGET_CFLAGS, TARGET_CPPFLAGS, TARGET_CXXFLAGS, and TARGET_LDFLAGS are no longer exported to the external environment. This change did not require any changes to core recipes, which is a good indicator that no changes will be required. However, if for some reason the software being built by one of your recipes is expecting these variables to be set, then building the recipe will fail. In such cases, you must either export the variable or variables in the recipe or change the scripts so that exporting is not necessary.

madisongh, 2020-04-27:
Yep, kernel build environment is different from recipes for userland cross builds, so KCFLAGS should be the right variable to export for this.  As for that note in the docs, the TARGET_xxx variables don't get exported directly, but they are used to set the CFLAGS, CPPFLAGS, etc. variables that do get exported when you're cross-building for the target machine.

I just wish I knew where those errors were coming from, as I've never run into them.  Maybe one of the other layers in your distro setup is changing things to try to enable those warnings?  If you could shoot me the output from `bitbake -e virtual/kernel`, that would help in tracking this down.

lfdmn, 2020-04-27:
Here's the [environment](https://github.com/madisongh/meta-tegra/files/4541003/environment.txt). I've removed all the bbappends for linux-tegra, the setting KCFLAGS directly to linux-tegra_4.9.bb
`export KCFLAGS = "-Wno-error=address-of-packed-member -Wno-error=missing-attributes"`

I added some echo to do_configure_prepend and do_compile_prepend. At least KCFLAGS is set properly.

madisongh, 2020-04-27:
So did that help with the compilation errors?

lfdmn, 2020-04-28:
No it didn't.

madisongh, 2020-04-28:
Try removing that `export` while keeping your setup with all the bbappends removed.  If the compiler doesn't support those warnings, having those `-W` options in the compilation command will trigger those `no option` messages (misleadingly so, since the messages omit the `no-` prefix).

If that works, then I suspect the problem is the KCFLAGS setting in the bbappend coming from boot2qt's bbappend.  You should be able to work around it by adding your own bbappend with
```
KCFLAGS_remove = "-Wno-error=address-of-packed-member -Wno-error=missing-attributes"
```

lfdmn, 2020-05-04:
Oh yeah I misread the compile output and was trying to add those flags again instead of removing them...

Sorry about that and thank you =)

closed 2020-05-04 by lfdmn

madisongh, 2020-05-04:
You and me both :)