OE4T/meta-tegra #305: Tegraflash package is not generated

ichergui, 2020-04-25:
Hi @madisongh 

I'm testing the test-distro build for the zeus-mender-l4t-r32.3.1 branch on a Jetson TX2 (MACHINE=jetson-tx2-cboot).

Build the image is **OK** but I did not find the tegraflash package ***.tegraflash.zip** in the deploy directory.

Is this the right behavior ? or I missed something in the conf/local.conf file?

Here the steps to build the image 
```bash
git clone https://github.com/madisongh/test-distro.git
cd test-distro
git checkout zeus-mender-l4t-r32.3.1
git submodule update --init --recursive
source ./setup-env
bitbake core-image-base
```

My conf/local.conf:
```bash
MACHINE = "jetson-tx2-cboot"
DISTRO = "testdistro"
PACKAGE_CLASSES = "package_deb"
EXTRA_IMAGE_FEATURES = "package-management"
#
# GCC version and fortran
GCCVERSION = "7.%"
# Enabling fortran
FORTRAN_forcevariable = ",fortran"
RUNTIMETARGET_append_pn-gcc-runtime = " libquadmath"
# Specify Cuda version
CUDA_VERSION = "10.0"
#
# Mender artifact name
#
MENDER_ARTIFACT_NAME = "jetson-tx2-image-1.0"

NVIDIA_DEVNET_MIRROR = "file:///home/ichergui/devel/jetpack_nvidia/4.3/nvidia/sdkm_downloads"

require contrib/conf/include/gcc-compat.conf
```

Thank you for your help.

madisongh, 2020-04-26:
For that MACHINE, yes, that's correct.  I have it configured for testing my encrypted rootfs partition setup with Mender updates, which requires an installer image to prepare the eMMC partitions.

Build `tegra-sysinstall` to get a tegraflash package with the installer. When it boots, run `tegra-sysinstall <rootfs-tarball> <dataimg-tarball>`.  The two tarballs (which should come from a build you've done of a regular image) can either be local (scp them to the device beforehand), or fetchable via http.  If the installer is successful, it will automatically reboot into the installed image, and from that point forward you use Mender to do updates.

Note that for the partition encryption to work properly, you'll have to generate an EKS keyblob using the KEK2 encryption key that you have programmed into your TX2 and replace the keyblob in the distro (which was generated for my fused TX2).

ichergui, 2020-04-26:
Thanks @madisongh for this informations, it is very usefull

closed 2020-04-26 by ichergui