OE4T/meta-tegra #306: Support for Xavier NX module

nandra, 2020-04-27:
Hi,

just want to ask if there is some work planned on adding Xavier NX module to this repo? I can support with testing (maybe integrating) as I have module. Thanks.

madisongh, 2020-04-27:
I ordered a module which hopefully will be in this week.  Assuming it actually works in one of my Nano carriers, I should get started on it shortly thereafter.

nandra, 2020-04-27:
@madisongh if you need some help pls tell me (testing integrating ...). I can support.

madisongh, 2020-04-30:
@nandra I've pushed changes to master and dunfell-l4t-r32.4.2 to add a new machine: `jetson-xavier-nx-devkit-emmc`.  Some basic tests on my setup work OK, but I can't do too much testing right now. If you could give it a try, let me know how it works for you.

nandra, 2020-04-30:
@madisongh board should be here this week but unfortunately seems beginning of next week only. I'll build image and give a try. Thanks.

acostach, 2020-05-04:
Hi @madisongh , saw you added the xavier nx as: 'Nvidia Jetson Xavier NX development kit with QSPI+eMMC'

I see the word devkit so can you please tell me if the setup you have is this module here:
https://www.arrow.com/en/products/900-83668-0000-000/nvidia 

on top of the b01 nano carrier here: https://www.arrow.com/en/products/945-13450-0000-100/nvidia

Or maybe you are using a different setup? Thanks!

madisongh, 2020-05-04:
@acostach Yes, that's what I'm using in my setup.  As far as I can tell, that's the configuration NVIDIA is expecting for `jetson-xavier-nx-devkit`, based on what I see in the L4T kit and what I've read on their developer forums.

dremsol, 2020-05-05:
@madisongh, may I ask which heatsink you are using in your NX setup? I've a NX module for over a week already but didn't find an available heatsink yet.

madisongh, 2020-05-05:
@dremsol That's one reason why my testing has been limited - I didn't have one, either.  Today I attached a generic 40x40mm CPU heatsink, figuring something was better than nothing... I'm more interested in making sure things are functional than performance/stress testing at this point, anyway.  Hopefully something better will come along soon.

dremsol, 2020-05-05:
@madisongh that's what I figured. It sounds like a proper temporary solution, does it cover the complete SoC? Maybe i will retrofit some massive aluminium obsolete heatsinks we have in house to be able to test with you.

FYI; I just got information from ConnectTech that they will ship their thermal solutions (active, passive, thermal plate) at June earliest.

madisongh, 2020-05-06:
> does it cover the complete SoC?

It has full contact with the chip itself, but not everything on the surface of the entire module.

madisongh, 2020-05-07:
I ran through some more tests yesterday, getting the DeepStream 5.0 samples running in a Docker container running on the Xavier NX.  Looking OK, I think.  I do see a bunch of warnings showing up on the serial console at boot time and once in a while after that, which I think are emitted by the BPMP firmware, but I see the same when running the stock L4T Ubuntu image, so I don't think they're due to any porting issues.

nandra, 2020-05-11:
@madisongh I tried to build image today but I'm hitting issue (using dunfell-l4t-r32.4.2l branch):
```
ERROR: linux-tegra-4.9+gitAUTOINC+dc9f649724-r0 do_fetch: Fetcher failure: Unable to find revision dc9f64972468f4754ba6b3524449e4c0e7a58ced in branch patches-l4t-rl4t-r32.1 even from upstream
ERROR: linux-tegra-4.9+gitAUTOINC+dc9f649724-r0 do_fetch: Fetcher failure for URL: 'git://github.com/madisongh/linux-tegra-4.9;branch=patches-l4t-rl4t-r32.1'. Unable to fetch URL from any source.
```
Thanks.

madisongh, 2020-05-11:
That branch name doesn't look right.   It's set to `"patches-${L4T_VERSION}"`, and L4T_VERSION should be coming from `classes/l4t_bsp.bbclass` (and should be set to `32.4.2` there).  Make sure you aren't setting that variable somewhere else... the output from `bitbake -e linux-tegra` can help with that.

nandra, 2020-05-12:
@madisongh you're right then was in other layer override in linux-tegra_%.bbappend. Sorry for noise.

nandra, 2020-05-14:
@madisongh I've build core-image-minimal, flash it and board boots fine. I have also attached CPU fan (on top of cooler) and connect it to slot. Fan was not spinning but I think it only start when CPU temp start to be high? Or do I need to do anything in this regard? USB + ethernet works so I'm not sure which other tests should be performed. Thanks.

madisongh, 2020-05-14:
> I've build core-image-minimal, flash it and board boots fine.

Great!

> Fan was not spinning but I think it only start when CPU temp start to be high?

Yes, and the pwm-fan driver should emit messages when the temp crosses the thresholds for raising or lowering the fan speed.  You may need to tweak the settings in the device tree for your specific fan.

nandra, 2020-05-15:
@madisongh can I perform some more test (I mean interface specific) or others? Please let me know. If it's enough we can close it ;). Thanks.

madisongh, 2020-05-15:
We should be good for now, at least.  Thanks!

closed 2020-05-15 by madisongh