OE4T/meta-tegra #308: Support for PREEMPT_RT

dljsjr, 2020-04-28:
Hey there!

I'm pretty new to Yocto/OE in general so this may be a non issue and I may just be doing something wrong. But it would be really nice if the `meta-tegra` layer and the `linux-tegra` kernel were able to support the PREEMPT_RT kernel patches for real-time embedded applications that have stricter latency requirements than just the PREEMPT kernel config can offer.

It looks like there's rudimentary support (if not slightly buggy) from Nvidia for building L4T w/ the PREEMPT_RT patches applied: https://forums.developer.nvidia.com/t/preempt-rt-patch-on-jetson-kernel/65766/19

This might be something that we undertake ourselves and contribute back but I wanted to open this issue in case there was just something obvious I was missing.

The only thing I've tried so far is setting the provider for the kernel to the stock Yocto RT kernel but I didn't expect that to work since we want `linux-tegra` for the Jetson boards and not `linux-yocto-rt` (it didn't work).

If you could give me some pointers on where to start looking to get these patches applied we'd love to help contribute to the project! Thanks!

madisongh, 2020-04-28:
There's an `rt-patches` directory in the `linux-tegra-4.9` repo that has the NVIDIA-supplied realtime patches.  You could fork the kernel repo and apply them there, then add a bbappend to set the SRC_URI to point to your fork/branch.  Then you'd have to generate a new defconfig for the kernel build that enables and disables the relevant configuration settings (see the `scripts/rt-patch.sh` script in the kernel repo).

I just tried a `git am` on the patches (on the `patches-l4t-r32.3.1` branch), and they all patch cleanly except for one (due to a gcc 8 compatibility patch I added at c88fb43493f42f41742ced88b1bdcf27aba5de55).  It's straightforward to resolve manually if you use `git am --reject`.

dljsjr, 2020-04-28:
Thanks for the tips! I'll keep you posted.

madisongh, 2020-11-08:
We have a [wiki page](https://github.com/OE4T/meta-tegra/wiki/Applying-PREEMPT-RT-Patches-dunfell-l4t-r32.4.3) describing how to apply the the PREEMPT_RT patches.

closed 2020-11-08 by madisongh