OE4T/meta-tegra #310: Branch thud-l4t-r32.3.1 does not build against meta-oe thud branch

wlindauer, 2020-04-30:
When building on the thud-l4t-r32.3.1 branch and including meta-oe you end up getting these parse errors:

ERROR: No recipes available for:
  poky/meta-tegra/external/openembedded-layer/recipes-support/libusbgx/libusbgx-config.bbappend
  poky/meta-tegra/external/openembedded-layer/recipes-support/libusbgx/libusbgx_%.bbappend
  poky/meta-tegra/external/openembedded-layer/recipes-support/opencv/opencv_4.1.0.bbappend

It appears that this commit "Backport L4T R32.3.1 from Zeus to Thud" (https://github.com/madisongh/meta-tegra/commit/8e6197fd414d33ac03f2e555c66385e811f9e089) added three bbappend files to recipes that do not exist in meta-oe on the thud branch.

Any guidance would be appreciated.

madisongh, 2020-04-30:
Drat, I should have checked on that before merging that backport.  I've pushed some changes to the thud-l4t-r32.3.1 branch that should make everything buildable again if you've got meta-oe in your layers.

wlindauer, 2020-05-01:
This is exactly what I was looking for. Thank you for being so responsive.

closed 2020-05-01 by wlindauer