OE4T/meta-tegra #312: u-boot env in sdcard partition

nandra, 2020-05-04:
Hi,
this is not an issue just more request for some help. For OTA update system I need to handle u-boot envs (board jetson nano SDcard boot). AFAIK when checked u-boot code there is no env support enabled at all (I'm using warrior branch from meta-tegra). I can add it but as I'm newcomer in Nvidia world I have no idea how to add extra partition (just raw no filesystem necessary). Any help appreciated. Thanks.

dwalkes, 2020-05-04:
> AFAIK when checked u-boot code there is no env support enabled at all (I'm using warrior branch from meta-tegra)

There are some examples in https://github.com/mendersoftware/meta-mender-community/tree/zeus-test/meta-mender-tegra/recipes-bsp/u-boot/patches for dealing with u-boot environment on Tegra platforms.  This might be a good place to start if you haven't looked at these already.  See https://github.com/mendersoftware/meta-mender-community/blob/zeus-test/meta-mender-tegra/recipes-bsp/u-boot/patches/0001-Move-jetson-nano-env-to-SDcard.patch for instance.

You may find the [meta-mender-community zeus-test branch](https://github.com/mendersoftware/meta-mender-community/tree/zeus-test/meta-mender-tegra) useful to see an example implementation for mender which supports u-boot environment variables on Tegra.  There are some options implemented there to locate uboot environment in spi flash or MMC on nano platforms using `TEGRA_MENDER_UBOOT_ENV_IN_SPIFLASH` which may be useful as well.

nandra, 2020-05-04:
@dwalkes thanks a lot. Yes I got some inspiration from mender patches and have a look at them. Bit of magic for me is still to understand how can I add dummy raw partition for env storage. I saw some xml files but I'm not sure if they are generated by some tool or manually? Thanks.

dwalkes, 2020-05-04:
@nandra sure, glad to help

> how can I add dummy raw partition for env storage

In the mender case I don't add a dummy partition to the xml file.  I just use mmcblk0boot1 space located in the [empty space before the secondary_gpt](https://github.com/mendersoftware/meta-mender-community/blob/f9e465c05b293715526ad721f702943d160c90e1/meta-mender-tegra/recipes-bsp/tegra-binaries/mender-custom-flash-layout/tegra186/flash_mender.xml#L189).  For the history of how the offset was selected you can see [this post on the nvidia forum](https://forums.developer.nvidia.com/t/mmcblk0boot1-usage-at-address-4177408-and-u-boot-parameter-storage-space-availability/82113) ultimately [reflected in the setting of MENDER_UBOOT_ENV_STORAGE_DEVICE_OFFSET](https://github.com/mendersoftware/meta-mender-community/blob/f9e465c05b293715526ad721f702943d160c90e1/meta-mender-tegra/recipes-bsp/u-boot/u-boot-mender-tegra.inc#L16).

> I saw some xml files but I'm not sure if they are generated by some tool or manually?

The xml files in meta-mender-community are manually created/updated today based on values provided in NVIDIA source for a given L4T release (see some early discussion about this [here](https://github.com/madisongh/meta-tegra/pull/114#issuecomment-435681615)), but then sizes for images and other parts are filled in automatically to correspond with yocto build parameters by the logic in meta-tegra, see examples related to [APPSIZE](https://github.com/mendersoftware/meta-mender-community/blob/f9e465c05b293715526ad721f702943d160c90e1/meta-mender-tegra/recipes-bsp/tegra-binaries/mender-custom-flash-layout/tegra186/flash_mender.xml#L222) for instance which you can find updated [here](https://github.com/madisongh/meta-tegra/blob/882d53169cd260f0131cccbbce2bfda5ad2748aa/classes/image_types_tegra.bbclass#L130).

madisongh, 2020-05-04:
>  I just use mmcblk0boot1 space located in the empty space before the secondary_gpt. 

Unfortunately, SDcards don't have the bootX partitions that eMMCs have, so creating a custom XML partition layout file is the only choice, if you want the U-Boot environment on the SDcard.

The warrior branch doesn't include any U-Boot environment support on the Nano, since there was no support for it in L4T R32.1.  IIRC, NVIDIA added support for that in R32.3.1, so the branches based off that version of L4T will have it by default.  I also added it to the R32.2.x-based U-Boot fork we use in the zeus branch... although the specific offset into the flash may be different than in R32.3.1+, since my implementation pre-dates NVIDIA's.

If using the QSPI flash is that the environment is present regardless of whether you swap SDcards. That might be a good thing or a bad thing, depending on what you need.  As @dwalkes pointed out, the meta-mender-tegra layer has some patches that switch the storage over to SDcard, and adding an extra entry in the partition layout XML file isn't too hard to do.

nandra, 2020-05-04:
@madisongh @dwalkes thanks for valuable info. I stole some bits and pieces from mender integration (I'm adding swupdate support). Now u-boot env is working fine. Again thanks a lot.

closed 2020-05-05 by nandra