OE4T/meta-tegra #313: Mender system update failed

ichergui, 2020-05-05:
Hi @madisongh ,

I'm testing the test-distro build for the zeus-mender-l4t-r32.3.1 branch on a Jetson TX2. When running the mender install I got the following error:

```bash
$ mender -install core-image-minimal-jetson-tx2.mender -log-file /data/system_updates_traces.log
```

**Traces:**

```bash
$ tail -n7 /data/system_updates_traces.log
time="2020-05-05T11:20:12Z" level=info msg="installer: authenticated digital signature of artifact" module=installer
time="2020-05-05T11:20:12Z" level=info msg="native sector size of block device /dev/mmcblk0p33 is 512, we will write in chunks of 1048576" module=dual_rootfs_device
time="2020-05-05T11:20:12Z" level=info msg="opening device /dev/mmcblk0p33 for writing" module=block_device
time="2020-05-05T11:20:12Z" level=info msg="partition /dev/mmcblk0p33 size: 15034482688" module=block_device
time="2020-05-05T11:24:20Z" level=info msg="wrote 14282760192/14282760192 bytes of update to device /dev/mmcblk0p33" module=dual_rootfs_device
time="2020-05-05T11:24:22Z" level=info msg="Enabling partition with new image installed to be a boot candidate: 33" module=dual_rootfs_device
time="2020-05-05T11:24:24Z" level=error msg="stderr collected while running script /var/lib/mender/scripts/ArtifactInstall_Leave_80_bl-update [cp: '/etc/machine-id' and '/tmp/nvbup.vl4AR3/etc/machine-id' are the same file\n]" module=executor

```

Do you have any suggestions on how to proceed with this?

Best regards,
Ilies.

madisongh, 2020-05-05:
From the error output, it looks like it's running the wrong script - it's attempting to copy the `/etc/machine-id` file from the current rootfs to the new one, but that should only happen if using cboot as bootloader.

That said, I haven't actually tested much on that branch for a while, other than some build testing.  I just tried a couple of runs today (jetson-tx2 and jetson-tx2-cboot) and noticed that I needed to port some mender-client patches from 2.1 to 2.2.  I've done that now, successfully did manual mender updates on both platforms, and pushed the changes to test-distro.

Note that there is an issue with the way the mender state scripts work in Yocto builds that can cause stale scripts to get deployed after changing build settings.  Try `rm -rf $BUILDDIR/tmp/deploy/images/jetson-tx2/mender-state-scripts` and rebuilding your image.  That could be the source of this particular problem.

ichergui, 2020-05-05:
Will do.
I will let you know.
Thank you very much @madisongh

ichergui, 2020-05-07:
I have done tests, it works fine
Thank you Matt

closed 2020-05-07 by ichergui