OE4T/meta-tegra #314: [jetson-nano] Add uvcvideo.ko module to image

elPrac, 2020-05-11:
Hi!

I was trying to install uvcvideo.ko module to my image using meta-tegra/.../jetson-nano.conf as $MACHINE but in order to do that i need to add it in the module list:
```
MACHINE_EXTRA_RDEPENDS += "kernel-module-uvcvideo"
```
Problem is that i want to keep meta-tegra intact and i want to add this in my own meta-layer
but i'm not sure how to do it i already tried to create my own machine recipe with nothing but
```
require conf/machine/jetson-nano.conf
COMPATIBLE_MACHINE = "my-machine"
```
but i got build errors, also i tried to add the module via conf/local.conf but because jetson-nano.conf set MACHINE_EXTRA_RDEPENDS with "=" my changes are overwritten. Also, i tried [these](https://www.yoctoproject.org/pipermail/yocto/2017-October/038448.html) suggestions but they didn't worked for me.

Can someone point me to the right direction on this please? Any clue or suggestion will be appreciated it
Also, this module needs to be added if someone is trying to use pyrealsense2 of [meta-intel-realsense](https://github.com/IntelRealSense/meta-intel-realsense)  otherwise it will complain about "NO device connected" I could add/test a recipe for this if @madisongh find it useful :+1:

madisongh, 2020-05-11:
If all you need to do is to add the kernel module to your own builds, then this line your `local.conf` file should do that:
```
MACHINE_EXTRA_RRECOMMENDS_append_jetson-nano = " kernel-module-uvcvideo"
```
(substitute the actual MACHINE name for `jetson-nano`, if you're using a different one)

More generally, to couple the inclusion of that module with  librealsense2, you could add this line either to the librealsense2 recipe, or to a bbappend for librealsense2 in your layer:
```
RRECOMMENDS_${PN} += "kernel-module-uvcvideo"
```
RRECOMMENDS is used instead of RDEPENDS so that the build doesn't fail if the module in question is compiled into the kernel (instead of as a separate module).

elPrac, 2020-05-11:
Thanks! @madisongh 

I tried with the .append recipe and it worked i also added these [patches](https://github.com/IntelRealSense/meta-intel-realsense/tree/pyro/recipes-kernel/linux)

Its a very simple recipe but not sure if you want to add it, i can deliver a pull request i'm using zeus branch.
If not, i can just close the issue
Thanks again!

madisongh, 2020-05-11:
Glad to hear it worked... there is already a [layer](http://layers.openembedded.org/layerindex/branch/master/layer/meta-intel-realsense/) for integrating librealsense into a build... you might want to submit a PR there for the RRECOMMENDS change - it would probably be useful to any platform using that camera.

closed 2020-05-11 by elPrac