OE4T/meta-tegra #318: Jetson NANO dev

outbackdingo, 2020-05-20:
previously there was an sdcard image we could dd to the card, now i updated and built but i dont see the sdcard image, how are we flashing now ?

madisongh, 2020-05-20:
If this is the first time you are updating your Nano with a build off the L4T R32.3.1 (or later) BSP, you should insert an SDcard in the Nano and flash via USB. For just updating the SDcard partitions, there's a `dosdcard.sh` script in the tegraflash zip package that can be used to create an SDcard image file or write directly to an SDcard inserted in your development host.  That script will, by default, still try to contact the Nano via USB to determine what version it is.  If you know which version of Nano you have, you can set the environment variables BOARDID and FAB to override this check.

The [wiki page](https://github.com/madisongh/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit) mentions this change.

outbackdingo, 2020-05-20:
ok still confused, this is a nano dev kit, there are no recovery / power buttons on this thing, it has the old version yocto running on sdcard, i just want to create a new sdcard image for it. there is no usb attached. so whats the command line to run ./dosdcard without it calling tegra-flash-helper

madisongh, 2020-05-20:
The Nano dev kit has a jumper for setting recovery mode - the exact location depends on which version of the carrier board you have.  I really recommend re-flashing via USB if you're upgrading, to ensure that the bootloader partitions store in the QSPI flash are correctly programmed for the new version of L4T.

You can try:
```
$ BOARDID=3448 FAB=300 ./dosdcard.sh
```
to create an SDCard image.  (If you know the right value of FAB for your particular Nano, use that value instead of `300`, so the image ends up with the correct device tree.)

There may be boot issues or other problems if the bootloader files that reside in the QSPI flash are out of date, though.

outbackdingo, 2020-05-21:
ok got a usb on it, got it flashed, it worked once, ghowever from curiousity, it doesnt seem to allow you to reflash a second time. 

./doflash.sh                                                               
ERR: chip does not identify as tegra210 ()

though it sees the board

 6765.995390] usb 1-4: new high-speed USB device number 19 using xhci_hcd
[ 6766.122221] usb 1-4: New USB device found, idVendor=0955, idProduct=7f21, bcdDevice= 1.02
[ 6766.122227] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 6766.122230] usb 1-4: Product: APX
[ 6766.122233] usb 1-4: Manufacturer: NVIDIA Corp.
 Fed    100%  dingo@localhost  ~/TEGRA  lsusb                                                                            14:50  21.05.20  55.62G RAM 
Bus 002 Device 002: ID 0781:5583 SanDisk Corp. Ultra Fit
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 5986:066d Acer, Inc BisonCam, NB Pro
Bus 001 Device 005: ID 0cf3:e300 Qualcomm Atheros Communications 
Bus 001 Device 003: ID 1c7a:0603 LighTuning Technology Inc. EgisTec_ES603
Bus 001 Device 013: ID 1b1c:1b17 Corsair APX
Bus 001 Device 012: ID 258a:0019  
Bus 001 Device 011: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 019: ID 0955:7f21 NVIDIA Corp. APX
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

outbackdingo, 2020-05-21:
nevermind, a reboot of my laptop fixed it

closed 2020-05-21 by outbackdingo