OE4T/meta-tegra #320: XENized Jetson NANO

outbackdingo, 2020-05-23:
Curios if anyone has done XEN from meta-virtualization for the NANO. I added the package xen to my build along with virtualization. seems its built, i flash a device and ls -al /boot shows
ls -al /boot/
total 37240
drwxr-xr-x    3 root     root          4096 Mar  9  2018 .
drwxr-xr-x   19 root     root          4096 May 23 02:37 ..
lrwxrwxrwx    1 root     root            39 Mar  9  2018 Image -> Image-4.9.140-l4t-r32.3.1+g47e7e1cb0b49
-rw-r--r--    1 root     root      37257224 Mar  9  2018 Image-4.9.140-l4t-r32.3.1+g47e7e1cb0b49
drwxr-xr-x    2 root     root          4096 Mar  9  2018 extlinux
-rw-r--r--    1 root     root       2089938 Mar  9  2018 initrd
-rw-r--r--    1 root     root        214717 Mar  9  2018 tegra210-p3448-0000-p3449-0000-a02.dtb
-rw-r--r--    1 root     root        214717 Mar  9  2018 tegra210-p3448-0000-p3449-0000-b00.dtb
lrwxrwxrwx    1 root     root            63 Mar  9  2018 u-boot-dtb.bin -> u-boot-jetson-nano-qspi-sd-2016.07+gitAUTOINC+565a18c42c-r0.bin
-rw-r--r--    1 root     root        483328 Mar  9  2018 u-boot-jetson-nano-qspi-sd-2016.07+gitAUTOINC+565a18c42c-r0.bin
lrwxrwxrwx    1 root     root            10 Mar  9  2018 xen -> xen-4.12.2
lrwxrwxrwx    1 root     root            10 Mar  9  2018 xen-4 -> xen-4.12.2
lrwxrwxrwx    1 root     root            10 Mar  9  2018 xen-4.12 -> xen-4.12.2
-rw-r--r--    1 root     root        853328 Mar  9  2018 xen-4.12.2
-rw-r--r--    1 root     root          2213 Mar  9  2018 xen-4.12.2.config

So id guess the missing pieces are booting xen from u-boot ? feeling like i need a u-boot_append to write the proper config ??

ichergui, 2021-03-28:
Hey @outbackdingo 
Did you successfully manage to resolve this issue ? if so, could please share here, I'm curious.
Thanks

outbackdingo, 2021-03-29:
while we can get xen to build for the jetson/xavier... i would need help with the u-boot booting xen piece

ichergui, 2021-10-04:
Hello @outbackdingo 
I'm closing this issue because no updates.
I didn't have time to take a look on it.
If I will get some news about `Xen` with `Jetson Nano` I will create a discussion channel and I will tag you.

closed 2021-10-04 by ichergui