OE4T/meta-tegra #322: reboot recovery

ericvh, 2020-05-23:
Was looking to build out some automation for using HW for regression testing -- nvidia forums suggest there is an option to reboot which forces it into recovery mode.  This option doesn't seem to be available in meta-tegra.

kekiefer, 2020-05-23:
`reboot forced-recovery` works in meta-tegra for me.

madisongh, 2020-05-23:
Yup, that's what I use.  There's no syntax checking for the option you enter there, so make sure you spell it correctly, or it won't take.

ericvh, 2020-05-24:
Weird -- I'm guessing my stripped down configuration is missing something.  Which recipe pulls in reboot with forced-recovery?

`root@smarter-jetson-xavier-4bcc25cc:/var/lib/docker# reboot forced-recovery
usage: reboot [-n] [-w] [-d] [-f] [-h] [-i]
        -n: don't sync before halting the system
        -w: only write a wtmp reboot record and exit.
        -d: don't write a wtmp record.
        -f: force halt/reboot, don't call shutdown.
        -h: put harddisks in standby mode.
        -i: shut down all network interfaces.`

madisongh, 2020-05-24:
Ah, right.  I use systemd in my builds, where `reboot` is `systemctl reboot` and takes the extra argument which it passes to `reboot()`.   I'll bet the sysvinit version doesn't support that.

ericvh, 2020-05-24:
ah, okay - thanks.

closed 2020-05-24 by ericvh