OE4T/meta-tegra #328: CMake: could not find CUDA

focusware-develop, 2020-05-31:
Hi Matt,

I'm trying to build my project with the latest version of the zeus-l4t-r32.3.1 branch. The projects that use "FindPackage(CUDA)" in their CMakeLists.txt file give an error during the configure phase 

`CMake Error at /media/yocto/jenkins-yocto/.yocto/tmp-glibc/work/armv8a_tegra194-oe-linux/jetnet/git-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
|   Could NOT find CUDA (missing: CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) (found
|   version "10.0")
| Call Stack (most recent call first):
|   /media/yocto/jenkins-yocto/.yocto/tmp-glibc/work/armv8a_tegra194-oe-linux/jetnet/git-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
|   /media/yocto/jenkins-yocto/.yocto/tmp-glibc/work/armv8a_tegra194-oe-linux/jetnet/git-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/FindCUDA.cmake:1085 (find_package_handle_standard_args)
|   CMakeLists.txt:9 (find_package)
`

I see that you've refactored to CUDA recipes to be able to build an SDK toolchain on march 31st (which is really nice :-). Before that date (commit f3433a7) the projects were building fine. Do you have any suggestions where to go, to find what might be causing the problem?

Regards,
Boris.

madisongh, 2020-06-01:
Hmm.  Yes, with those changes I focused mainly on improving the handling of CMake's newer CUDA-as-language support - so you could build either in bitbake or externally with the SDK with no extra hackery - but I thought that the older FindCUDA.cmake module was still working.  I'll take a look.

madisongh, 2020-06-01:
@focusware-develop Can you try out the changes in #330?  It turns out that I don't have a good project around that uses CMake and the old FindCUDA support, but from what testing I've been able to do, the additional environment variables make CMake happy again.

focusware-develop, 2020-06-01:
That works. Thanks a lot! I'll try the SDK build also, but that takes some time to finish. I'll let you know if it works for the SDK too.

madisongh, 2020-06-03:
Thanks for verifying.  I was able to try an SDK build, and I found a typo in one of the environment variable settings.  I've fixed that and have merged the change now.

closed 2020-06-03 by madisongh

focusware-develop, 2020-06-03:
I've tried the SDK build also but was not able to compile any cuda files (.cu). CMake works in the SDK now, but the actual build fails. Have you been able to compile cuda files with the SDK?

madisongh, 2020-06-03:
That may have been due to the typo I mentioned - The CUDA_NVCC_EXECUTABLE variable was set to just the directory containing `nvcc`, rather than the full path of the `nvcc` executable.  Once I fixed that, things worked better for me.

focusware-develop, 2020-06-03:
I'll give that a try!

focusware-develop, 2020-06-03:
I've changed the CMakeLists.txt file to use the "new" method, instead of `FIND_PACKAGE(CUDA)`, with `project (test LANGUAGES C CXX CUDA)` and changed `cuda_add_library()` call to `add_library()`.

Now everything cross-compiles with the SDK and the executable runs on the target! Thanks again for your help.

Boris.