OE4T/meta-tegra #329: warrior-l4t-r32.2 compilation error

lfdmn, 2020-06-01:
Hi,

[This commit](https://github.com/madisongh/meta-tegra/commit/28962ef4a13b63c73f9d2392b5c0309994e0198c) introduced a new **docker-ce_%.bbappend** and we don't have meta-virtualization in our configuration.

Consider using dynamic inclusion for external overwrite layers in your layer.conf

From this
```
# Recipe modifications for other layers that may be included in the build
BBFILES += " \
    ${LAYERDIR}/external/openembedded-layer/recipes*/*/*.bb \
    ${LAYERDIR}/external/virtualization-layer/recipes*/*/*.bb \
    ${LAYERDIR}/external/openembedded-layer/recipes*/*/*.bbappend \
    ${LAYERDIR}/external/virtualization-layer/recipes*/*/*.bbappend \
"
```

to this 
```
# Recipe modifications for other layers that may be included in the build
BBFILES_DYNAMIC += " \
openembedded-layer:${LAYERDIR}/external/openembedded-layer/recipes-support/*/*.bb \
openembedded-layer:${LAYERDIR}/external/openembedded-layer/recipes-support/*/*.bbappend \  
multimedia-layer:${LAYERDIR}/external/openembedded-layer/recipes-multimeda/*/*.bb \
multimedia-layer:${LAYERDIR}/external/openembedded-layer/recipes-multimedia/*/*.bbappend \  
virtualization-layer:${LAYERDIR}/external/virtualization-layer/recipes*/*/*.bb \       
virtualization-layer:${LAYERDIR}/external/virtualization-layer/recipes*/*/*.bbappend \ 
"
```
Cheers,
-Damien

madisongh, 2020-06-01:
The logic in meta-tegra's `layer.conf` file is similar... it should only be including that bbappend when `virtualization-layer` is in BBFILE_COLLECTIONS (which should only happen if you have meta-virtualization in your `bblayers.conf`).

Is that not working in your case?

madisongh, 2020-06-01:
Oh, I see.  I wasn't looking at the right branch.  I must have missed that change when I merged in the PR that made it.

madisongh, 2020-06-01:
I've reverted the commit that introduced that change.

closed 2020-06-01 by madisongh