OE4T/meta-tegra #331: Unable to build r34.4.2 core-image-minimal

UTSAnonymous, 2020-06-03:
Hi all, 

I am pretty new to Yocto and currently trying to get a simple image up and running to test things out. I'm currently using dunfell with meta-tegra branch dunfell-l4t-r34.4.2.

I edited the local.conf of poky and bblayers.conf to add the Nvidia SDK, GCC version, CUDA version and the correct machine as well (jetson-nano-qspi-sd). 

When building I got this error 
`he stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 3, function: <module>
     0001:__anon_20__media_SSD_presien_work_yocto_poky_nano_meta_conf_machine_include_arm_feature_arm_thumb_inc(d)
     0002:__anon_32__media_SSD_presien_work_yocto_poky_nano_meta_classes_patch_bbclass(d)
 *** 0003:__anon_707__media_SSD_presien_work_yocto_poky_nano_meta_classes_base_bbclass(d)
     0004:__anon_202__media_SSD_presien_work_yocto_poky_nano_meta_classes_reproducible_build_bbclass(d)
     0005:__anon_1106__media_SSD_presien_work_yocto_poky_nano_meta_classes_insane_bbclass(d)
     0006:__anon_1346__media_SSD_presien_work_yocto_poky_nano_meta_classes_insane_bbclass(d)
     0007:__anon_252__media_SSD_presien_work_yocto_poky_nano_meta_classes_package_bbclass(d)
File: '/media/SSD/presien_work/yocto/poky-nano/meta/classes/base.bbclass', lineno: 590, function: __anon_707__media_SSD_presien_work_yocto_poky_nano_meta_classes_base_bbclass
     0586:
     0587:    needsrcrev = False
     0588:    srcuri = d.getVar('SRC_URI')
     0589:    for uri in srcuri.split():
 *** 0590:        (scheme, _ , path) = bb.fetch.decodeurl(uri)[:3]
     0591:
     0592:        # HTTP/FTP use the wget fetcher
     0593:        if scheme in ("http", "https", "ftp"):
     0594:            d.appendVarFlag('do_fetch', 'depends', ' wget-native:do_populate_sysroot')
File: '/media/SSD/presien_work/yocto/poky-nano/bitbake/lib/bb/fetch2/__init__.py', lineno: 357, function: decodeurl
     0353:    """
     0354:
     0355:    m = re.compile('(?P<type>[^:]*)://((?P<user>[^/;]+)@)?(?P<location>[^;]+)(;(?P<parm>.*))?').match(url)
     0356:    if not m:
 *** 0357:        raise MalformedUrl(url)
     0358:
     0359:    type = m.group('type')
     0360:    location = m.group('location')
     0361:    if not location:
Exception: bb.fetch2.MalformedUrl: The URL: '/media/SSD/presien_work/nvidia_jetson/sdkm_downloads/cuda-repo-l4t-10-2-local-10.2.89_1.0-1_arm64.deb' is invalid and cannot be interpreted
`

Any help would be nice.

madisongh, 2020-06-03:
Make sure your setting of NVIDIA_DEVNET_MIRROR is in the form of a URL.  From this:

>`bb.fetch2.MalformedUrl: The URL: '/media/SSD/presien_work/nvidia_jetson/sdkm_downloads/cuda-repo-l4t-10-2-local-10.2.89_1.0-1_arm64.deb' is invalid and cannot be interpreted`

it looks like you forgot to prefix the pathname with `file://`.

UTSAnonymous, 2020-06-03:
oh, I thought the file:// prefix is only for the windows path. Thanks!

UTSAnonymous, 2020-06-03:
@madisongh 

so I got everything setup with the correct gcc version as well, but I'm facing this issue here.

`ERROR: libtool-native-2.4.6-r0 do_populate_sysroot_setscene: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_populate_sysroot_setscene(d)
     0003:
File: '/media/SSD/presien_work/yocto/poky-nano/meta/classes/staging.bbclass', lineno: 127, function: do_populate_sysroot_setscene
     0123:do_populate_sysroot[sstate-outputdirs] = "${COMPONENTS_DIR}/${PACKAGE_ARCH}/${PN}"
     0124:do_populate_sysroot[sstate-fixmedir] = "${COMPONENTS_DIR}/${PACKAGE_ARCH}/${PN}"
     0125:
     0126:python do_populate_sysroot_setscene () {
 *** 0127:    sstate_setscene(d)
     0128:}
     0129:addtask do_populate_sysroot_setscene
     0130:
     0131:def staging_copyfile(c, target, dest, postinsts, seendirs):
`

full log here:
[log.log](https://github.com/madisongh/meta-tegra/files/4723341/log.log)

I don't know what to make of this.

madisongh, 2020-06-03:
From these errors in the log:
```
Subprocess output:
cp: cannot create hard link '/media/SSD/presien_work/yocto/poky-nano/build/tmp/sysroots-components/x86_64/libtool-native/usr/lib/libltdl.so' to './usr/lib/libltdl.so': Operation not permitted
cp: cannot create hard link '/media/SSD/presien_work/yocto/poky-nano/build/tmp/sysroots-components/x86_64/libtool-native/usr/lib/libltdl.so.7' to './usr/lib/libltdl.so.7': Operation not permitted
```
it looks like the filesystem on `/media/SSD` is not suitable for running bitbake builds.  It should be a Linux native filesystem type like ext4, rather than something like vfat or a shared folder in a virtual machine.

UTSAnonymous, 2020-06-04:
Hey thanks a lot for the help, so I was able to build it but halfway through the build an error occurred. The error seems to be from compiling the Linux kernel 4.9. 

`ERROR: linux-tegra-4.9.140+gitAUTOINC+a26c322c04-r0 do_compile: oe_runmake failed
ERROR: linux-tegra-4.9.140+gitAUTOINC+a26c322c04-r0 do_compile: Execution of '/home/nicholas/poky-nano/build/tmp/work/jetson_nano_qspi_sd-poky-linux/linux-tegra/4.9.140+gitAUTOINC+a26c322c04-r0/temp/run.do_compile.7877' failed with exit code 1:
  GEN     ./Makefile
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  GEN     ./Makefile`

`/home/nicholas/poky-nano/build/tmp/work-shared/jetson-nano-qspi-sd/kernel-source/arch/arm64/boot/dts/Makefile:132: target 'arch/arm64/boot/dts/_ddot_' given more than once in the same rule 
make[3]: *** No rule to make target 'arch/arm64/boot/dts/_ddot_/_ddot_/_ddot_/_ddot_/nvidia/platform/t210/porg/kernel-dts/tegra210-p3448-0000-p3449-0000-a02.dtb'.  Stop.
| make[2]: *** [arch/arm64/Makefile:151: _ddot_/_ddot_/_ddot_/_ddot_/nvidia/platform/t210/porg/kernel-dts/tegra210-p3448-0000-p3449-0000-a02.dtb] Error 2
| make[1]: *** [Makefile:169: sub-make] Error 2
| make: *** [Makefile:24: __sub-make] Error 2
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/nicholas/poky-nano/meta-tegra/recipes-kernel/linux/linux-tegra_4.9.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2778 tasks of which 1 didn't need to be rerun and 1 failed.
`

Not too sure why this is happening.

madisongh, 2020-06-04:
What version of make do you have on your build host?

UTSAnonymous, 2020-06-04:
GNU Make 4.3

madisongh, 2020-06-04:
OK.  It wouldn't surprise me if this is due to some change in that version of GNU make that is causing this.  Latest version I've used is 4.2.1. Any chance you could downgrade to that version and see if it works better?

madisongh, 2020-06-05:
From the GNU make 4.3 release notes:
```
* WARNING: Backward-incompatibility!
  Previously appending using '+=' to an empty variable would result in a value
  starting with a space.  Now the initial space is only added if the variable
  already contains some value.  Similarly, appending an empty string does not
  add a trailing space.
```
This is the change that is causing the failure.  NVIDIA used this trick to define a make variable with a single space.  I'll get a fix/workaround for this in place shortly.

madisongh, 2020-06-05:
I've pushed updates to all current branches, including dunfell-l4t-r32.4.2, that updates the SRCREV of the kernel to include a patch that fixes the GNU Make 4.3 compatibility problem.

madisongh, 2020-11-08:
Haven't heard anything new on this, so I'll close it under the assumption that fixing the make compatibility issue addressed the problem.

closed 2020-11-08 by madisongh