OE4T/meta-tegra #332: Jetson Xavier NX core-image-sato does not load UI

dreambeyondorange, 2020-06-03:
I have a Jetson Xavier Devkit from NVIDIA and am using the dunfell-l4t-r32.4.2.  This is the one with an SD card and the wifi integrated into the carrier board.  I successfully built core-image-sato and it boots into the OS but I'm stuck in a terminal.  I did this same process using R32.3.1 with the Nano devkit and was able to get into the UI and run cuda samples.  Are there extra deps I need to add with the new brach?  I'm not sure why it doesn't bring up the UI when the nano does with the older branch.

madisongh, 2020-06-03:
I've run into this sometimes, where some race condition at boot time prevents the X display from coming up initially.  If you can get logged in and use
```
# systemctl restart xserver-nodm
```
does that help?  That's what has worked for me in these cases.

If not, take a look at `/var/log/Xorg.0.log` to see what the X server is reporting.

dreambeyondorange, 2020-06-03:
It seems like something deeper is wrong, when I try to use systemctl, it says not found.  Here's changes to the config file local.conf:

IMAGE_CLASSES += "image_types_tegra"  
IMAGE_FSTYPES = "tegraflash"  
MACHINE = "jetson-xavier-nx-devkit"  
LICENSE_FLAGS_WHITELIST = "commercial"  
IMAGE_INSTALL_append = "cuda-samples"  
INHERIT += "rm_work"  
GCCVERSION = "8.%"  
require contrib/conf/include/gcc-compat.conf  
CUDA_VERSION="10.2"  
NVIDIA_DEVNET_MIRROR = "file:///home/ANT.AMAZON.COM/ndry/Downloads/nvidia/sdkm_downloads"  

and here's my bblayers.conf changes

BBLAYERS ?= " \
  /portable/jetson-nx/meta \
  /portable/jetson-nx/meta-poky \
  /portable/jetson-nx/meta-yocto-bsp \
  /portable/jetson-nx/meta-tegra \
  /portable/jetson-nx/meta-tegra/contrib \
  "
Everything else is stock from meta-tegra

dreambeyondorange, 2020-06-03:
I'm doing it on this [devkit](https://developer.nvidia.com/embedded/jetson-xavier-nx-devkit) from NVIDIA, not the Nano dev kit with an NX module

madisongh, 2020-06-03:
Well, if `systemctl` isn't there, you're probably using sysvinit instead of systemd as the init manager.  In which case, the command would be
```
# /etc/init.d/xserver-nodm restart
```
instead.

Most of my testing is done with distros that use systemd as the init manager.  sysvinit-based builds should work OK, but there may be some dependency issues with the order of services starting that need to be sorted out.

Adding `INIT_MANAGER = "systemd"` in your build configuration will switch to systemd, if you want to give that a try, but be prepared for a long rebuild when you make that change.

dreambeyondorange, 2020-06-03:
[image]

madisongh, 2020-06-03:
Looks like the SDcard didn't get written properly.  Did you flash it via USB, or use the `dosdcard.sh` script in the tegraflash package to generate it?

dreambeyondorange, 2020-06-03:
I used the doflash.sh script and flashed it via USB and I end up with the screenshot above.  

I just tried the dosdcard.sh and flashed the image with dd.  I get the same result.  I cannot mount the sd card on my host after flashing

The systemd image is still building, but that doesn't seem like it will help

madisongh, 2020-06-03:
Could be a problem with the SDcard itself.  Do you have another you can try?

dreambeyondorange, 2020-06-03:
I just loaded the sd card image from my nano build (core-image-sato) and it mounts OK.  The nano image was built using Jetpack 4.3 and the meta-tegra dunfell branch with gcc 7.

madisongh, 2020-06-03:
Can you overwrite that working SDcard with your NX image and see if that works better?

dreambeyondorange, 2020-06-03:
I ran from the NX tegraflash dir:
`sudo dd if=core-image-sato.sdcard of=/dev/mmcblk0 bs=1M status=progress`
on the sd card that was able to mount with the nano image and i get mounting errors:
mount: /mount: wrong fs type, bad option, bad superblock on /dev/mmcblk0p1, missing codepage or helper program, or other error.
and from the file explorer I get a similar error

dreambeyondorange, 2020-06-04:
i tried again after switching to systemd.  I wonder if something got wonky so ill try a fresh overnight build and try to flash again tomorrow morning.  I assume you have testing the NX devkit with an sd card image already?

madisongh, 2020-06-04:
Yes, I've tested with an NX module in a Nano carrier as well as with the just-release NX dev kit.

dreambeyondorange, 2020-06-04:
Great!  I tried to do it directly and got the following error.  On two different sd cards
`[   5.7833 ] `
`About to make an SDcard image on /dev/mmcblk0. OK? yes`
`Init...APP...kernel...kernel_b...kernel-dtb...kernel-dtb_b...recovery...recovery-dtb...kernel-bootctrl...kernel-bootctrl_b...RECROOTFS...UDA...[OK]`
`Writing...ERR: cannot locate block device /dev/mmcblk01
`

madisongh, 2020-06-04:
Oops.. that's a problem in the make-sdcard script, which isn't adding the 'p' separator in the two lines that set the destination device name:
```
	dest="/dev/$DEVNAME$partnumber"
```
Because I've only tested with USB card readers that show up as `/dev/sd` devices and not `/dev/mmcblk` devices.

dreambeyondorange, 2020-06-04:
can you share your local.conf and bblayers files for the nx dev kit sato build?  I'd like to see if I'm doing something there that is causing a problem, since those are the only files I edit.

madisongh, 2020-06-04:
My test distro is [here](https://github.com/madisongh/test-distro).

dreambeyondorange, 2020-06-04:
I did a bit more testing and it seems to be a problem with the ext4 image file that gets extracted with the tegraflash.zip.  I tried mounting it by itself with the following:

$ sudo mount -o loop deploy/core-image-sato-jetson-xavier-nx-devkit-20200604135756.tegraflash/core-image-sato.ext4 /mount
mount: /mount: wrong fs type, bad option, bad superblock on /dev/loop4, missing codepage or helper program, or other error.
$ sudo mount -o loop deploy/core-image-sato-jetson-xavier-nx-devkit.tegraflash/core-image-sato.ext4 /mount
mount: /mount: wrong fs type, bad option, bad superblock on /dev/loop4, missing codepage or helper program, or other error.

I tried both extracting the symlink and extracting the linked archive and got the same result with the core-image-sato.ext4 files extracted from those archives

However, when I use the ext4 file from the tmp/deploy/images dir I can successfully mount the image.
$ sudo mount -o loop jetson-nx/build/tmp/deploy/images/jetson-xavier-nx-devkit/core-image-sato-jetson-xavier-nx-devkit.ext4 /mount

dreambeyondorange, 2020-06-04:
Looks like it was a problem with using the file explorer in Ubuntu 18.04 to extract the files vs using tar.  Thanks for your help!

closed 2020-06-04 by dreambeyondorange

MJLHThomassen-Sorama, 2020-06-10:
Just a note for future reference for people with similar issues. I've been having similar problems with building an image for the Jetson Nano Devkit sd card on dunfell-l4t-r32.4.2.

When flashing using doflash and dosdcard, i get an image that boots the Nano into the splashscreen (nvidia logo shows) but does not go further. On in spection via the UART port, i see U-Boot attempting a boot from mmc, which fails, it then starts to boot from USB which also fails since there is no usb boot device, it then tries to use BOOTP which ofcourse also fails.

Here is the uboot output:

```
U-Boot 2020.04 (Apr 26 2020 - 16:16:28 +0000)

SoC: tegra210
Model: NVIDIA Jetson Nano Developer Kit
Board: NVIDIA P3450-0000
DRAM:  4 GiB
MMC:   sdhci@700b0000: 1, sdhci@700b0600: 0
Loading Environment from SPI Flash... SF: Detected mx25u3235f with page size 256 Bytes, erase size 4 KiB, total 4 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
starting USB...
Bus usb@7d000000: tegrausb: Invalid dr_mode 2 for host mode
probe failed, error -1
No working controllers found

Warning: eth_rtl8169 using MAC address from ROM
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
...
```

And here is the output of U Boot from the default nvidia provided image:

```
U-Boot 2016.07-g45dfa3dff4 (Apr 08 2020 - 18:04:08 -0700)

TEGRA210
Model: NVIDIA P3450-Porg
Board: NVIDIA P3450-PORG
DRAM:  4 GiB
MMC:   Tegra SD/MMC: 0, Tegra SD/MMC: 1
SF: Detected MX25U3235F with page size 256 Bytes, erase size 4 KiB, total 4 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found /boot/extlinux/extlinux.conf
Retrieving file: /boot/extlinux/extlinux.conf
733 bytes read in 121 ms (5.9 KiB/s)
1:      primary kernel
Retrieving file: /boot/initrd
5487746 bytes read in 295 ms (17.7 MiB/s)
Retrieving file: /boot/Image
34265096 bytes read in 1536 ms (21.3 MiB/s)
append: tegraid=21.1.2.0.0 ddr_die=4096M@2048M section=512M memtype=0 vpr_resize usb_port_owner_info=0 lane_owner_info=0 emc_max_dvfs=0 touch_id=0@63 video=tegrafb no_console_suspend=1 console=ttyS0,115200n8 debug_uartport=lsport,2 earlyprintk=uart8250-32bit,0x70006000 maxcpus=4 usbcore.old_scheme_first=1 lp0_vec=0x1000@0xff780000 core_edp_mv=1075 core_edp_ma=4000  root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 console=ttyS0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0    root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 console=ttyS0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0 quiet
## Flattened Device Tree blob at 83100000
   Booting using the fdt blob at 0x83100000
   reserving fdt memory region: addr=80000000 size=20000
   Using Device Tree in place at 0000000083100000, end 00000000831777ed

Starting kernel ...
```

Also, noteworthy, the nvidia image prints this line a few moments before uboot starts:

```
DTB cmdline: root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 console=ttyS0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0
boot image cmdline: root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 console=ttyS0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0
```

While the image i flashed prints:

```
DTB cmdline: console=ttyS0,115200 console=tty0 fbcon=map:0 net.ifnames=0
```

IT seems like something is going wrong with indicating the propper boot command (btw, i have 0 experience with UBOOT so i might be completely wrong about this).

Also, im getting the same problems as @dreambeyondorange when i inspect the SD card that contains my own image with the Disks tool in Ubuntu: " wrong fs type", while if i insert the image from Nvidia, the rootfs partitions is mounted and readily available.

My solution to this was the same as @dreambeyondorange :

I untared the tegraflash.tar.gz using ubuntu file manager. This resulted in the non-functional image. Untarring using command line `tar -xzvf <yimage>.tegraflash.tar.gz` and then running `sudo ./dosdcard.sh /dev/sdc` (my sd card was mounted on in the reader on sdc) resulted in an image that boots to console properly (and also the APP partition is now mountable on Ubuntu host machine).