OE4T/meta-tegra #334: Can't run cuda samples on latest dunfell branch

fasm22, 2020-06-03:
Hello!

I am trying to use CUDA with latest dunfell branch. But for example when I try to use the deviceQuery example it fails as follows:

```
jetson-tx2:/usr/local/cuda-10.2/samples/1_Utilities/deviceQuery$ ./deviceQuery 
./deviceQuery Starting...

 CUDA Device Query (Runtime API) version (CUDART static linking)

cudaGetDeviceCount returned 100
-> no CUDA-capable device is detected
Result = FAIL
```

My local.conf file is as follows:

```
CONF_VERSION = "1"

MACHINE ?= "jetson-tx2"

IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash"

NVIDIA_DEVNET_MIRROR = "file:///home/fsolano/Downloads/nvidia/sdkm_downloads"

BB_NUMBER_THREADS = '6'
PARALLEL_MAKE = '-j6'
SSTATE_DIR ?= "/home/fsolano/Yocto/shared-sstate-cache"
DL_DIR ?= "/home/fsolano/Yocto/downloads"

#Cuda support for gcc
GCCVERSION = "7.%"
require contrib/conf/include/gcc-compat.conf

# GCC 7 doesn't support fmacro-prefix-map, results in "error: cannot compute suffix of object files: cannot compile"
DEBUG_PREFIX_MAP_remove = "-fmacro-prefix-map=${WORKDIR}=/usr/src/debug/${PN}/${EXTENDPE}${PV}-${PR}"

SECURITY_CFLAGS_pn-tini_append = " ${SECURITY_NOPIE_CFLAGS}"

IMAGE_INSTALL_append = " cuda-binaries cuda-toolkit cuda-compiler cuda-nvcc cuda-libraries cuda-driver cuda-cudart cuda-samples cuda-command-line-tools"
IMAGE_INSTALL_append = " cudnn tensorrt gstreamer1.0-plugins-nvvideo4linux2"

DISTRO_FEATURES_append = " ldconfig"

EXTRA_IMAGE_FEATURES = " tools-sdk"
```

madisongh, 2020-06-03:
Since I'm seeing cuda-10.2, I'm assuming this is the `dunfell-l4t-r32.4.2` branch, rather than `dunfell` (which is L4T R32.3.1 with CUDA 10.0), right?  And it looks like you're trying to build the CUDA samples on the device?

Did you build the cuda-samples package into your image?  If so, can you run `/usr/bin/cuda-samples/deviceQuery` and see if that works?

fasm22, 2020-06-03:
Yes, the branch is dunfell-l4t-r32.4.2, sorry about the confusion.

Yes, it added cuda-samples package into the image.

The error also appears if I try to run /usr/bin/cuda-samples/deviceQuery

madisongh, 2020-06-03:
Sure sounds like there's a device missing, or something's messed up with the drivers.  What sort of image are you building?

fasm22, 2020-06-03:
I am building **core-image-sato-dev**

madisongh, 2020-06-03:
OK, well, everything should be in there, and then some.  Definitely odd.  Do you have `strace` installed in your image?  If so, can you do
```
# mkdir /tmp/strace-output
# cd /tmp/strace-output
# strace -ff -oxx /usr/bin/cuda-samples/deviceQuery
```
and post the `xx.<pid>` file that gets generated?   That could help with figuring out what's wrong.

fasm22, 2020-06-04:
I tested the **dunfell** branch instead of the **dunfell-l4t-r32.4.2** and the CUDA examples ran as expected. So maybe something is broken in latest branch. I will test with strace in **dunfell-l4t-r32.4.2** and post the results

madisongh, 2020-06-11:
Good to know.  All my builds use gcc 8, but I would have expected gcc 7 to be OK to use as well.

fasm22, 2020-06-11:
Hi sorry to re-open the issue. I really appreciate your help and time. The last test with gcc8 was on Jetson Nano and everything worked as expected. However on TX2 I am still seeing the following error:

```
# /usr/bin/cuda-samples/deviceQuery
/usr/bin/cuda-samples/deviceQuery starting...

 CUDA Device Query (Runtime API) version (CUDART static linking)

NVMAP_IOC_QUERY_HEAP_PARAMS failed [Inappropiate ioctl for device]
cudaGetDeviceCount returned 999
-> unknown error
Result - FAIL
```

It seems that this instruction triggers the error:
```
ioctl(4, _IOC(_IOC_READ, 0x4e, 0x69, 0x28), 0x7fcf3e9538) = -1 ENOTTY (Inappropriate ioctl for device)
```

Here is a comparison between dunfell and dunfell-l4t-r32.4.2 strace command output for deviceQuery CUDA sample:

https://www.diffchecker.com/scPLoAQ3

Regards,
Fabian

madisongh, 2020-06-12:
That is an ioctl to `/dev/nvmap`, which is returning ENOTTY because it isn't recognizing the ioctl code.

Do you have a modified kernel configuration for your TX2, by any chance?  There is a kernel config option, CONFIG_TEGRA_NVMAP_V2, which, if it's enabled, would cause this.  That config is disabled in the stock configuration, and should remain disabled.

I did run a test build of my own this morning using gcc7 instead of gcc8, and it worked fine, too, running on my TX2.

So something else must be different in your build - in particular, check your kernel config and build.

cg3dland, 2020-09-07:
@fasm22, I am new to Yocto and meta-tegra, I get an error while building cuda-binaries, I add following into my local.conf:
IMAGE_INSTALL_append += " cuda-binaries cuda-toolkit cuda-compiler cuda-nvcc cuda-libraries cuda-driver cuda-cudart cuda-command-line-tools"

and I get following errors:
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'cuda-binaries' (but /data/yocto1/tegra/meta/recipes-sato/images/core-image-sato.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'cuda-binaries' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['cuda-binaries']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'cuda-binaries']

but if I change the IMAGE_INSTALL_append line to:
IMAGE_INSTALL_append += " cuda-toolkit cuda-compiler cuda-nvcc cuda-libraries cuda-driver cuda-cudart cuda-command-line-tools"

the build process could go to end normally and the core image could b built successfully.

May I learn from you what wrong do I make in my local.conf?

Thanks,
George

madisongh, 2020-09-07:
@cg3dland The `cuda-binaries` recipe does not provide any installable packages.  Recipes just provide instructions for building things... in most cases, the thing a recipe builds is a package to be installed, but not always.

cg3dland, 2020-09-08:
So， it is invalid to add cuda-binaries into the IMAGE_INSTALL_append because the cuda-binaries recipe does not provide any installable packages, am I right?

madisongh, 2020-09-08:
Correct.

madisongh, 2020-11-02:
With nothing new on this for a while, I'll assume we're all good here and close.

closed 2020-11-02 by madisongh