OE4T/meta-tegra #336: OTA update for board without u-boot

nandra, 2020-06-04:
Hi,

this is not an issue but maybe getting more info about possibilities and others to share ;). I saw that for u-boot based board like nano mender + other OTA can be used without an issue. But what about boards like NX which don't contain u-boot in boot chain? I had an idea to modify cboot to mimic u-boot functionality (store bootcount + bootlimit) and then swupdate can be used. Other opinion was to start u-boot from cboot but this seems not straightforward (maybe even impossible). Are there any other documents, tips where to look? Thanks a lot.

dwalkes, 2020-06-04:
@nandra @madisongh has added support for cboot based update on mender in the zeus branch.  See the full history of changes for zeus at https://github.com/mendersoftware/meta-mender-community/pull/147/commits

The way he's done it for mender is to [make the state scripts use NVIDIA's redundant boot](https://github.com/mendersoftware/meta-mender-community/commit/f0dcd9100566fe0ee0833ebb89c919beefb62cb3#diff-8e4d3b414d01d4c34c014c97054695dd) and nv_update_engine described in [NVIDIA boot redundancy documentation](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/bootloader_update_agx_tx2.html) and make the mender fw_printenv and fw_setenv calls [do nothing other than set a flag /var/lib/mender/upgrade_available](https://github.com/mendersoftware/meta-mender-community/commit/49285bece4488180f323b2f24d5386f6d2f75b3b#diff-8e4d3b414d01d4c34c014c97054695dd).  He can explain it a lot better but these are a few pointers.

Also FYI I'm working on a mender port to dunfell how [see this link for work in progress](https://github.com/madisongh/test-distro/issues/5) which will target cboot platforms first since there's additional work for u-boot support in l4t-r32.4.2

madisongh, 2020-06-04:
There are some drawbacks to relying entirely on NVIDIA's redundant boot support - there are some cases where it will simply stop trying and switch to recovery mode, which isn't desirable on devices that run unattended.  I created [some tools](https://github.com/madisongh/tegra-boot-tools) to deal with that.

nandra, 2020-06-11:
@dwalkes @madisongh thanks a lot for pointers. I'll give a try.

nandra, 2020-06-11:
@madisongh would be possible to cherrypick nx module support for zeus branch? I checked dunfell and there cherrypick 3 commits whihc added support for nx emmc but build is failing. Do I need also some other commits? Thanks.

madisongh, 2020-06-11:
It's a bit more involved than that, since support for NX didn't come until L4T R32.4.2. The dunfell-l4t-r32.4.2 branch should have everything needed, and I build both nx-devkit and nx-emmc with my test distro that uses that branch.

nandra, 2020-06-11:
@madisongh ok thanks. I would like to try mender for nx (not sure when dunfell land in meta-mender as zeus is not yet ready :))

madisongh, 2020-06-11:
@dwalkes has something in the works, I believe - see [this test-distro issue](https://github.com/madisongh/test-distro/issues/5).

dwalkes, 2020-06-11:
@nandra @madisongh 

> see [this test-distro issue](https://github.com/madisongh/test-distro/issues/5)

Correct, the branches [here](https://github.com/Trellis-Logic/test-distro/blob/dunfell-mender-l4t-r32.4.2/.gitmodules) work with jetson-tx2-cboot as long as you don't care about the test-distro code changes mentioned there.  I haven't tried Xavier NX yet but help with that is welcome.  I also hope to have an update on test-distro proposal in https://github.com/madisongh/test-distro/issues/5 within the next week.

nandra, 2020-06-15:
@dwalkes I've cloned test distro and hopefuly setup everything be when start building core-image-minimal I got:

```
--- Starting bitbake server pid 17410 at 2020-06-15 19:53:50.574587 ---
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/mender-full.bbclass
```

and also in build I don't have bblayers.conf? I just source ```./setup-env``` Thanks.

dwalkes, 2020-06-15:
@nandra 

> I've cloned test distro 

Presumably you mean my hacked together mender branch test-distro repo at https://github.com/Trellis-Logic/test-distro/tree/dunfell-mender-l4t-r32.4.2 correct?

Not sure if this was obvious but you'll need to also
```
git checkout --track origin/dunfell-mender-l4t-r32.4.2
git submodule update --init
```
Then
```
source setup-env
```
Then follow instructions [here](https://github.com/Trellis-Logic/test-distro/tree/dunfell-mender-l4t-r32.4.2#security-options) to clear security options in local.conf unless you've setup signing/encryption.

We have some hacks for jetson-tx2 cboot we can share that can get you running Mender successfully on that platform if you like.  We'll try to get those out later today.

All this will be cleaned up with https://github.com/madisongh/test-distro/issues/5 but this should get you going for now.

dwalkes, 2020-06-15:
Also I haven't tried `core-image-minimal` yet, the only thing we've got working at the moment is `core-image-base` with jetson-tx2_cboot

nandra, 2020-06-15:
@dwalkes  I performed steps you mentioned  and still have same issue when running:
```
MACHINE=jetson-xavier-nx-devkit-emmc nice bitbake core-image-base
``` 

Maybe it's not prepared yet for this machine? Thanks.

dwalkes, 2020-06-15:
@nandra 

> Maybe it's not prepared yet for this machine? Thanks.

Correct, `jetson-tx2-cboot` is the only build we've tried thus far.

nandra, 2020-06-16:
@dwalkes ok let me try to look and check what is going on ;). Thanks.

dwalkes, 2020-06-17:
The `jetson-tx2-cboot` mender image now works at https://github.com/Trellis-Logic/test-distro/tree/dunfell-mender-l4t-r32.4.2 without signing turned on, see [README instructions](https://github.com/Trellis-Logic/test-distro/tree/dunfell-mender-l4t-r32.4.2#mender-hack-build)

nandra, 2020-06-20:
@dwalkes I've tried added support for jetson-nx emmc and I have core-image-minimal build working but I have few qestions:

1. for tx2 you created new machine which inherit original one + add mender stuff on top (I did same for nx). Is that right?
2. in recipe bootfiles there is ```eks.img``` Can I reuse it also fro nx?

My dummy modifications was only to add (copy'n'paste) in ```tegra-bootfiles_32.4.2.bbappend``` recipe + added compatible machine to bootfiles. Thanks a lot.

dwalkes, 2020-06-20:
@nandra 

> I've tried added support for jetson-nx emmc and I have core-image-minimal build working

Great!

> for tx2 you created new machine which inherit original one + add mender stuff on top (I did same for nx). Is that right?

I actually didn't create it, as with most of the stuff in this test branch it came from Matt's [zeus changes](https://github.com/madisongh/test-distro/blob/zeus-mender-l4t-r32.3.1/layers/meta-testdistro/conf/machine/jetson-tx2-cboot.conf), but yes, the `jetson-tx2-cboot` machine was originally added with mender specific cboot support.  Based on the comment [here](https://github.com/madisongh/test-distro/commit/c6c3acf791f6a65fcc99b97278b419b788cb2b3a#diff-ac9e399e04d088440931f0cd3d5e8891) it looks like this was just to make it easier to support both cboot and uboot options for TX2, which [supports both](https://github.com/madisongh/meta-tegra/wiki/Using-cboot-as-bootloader).  So I don't think it would be necessary to make a new machine for Xavier NX which appears to [only support cboot](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fsoftware_features_jetson_xavier_nx.html%23wwpID0E0SC0HA).

> in recipe bootfiles there is eks.img Can I reuse it also fro nx?

This is related to secure boot encryption support.  You can see the commit [here](https://github.com/madisongh/test-distro/commit/2e5d0c83268828494bee80aa178ab517e2c86874) where it was added and comments from Matt [here](https://github.com/madisongh/test-distro/issues/5#issuecomment-637780163) which is where I'd start if you need to support encryption.  When you use my [security options hack](https://github.com/Trellis-Logic/test-distro/tree/dunfell-mender-l4t-r32.4.2#security-options) it won't be used so I don't think it will matter.  It would be better to remove the dependency I suppose and condition the bootfiles logic which references it on whether signing is supported.  I expect this is something which will be cleaned up in https://github.com/madisongh/test-distro/issues/5 using secureboot MACHINEOVERRIDES as discussed [here](https://github.com/madisongh/test-distro/issues/5#issuecomment-643761920)

nandra, 2020-06-22:
@dwalkes I was too fast with reply that something works ;). I build master branch not dunfell-mender. When building later one I'm getting some signing issue [here](https://pastebin.com/pyejNu6M). kernel-bup-payload.bb:do_deploy task fails with this trace. Ideas? I have ```SECURITY_OPTIONS=""``` setup in local.conf. Thanks.

dwalkes, 2020-06-23:
@nandra 
It looks like the first error is here:
```
| [   2.9429 ] File DATAFILE open failed
| [   3.5520 ]
| Error: Return value 19
```
Which I think means there's a problem with the sed commands [here](https://github.com/Trellis-Logic/meta-mender-community/blob/dunfell-l4t-r32.4.2/meta-mender-tegra/classes/image_types_mender_tegra.bbclass#L12) not getting which I haven't tried to trace.

There's another issue here;
```
| File /home/jenkins/test-distro/build/tmp/work/jetson_xavier_nx_devkit_emmc-td-linux/kernel-bup-payload/1.0-r0/bup-payload/signed/xusb_sil_rel_fw does not exist
...
|   File "/home/jenkins/test-distro/build/tmp/work/jetson_xavier_nx_devkit_emmc-td-linux/kernel-bup-payload/1.0-r0/bup-payload/BUP_generator.py", line 266, in fill_image
|     binary_handle = open(entry_info[0], 'rb')
| TypeError: expected str, bytes or os.PathLike object, not NoneType
|
| FAILURE: xusb_only_payload not created
```
I don't have my head deep wrapped around the signing bup logic enough to be able to give a suggestion here off the cuff.  I'd probably poke around in the run script to see where [xusb_sil_rel_fw](https://github.com/madisongh/meta-tegra/blob/dunfell-l4t-r32.4.2/classes/image_types_tegra.bbclass#L287) is getting referenced.  

You can watch [this branch](https://github.com/kekiefer/test-distro/tree/master+wip-mender) for some work in progress changes related to mender support but nothing related to Xavier NX yet.

lfdmn, 2020-06-23:
I had the same error yesterday.

The xusb_only_payload failure is due to DATAFILE not being found and stopping the script. So trip the <filename> tags from your own partitions as @dwalkes metioned and the bup signing will succeed.

nandra, 2020-06-25:
@lfdmn you mean drop DATAFILE partition? I'm building mender image and would like to use data partition.
Can you please be more specific how do you fix it? Thanks.

I looked at image_types_mender_tegra.bbclass but I'm not sure I understand this line:
```
sed -i -e"1a sed -i -e'/DATAFILE/d' ./flash.xml.in" $outfile
``` 
Also I searched work directory but cannot find any reference of ```generate_bup_payload.sh``` script which should be updated. Any pointers?

madisongh, 2020-06-25:
That's a `sed` command to insert (`1a` means "append after line 1") an extra `sed` command to remove the `DATAFILE` tag from the `flash.xml.in` file.  The script being changed is either `generate_bup_payload.sh` when creating a normal tegraflash package, or `doflash.sh` when generating a BUP payload.  Take a look at `image_types_tegra.bbclass` in meta-tegra to see the functions the `_append`s in `image_types_mender_tegra.bbclass` apply to.

Yeah, it's ugly and complicated.

kekiefer, 2020-07-02:
The `kernel-bup-payload` error should be fixed up by https://github.com/BoulderAI/meta-mender-community/commit/78b441c333d05ea758563c4c956363caedbd81e2 .

See [here](https://github.com/kekiefer/test-distro/tree/master+wip-mender) for the work-in-progress -- jetson-tx2 (only virtual/bootloader=cboot for now) builds and works and jetson-xavier at least builds now.

I haven't looked at all at xavier-nx yet, but if it shares the same flash layout as jetson-xavier, the meta-mender-tegra layer will handle it the same, and it might just work. Let me know ;)

madisongh, 2020-07-02:
> I haven't looked at all at xavier-nx yet, but if it shares the same flash layout as jetson-xavier, the meta-mender-tegra layer will handle it the same, and it might just work. Let me know ;)

This is NVIDIA, so of course it's different... on both the dev kit and eMMC flavors, the bootloader partitions are in SPI flash, with the kernel/kernel-dtb/etc. residing on SDcard/eMMC.  That said, since u-boot isn't involved, it *might* just work.

kekiefer, 2020-07-02:
Hope springs eternal, I guess.

If needed, an additional override for the flash layout on the NX can be added here: https://github.com/BoulderAI/meta-mender-community/tree/dunfell/meta-mender-tegra/recipes-bsp/tegra-binaries/mender-custom-flash-layout

Here's the diff vs the stock nvidia one for t194. Probably you can just add an override for the nx spi/emmc or spi/sd machine with similar changes:
```
--- flash_t194_sdmmc.xml	2020-04-08 18:26:25.000000000 -0700
+++ flash_mender.xml	2020-07-02 14:22:50.959162900 -0700
@@ -212,7 +212,6 @@
--- flash_t194_sdmmc.xml	2020-04-08 18:26:25.000000000 -0700
+++ flash_mender.xml	2020-07-02 14:25:49.252317203 -0700
@@ -212,7 +212,6 @@
             <allocation_attribute> 0x8 </allocation_attribute>
             <align_boundary> 4096 </align_boundary>
             <percent_reserved> 0 </percent_reserved>
-            <unique_guid> APPUUID </unique_guid>
             <filename> APPFILE </filename>
             <description> **Required.** Contains the rootfs. This partition must be defined after
               `primary_gpt` so it can be accessed as the fixed known special device
@@ -630,6 +629,16 @@
             <description> **Optional.** Reserved for future use by the recovery filesystem;
               removable. </description>
         </partition>
+        <partition name="APP_b" type="data">
+            <allocation_policy> sequential </allocation_policy>
+            <filesystem_type> basic </filesystem_type>
+            <size> APPSIZE </size>
+            <file_system_attribute> 0 </file_system_attribute>
+            <allocation_attribute> 0x8 </allocation_attribute>
+            <align_boundary> 4096 </align_boundary>
+            <percent_reserved> 0 </percent_reserved>
+            <filename> APPFILE </filename>
+        </partition>
         <partition name="UDA" type="data">
             <allocation_policy> sequential </allocation_policy>
             <filesystem_type> basic </filesystem_type>
@@ -637,6 +646,7 @@
             <allocation_attribute> 0x808 </allocation_attribute>
             <percent_reserved> 0 </percent_reserved>
             <align_boundary> 4096 </align_boundary>
+            <filename> DATAFILE </filename>
             <description> **Required.** Automatically takes all remaining space on the device except space
               occupied by `secondary_gpt`. Allocation attribute must be set to 0x808. This
               partition may be mounted and used to store user data. </description>
```

madisongh, 2020-07-03:
I gave it a quick try, and after some hackery I was able to get it to work on my Xavier NX dev kit (the SDcard one).

1. I created a new mender flash layout file by applying the diffs above to the stock flash_l4t_t194_spi_sd_p3668.xml file.
2. I commented out the MENDER_STORAGE_TOTAL_SIZE_MB setting in `testdistro-mender.conf` - there is no EMMC_SIZE setting for the SDcard-based Jetsons, so this was causing parsing errors.
3. I added these settings to `tegra-mender-setup.bbclass`:
```
MENDER_DATA_PART_NUMBER_jetson-xavier-nx-devkit = "12"
MENDER_ROOTFS_PART_B_NUMBER_jetson-xavier-nx-devkit = "11"
```
4. I set IMAGE_OVERHEAD_FACTOR to "1.0" so the partition sizes wouldn't be too large for the total SDcard space.

The big problem appears to be with the NVIDIA flashing tools, which were crapping out on the additions to the flash layout file - looks like `tegraparser_v2` in particular uses `mmap()` to map in the files, causing it to get ENOMEM errors and/or segfaults when there is more than one multi-gigabyte partition that needs mapping.  I worked around it by removing the filenames for APP_b and UDA from the XML file and then writing those partitions manually after running `dosdcard.sh`.

Once I got past those obstacles, I got a bootable image on the SDcard, could register it with my hosted Mender setup, and could use Mender to update.

If I get a chance, I'll see if I can get it working with my eMMC Xavier NX.  I suspect, though, that the tegraparser_v2 problem will prevent me from making it work, since I won't have a way to flash the extra partitions.

madisongh, 2020-07-04:
> The big problem appears to be with the NVIDIA flashing tools, which were crapping out on the additions to the flash layout file

Turns out this is not a new thing, just something I hadn't run into before.  After several experiments I think I've hit on the right way to deal with this... and the good news is that it simplifies data partition handling.  See #377 for more info.

madisongh, 2020-07-04:
I was also able to get a Mender-enabled image built for the eMMC-based Xavier NX module, after a couple of further tweaks:
1. I put back the MENDER_STORAGE_TOTAL_SIZE_MB setting, adjusting it to be:
```
MENDER_STORAGE_TOTAL_SIZE_MB ?= "${@tegra_mender_calc_total_size(d) - int(d.getVar('TEGRA_FLASH_PARTITIONS_MB'))}"
```
2. I change the part number settings to be:
```
MENDER_DATA_PART_NUMBER_xavier-nx = "12"
MENDER_ROOTFS_PART_B_NUMBER_xavier-nx = "11"
```
since the same numbering is used on both the SDcard and the eMMC for Xavier NX (thanks to the use of SPI flash for booting on both).

kekiefer, 2020-07-04:
That's great! Thanks for taking at look at those, Matt. I'll start incorporating these changes to the dunfell branch we've got going for meta-mender-community this weekend, as time permits, unless you've already got a working branch. At some point we'll consolidate all this into one downstream since mendersoftware/meta-mender-community is still on zeus.

madisongh, 2020-07-05:
> At some point we'll consolidate all this into one downstream since mendersoftware/meta-mender-community is still on zeus.

Sounds good.  I've merged #377 after running some test builds and flashing operations across the platforms. With those changes in place, you should be able to remove all of the `tegraflash_XXX` functions/appends from `image_types_mender_tegra.bbclass` and just set
```
IMAGE_TEGRAFLASH_DATA ?= "${IMGDEPLOYDIR}/${IMAGE_LINK_NAME}.dataimg"
```

arrow53, 2020-10-02:
If i'd like to work with mender on the xavier nx is my best present path to use `OE4T/meta-mender-community` and the zeus branch? It seems this is still a work in progress but it appears it's far enough along to try out?

kekiefer, 2020-10-02:
Mender support is now complete for all machines in the current dunfell-l4t-r32.4.3 and master branches. Have a look at the `tegrademo-mender` distro here: https://github.com/OE4T/tegra-demo-distro

arrow53, 2020-10-02:
Thanks! I've never used this demo project repo before but after following the instructions it's building so greatly appreciate the work.

outbackdingo, 2020-10-03:
> Mender support is now complete for all machines in the current dunfell-l4t-r32.4.3 and master branches. Have a look at the `tegrademo-mender` distro here: https://github.com/OE4T/tegra-demo-distro

tegrademo-mender ? i dont see a branch or such, i see how to build it, but nothing explains much on how to get a mender build out of it. Nothing in the wiki.......  i can build demo-oimage-full but dont see mender is included. nothing in the conf/local.conf file for it

kekiefer, 2020-10-03:
Mender support is provided when your `DISTRO` is set to `tegrademo-mender`. When you run `setup-env`, you can pass the `--distro tegrademo-mender` option, or set `DISTRO` to `tegrademo-mender` in the `$BUILDDIR/conf/local.conf`

outbackdingo, 2020-10-03:
> Mender support is provided when your `DISTRO` is set to `tegrademo-mender`. When you run `setup-env`, you can pass the `--distro tegrademo-mender` option, or set `DISTRO` to `tegrademo-mender` in the `$BUILDDIR/conf/local.conf`

great, now i have a clue.... much appreciated!

outbackdingo, 2020-10-03:
okay now what did i miss!

. ./setup-env --machine jetson-xavier-nx-devkit  --distro tegrademo-mender
NVIDIA_DEVNET_MIRROR location not found; some recipes may not be available
### Shell environment set up for builds. ###

You can now run 'bitbake <target>'
Targets for building Tegra demo images with test applications:
    core-image-minimal  - minimally bootable image (no demo apps, no graphics)
    demo-image-base     - basic image with no graphics
    demo-image-egl      - basic image with DRM/EGL, no window manager
    demo-image-sato     - X11 image with 'sato' UI
    demo-image-weston   - Wayland with Weston compositor
    demo-image-full     - X11/sato UI plus docker, openCV, multimedia API samples
dingo@gitlab:~/tegra-demo-distro/build$ bitbake demo-image-full
ERROR: ParseError in configuration INHERITs: Could not inherit file classes/mender-full.bbclass

outbackdingo, 2020-10-03:
nevermind... missing layers in bblayer.conf ... fixed

dwalkes, 2020-10-04:
FYI I submitted a PR at https://github.com/mendersoftware/meta-mender-community/pull/175 to update documentation in meta-mender-community, and started rework on mender hub pages.

arrow53, 2020-10-04:
This isn't really an issue, but I'm looking for more info on meta-tegrademo. 

Is it best to keep the demo layer and then build on top of it with my own layer?

Or, is it recommended method to re-work the demo layer into my own layer?

BTW, thanks again for adding in mender support to the xavier nx. What a time saver.

dwalkes, 2020-10-04:
> Is it best to keep the demo layer and then build on top of it with my own layer?
> Or, is it recommended method to re-work the demo layer into my own layer?

We had this same discussion in https://github.com/madisongh/test-distro/issues/5#issuecomment-687850353 and you can see where we ended up based on Kurt's suggestion [here](https://github.com/madisongh/test-distro/issues/5#issuecomment-689096343).

I actually haven't had a chance to dig in to the final implementation myself yet but it's on my list to put some documentation together about this.  I think we could either use an update to the README or a new wiki page on tegra-demo-distro about this.

arrow53, 2020-10-05:
Thanks @dwalkes that was a very useful read.

madisongh, 2020-11-02:
I think the original question here has been answered, so am going to close.

closed 2020-11-02 by madisongh