OE4T/meta-tegra #343: Question on Xavier NX SPI boot rom

acostach, 2020-06-09:
Hi @madisongh , do you happen to know how it would be possible to update the boot blob inside the 32MB SPI chip from the OS, without connecting it to a PC? I mean the first one in the xml, which contains nvtboot, cboot, etc.

The boot hw partitions are empty and I've tried flashrom on the exposed spidevs, as the [RPi4 uses this procedure](https://github.com/raspberrypi/rpi-eeprom/blob/45935788fddd94fb2c17105855500d345d751187/rpi-eeprom-update#L213), but it didn't detect anything. Guess this was quite expected.

madisongh, 2020-06-09:
There is bootloader update support. See [this section](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fbootloader_update_agx_tx2.html%23) of the L4T documentation. You can generate a BUP payload for the Xavier with the `kernel-bup-payload` recipe, and the tools are installed by including `tegra-redundant-boot` in your image.

I have tested this on the Xavier NX, and it does work.

acostach, 2020-06-11:
Thank you very much Matt, this info helped a lot!

closed 2020-06-11 by acostach